Koenkk/zigbee2mqtt #16140: Z2MQTT Add-on stops when losing connection to coordinator - error Failed to stop Zigbee2MQTT

mihsu81, 2023-01-13:
### What happened?

The Z2MQTT Add-on stops when losing connection to the coordinator.
The HA server and LilyZig coordinator are connected to the same router.
After I reboot the router (takes about 30-40 seconds), within ~20 seconds the Z2MQTT Add-on stops with the below error:
```
Zigbee2MQTT:error 2023-01-13 11:12:21: Adapter disconnected, stopping
Zigbee2MQTT:debug 2023-01-13 11:12:21: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2023-01-13 11:12:21: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
Zigbee2MQTT:info  2023-01-13 11:12:21: Disconnecting from MQTT server
Zigbee2MQTT:info  2023-01-13 11:12:21: Stopping zigbee-herdsman...
Zigbee2MQTT:error 2023-01-13 11:12:21: Failed to stop Zigbee2MQTT
```

### What did you expect to happen?

The Z2MQTT add-on retries connecting to the coordinator a configurable number of times.

### Zigbee2MQTT version

1.29.1-1

### Adapter firmware version

20220219

### Adapter

ZigStar LilyZig POE

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10410404/log.txt)

mihsu81, 2023-02-13:
The issue is still present.

midlan, 2023-05-26:
I discovered similar issue.

After I disconnected SLZB-06 from ethernet cable, the zigbee2mqtt service crashed (understandably). Then I connected again my SLZB-06 gateway, but the zigbee2mqtt service) stayed down. I waited at least 10 minutes, but the watchdog seems not working, seems it does not try to start the service.

mihsu81, 2023-06-26:
The issue is still present in the Z2MQTT 1.31.2-1 add-on and HA 2023.6.3.

mihsu81, 2023-07-28:
The issue is still present in the Z2MQTT 1.32.1-1 add-on and HA 2023.7.3.

mihsu81, 2023-08-29:
The issue is still present in the Z2MQTT 1.32.2-1 add-on and HA 2023.8.4.

lhorak, 2024-01-18:
I have most likely the same or similar issue. I have an ethernet coordinator (TCP). When I, for example, restart my router, zigbee2mqtt crashes and doesn't recover even though the coordinator goes back online. The only way to get things back up and running is by manually restarting the zigbee2mqtt service (mine running barebones in VM). 

I wonder if this is expected behavior or a bug? 

Some logs:

`error 2024-01-16 21:59:38: Adapter disconnected, stopping
debug 2024-01-16 21:59:38: Saving state to file /var/lib/zigbee2mqtt/state.json
info  2024-01-16 21:59:38: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
info  2024-01-16 21:59:38: Disconnecting from MQTT server
info  2024-01-16 21:59:38: Stopping zigbee-herdsman...
error 2024-01-16 21:59:38: Failed to stop Zigbee2MQTT`

mihsu81, 2024-01-18:
I set up an automation which starts the Add-On  if it's not running for 1 minute and 10 seconds.
Nonetheless, I hope this bug will be fixed eventually.
```yaml
alias: Start Zigbee2MQTT Add-On if stopped
description: ""
trigger:
  - type: not_running
    platform: device
    device_id: 2cd20c96528ec97880b06007e39d7c06
    entity_id: binary_sensor.zigbee2mqtt_running
    domain: binary_sensor
    for:
      hours: 0
      minutes: 1
      seconds: 10
condition: []
action:
  - service: hassio.addon_start
    data:
      addon: 45df7312_zigbee2mqtt
mode: single
```

3vilson, 2024-01-21:
Got it woking on Proxmox **Alpine LXC**.

Create the directory:
```
mkdir /opt/zigbee2mqtt
```
Create a script to check if the Zigbee2MQTT service crashed then execute a reboot.
```
cat <<EOF >/opt/zigbee2mqtt/check_zigbee_service.sh
if rc-service zigbee2mqtt status | grep -q "started"; then
    echo "zigbee2mqtt is running."
else
    echo "zigbee2mqtt is not running. Restarting..."
    #rc-service zigbee2mqtt restart
	reboot
fi
EOF
```
Give it execute permissions:
```
chmod +x /opt/zigbee2mqtt/check_zigbee_service.sh
```

Open your crontab file for editing:
```
nano /etc/crontabs/root
```
Add a line to schedule the script to run at your desired interval. For example, to run every 5 minutes:
```
*/5		*	*	*	*	/opt/zigbee2mqtt/check_zigbee_service.sh
```

**reboot**

lhorak, 2024-01-23:
Just to give an update here, I was running  z2m on Proxmox Alpine LXC, I checked the build version of the Alpine package: https://pkgs.alpinelinux.org/package/edge/community/aarch64/zigbee2mqtt and found out the latest version in the repository is 1.34.0, while z2m is on 1.35.1 currently. 

I prefer to have things updated, so I've migrated to LXC with z2m running under docker, and I just tested and when unplugging the IP coordinator and plugging back in, z2m successfully starts up on it's own, so this has solved it for me (and as an added bonus I get instant updates instead of waiting for the Alpine package to get updated 🙂 )

I know this does not solve the issue and brings a little overhead with running Docker instead of barebones, but I just wanted to add this here as another option that is proven to work.

mrbrdo, 2024-03-08:
I have the same issue. After unplugging my ethernet coordinator, the z2m addon crashes and does not come back up after reconnecting.
This is a bug with z2m, because the watchdog in HA supervisor is not designed to handle such cases, the addon itself must handle it. See https://github.com/home-assistant/supervisor/pull/3779 when this behavior was updated.
z2m should be updated to handle connection retrying (indefinitely) instead of crashing. If this behavior is problematic for another use case, then it can be a configurable setting.

The issue is still present with HA 2024.2.5, supervisor 2024.02.1, HAOS 12.0 and Zigbee2MQTT 1.36.0-1.

@3vilson @lhorak in my opinion what you wrote is not really relevant to this issue. The issue does not pertain to running on Proxmox and therefore this cannot be a solution. You have a different setup.

psarossy, 2024-04-19:
Sill an issue on HA 2024.4.3, supervisor 2024.04.0, HAOS 12.2 Zigbee2MQTT 1.36.1-1

r01k, 2024-04-27:
How come it completely stops when it can't connect to the coordinator (which is very likely to occur specially when using network-based coordinators), instead of just checking every few minutes for coordinator availability?

dinhchinh82, 2024-06-02:
I got exactly issue with the latest version z2m 1.38.0. When I unplug the ethernet coordinator, the z2m got crashed and unable to work again even the ethernet coordinator is plugged again with the same IP address.
It seems to be hang forever until I restart the z2m manually again.

Here is the latest log from z2m:

> [2024-06-02 17:13:21] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2024-06-02 17:13:21] error: 	zh:ember:uart:ash: Failed to init port with error Error: connect ECONNREFUSED 192.168.86.27:8888
[2024-06-02 17:13:21] error: 	zh:ember: Failed to reset and init NCP. Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
[2024-06-02 17:13:21] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-06-02 17:13:21] info: 	zh:ember:uart:ash:   Total frames: RX=0, TX=0
[2024-06-02 17:13:21] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-06-02 17:13:21] info: 	zh:ember:uart:ash:   DATA frames : RX=0, TX=0
[2024-06-02 17:13:21] info: 	zh:ember:uart:ash:   DATA bytes  : RX=0, TX=0
[2024-06-02 17:13:21] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2024-06-02 17:13:21] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=0
[2024-06-02 17:13:21] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-06-02 17:13:21] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-06-02 17:13:21] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-06-02 17:13:21] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-06-02 17:13:21] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-06-02 17:13:21] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-06-02 17:13:22] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-06-02 17:13:22] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-06-02 17:13:22] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-06-02 17:13:22] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-06-02 17:13:22] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2024-06-02 17:13:22] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-06-02 17:13:22] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2024-06-02 17:13:22] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2024-06-02 17:13:22] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2024-06-02 17:13:22] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2024-06-02 17:13:22] error: 	z2m: Adapter disconnected, stopping
[2024-06-02 17:13:22] info: 	z2m: Disconnecting from MQTT server
[2024-06-02 17:13:22] info: 	z2m: Stopping zigbee-herdsman...
[2024-06-02 17:46:53] info: 	z2m: Disconnecting from MQTT server
[2024-06-02 17:46:53] info: 	z2m: Stopping zigbee-herdsman...

Bjk8kds, 2024-09-07:
Happened to me too, when i unplugged the coordinator a few seconds its still working properly, but when its unplugged for a longer time, the Z2M just not automatically running, i had to start it manually

Bogey, 2024-12-06:
> I discovered similar issue.
>
> After I disconnected SLZB-06 from ethernet cable, the zigbee2mqtt service crashed (understandably). Then I connected again my SLZB-06 gateway, but the zigbee2mqtt [...]

Same here with a SLZB-06M. It requires restarting the zigbee2mqtt service manually.

snok3r-krk, 2025-01-23:
Is there any technical reason why its designed this way?

ipeacocks, 2025-03-30:
> I set up an automation which starts the Add-On if it's not running for 1 minute and 10 seconds. Nonetheless, I hope this bug will be fixed eventually.
> 
> alias: Start Zigbee2MQTT Add-On if stopped
> description: ""
> trigger:
>   - type: not_running
>     platform: device
>     device_id: 2cd20c96528ec97880b06007e39d7c06
>     entity_id: binary_sensor.zigbee2mqtt_running
>     domain: binary_sensor
>     for:
>       hours: 0
>       minutes: 1
>       seconds: 10
> condition: []
> action:
>   - service: hassio.addon_start
>     data:
>       addon: 45df7312_zigbee2mqtt
> mode: single

I've updated a bit to run periodically:
```
alias: Start Zigbee2MQTT Add-On if stopped
description: ""
triggers:
  - type: not_connected
    device_id: d4137a058a5843754a4e38af9829816a
    entity_id: 2b7a2f6f438ba91431d5796d025f2b9c
    domain: binary_sensor
    trigger: device
    for:
      hours: 0
      minutes: 1
      seconds: 10
  - trigger: time_pattern
    minutes: /5
    enabled: true
conditions: []
actions:
  - data:
      addon: 45df7312_zigbee2mqtt
    action: hassio.addon_start
mode: single
```

sdeep211, 2025-04-01:
This is also an issue for me. 
Z2M has to be restarted manually if the coordinator disconnects from the network for any reason e.g. router reboot, power cut, cable disconnect. Only found out this was an issue after a recent power cut.
Recently moved to using an SLZB-06 over POE coming from a Sonoff USB coordinator previously.
Z2M running as an add on in HASOS.
Guess I will also implement an automation to restart Z2M periodically if not running.

christerdk, 2025-04-17:
Also an issue here. I'd expect it to retry once in a while until the coordinator comes back online.

KevinGaudry, 2025-05-06:
Same problem here, I have Z2M in an LXC under proxmox and as soon as HA reboots or the LXC looses connection, it crashes and must be restared manually. PoE SLZB-06 coordinator. 14:07 is the time of reboot

`[2025-05-06 13:03:04] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-05-06 13:03:29] error:    zh:zstack:znp: Socket error Error: read ETIMEDOUT
[2025-05-06 13:03:29] info:     zh:zstack:znp: Port closed
[2025-05-06 13:03:29] debug:    zh:controller: Adapter disconnected
[2025-05-06 13:03:29] info:     zh:zstack:znp: closing
[2025-05-06 13:03:29] error:    z2m: Adapter disconnected, stopping
[2025-05-06 13:03:29] debug:    z2m: Saving state to file /opt/zigbee2mqtt/data/state.json
[2025-05-06 13:03:29] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-05-06 13:03:29] info:     z2m: Disconnecting from MQTT server
[2025-05-06 13:21:50] debug:    zh:controller:database: Writing database to '/opt/zigbee2mqtt/data/database.db'
[2025-05-06 14:07:41] debug:    z2m: Saving state to file /opt/zigbee2mqtt/data/state.json
[2025-05-06 14:07:41] error:    z2m: Not connected to MQTT server!
[2025-05-06 14:07:41] error:    z2m: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: '{"state":"offline"}
[2025-05-06 14:07:41] info:     z2m: Disconnecting from MQTT server
[2025-05-06 14:07:41] info:     z2m: Stopping zigbee-herdsman...
[2025-05-06 14:07:41] debug:    zh:controller:database: Writing database to '/opt/zigbee2mqtt/data/database.db'
[2025-05-06 14:07:41] info:     z2m: Stopped zigbee-herdsman
[2025-05-06 14:07:41] info:     z2m: Stopped Zigbee2MQTT`

christerdk, 2025-05-08:
Had a power blackout yesterday in our building, same thing happened. Doesn't recover with the automation above. 

Let get this straight: a queuing system made to facilitate resilient communication cannot recover from a glitch without human intervention. Come on...

KevinGaudry, 2025-06-09:
any update on this issue ? would be nice to have.

gromitn, 2025-08-10:
Yes please!!! Urgh this is very frustrating. I am having an issue with my UZG-01 losing connection intermittently, (at this stage not sure if a brief network disturbance etc) but was led down the garden path of Z2M giving up and shutting down, leaving every device in the house unusable, when all I needed to fix it was to manually start the add-on again. Seems kinda crazy that I'm looking at needing to build an automation to restart it.

gromitn, 2025-10-20:
Others have posted here if you trawl back up the thread - I use this set at 5 mins so it allows some manual restarts etc, and things aren't that minute-critical for me. Tailor as complex as you like, mine is pretty simple, but works for my use case.

```
alias: Restart Zigbee2MQTT if stopped
description: Restarts Zigbee2MQTT if the bridge is offline for more than 5 minutes
triggers:
  - entity_id: binary_sensor.zigbee2mqtt_bridge_connection_state
    to: "off"
    for:
      minutes: 5
    trigger: state
conditions: []
actions:
  - data:
      addon: 45df7312_zigbee2mqtt
    action: hassio.addon_start
  - data:
      title: Zigbee2MQTT Restarted
      message: Zigbee2MQTT bridge was offline and has been restarted automatically.
    action: persistent_notification.create
  - data:
      title: Zigbee2MQTT Restarted
      message: Zigbee2MQTT bridge was offline and has been restarted automatically.
    action: notify.mobile_app_sm_s921b
mode: single
```

valeriob1, 2025-10-30:
I'm facing the same issue. I'll be interested as well in a fix or applicable workaround.
I'm not very familiar with scrips, hopefully it gets solved.

lyricnz, 2025-11-04:
Same. This is super annoying. Anyone got a workaround?
(my HA is on UPS/battery etc, but when we have a power outage my network switches/APs, and my zig coordinator, all go out - and z2m addon never recovers)

dragosmiron, 2025-11-07:
this is my automation btw

```yml
alias: Restart Zigbee2MQTT if needed
description: >-
  Restarts Zigbee2MQTT if the bridge is offline for more than 5 minutes,
  retrying until success, or when the SLZB-06M ethernet connection comes online
  and remains stable for 5 minutes
triggers:
  - entity_id: binary_sensor.zigbee2mqtt_bridge_connection_state
    to: "off"
    for:
      minutes: 5
    trigger: state
    alias: Bridge offline
  - entity_id: binary_sensor.slzb_06m_ethernet
    to: "on"
    for:
      minutes: 5
    trigger: state
    alias: Ethernet connected
conditions: []
actions:
  - data:
      action: hassio.addon_start
      addon: 45df7312_zigbee2mqtt
      retries: 10
      validation: >-
        [[ states('binary_sensor.zigbee2mqtt_bridge_connection_state') == 'on'
        ]]
      backoff: "[[ 30 * 2 ** attempt ]]"
    action: retry.action
  - if:
      - condition: state
        entity_id: binary_sensor.zigbee2mqtt_bridge_connection_state
        state: "off"
    then:
      - data:
          title: Zigbee2MQTT Failed to Restart
          message: >-
            All restart attempts failed. Zigbee2MQTT bridge is still offline
            after 10 retries.
        action: persistent_notification.create
      - device_id: f77e35527aeb2123c4c015c5277c997d
        domain: mobile_app
        type: notify
        message: Zigbee2MQTT Failed to start
    else:
      - data:
          title: Zigbee2MQTT Restarted
          message: Zigbee2MQTT bridge was offline and has been restarted automatically.
        action: persistent_notification.create
mode: queued
```

hheimbuerger, 2025-11-08:
This is pretty frustrating, I honestly don't think the container should terminate, just because the device is unavailable for a second…

As an intermediate solution: does anybody have an idea how to restart a container running in Synology Container Station? (I guess I could SSH in, but I'd really rather not...)

dragosmiron, 2025-11-08:
> container running in Synology Container Station?

do a VM in the nas with HAOS, any other type of home assistant will or already is [deprecated](https://www.home-assistant.io/blog/2025/05/22/deprecating-core-and-supervised-installation-methods-and-32-bit-systems/)

Impact123, 2025-11-08:
The container install is not deprecated.

hheimbuerger, 2025-11-15:
> > container running in Synology Container Station?
> 
> do a VM in the nas with HAOS, any other type of home assistant will or already is [deprecated](https://www.home-assistant.io/blog/2025/05/22/deprecating-core-and-supervised-installation-methods-and-32-bit-systems/)

As Impact123 already said, Container is not deprecated, and in fact they recommend that Core (deprecated) users should migrate to Container, and Supervised users can migrate to Container.

stopeject, 2025-11-20:
Another user with the same annoying problem.

panther7, 2026-09-02:
My fix:

Helper, return available entities in %.
```yaml
{% set all = states | count %}
{% set unv = states | selectattr('state', 'in', ['unavailable', 'unknown']) | list | count %}
{{ ((1 - unv / all) * 100) | round(1) }}
```

Automation:
```yaml
alias: Restart Zigbee2MQTT if entities not available
description: ''
triggers:
  - trigger: numeric_state
    entity_id:
      - sensor.available_entities
    for:
      minutes: 10
    below: 66
conditions:
  - condition: template
    value_template: '{{ now() - as_datetime(states(''sensor.uptime'')) > timedelta(minutes=60) }}'
actions:
  - action: hassio.addon_restart
    data:
      addon: core_mosquitto
  - action: hassio.addon_restart
    data:
      addon: 45df7312_zigbee2mqtt
mode: single
```