Koenkk/zigbee2mqtt #16141: HEIMAN HS2DB - action property is never updated

2t0m, 2023-01-13:
### What happened?

**Problem**

I'm trying to use a HEIMAN HS2DB doorbell button, but the `action` property is never updated and the `battery` status seams unrealistic.

I am able to get:
- battery status with 0% (even with a new battery)
- battery low = 'True'
- Link quality which is varying at each action on the button
- action = N/A (not affected by an action on the button)
- temper = false (not affected by an action on the button)

**database.db entry of the device**

`{"id":2,"type":"EndDevice","ieeeAddr":"0x000d6f00161f6b79","nwkAddr":27308,"manufId":4619,"manufName":"HEIMAN","powerSource":"Battery","modelId":"DoorBell-EM","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"DoorBell-EM","manufacturerName":"HEIMAN","powerSource":3,"zclVersion":2,"appVersion":21,"stackVersion":2,"hwVersion":16,"dateCode":"2018.10.12"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0024c12a2d","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":0}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c12a2d","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":21,"stackVersion":2,"hwVersion":16,"dateCode":"2018.10.12","zclVersion":2,"interviewCompleted":true,"meta":{"configured":491743244},"lastSeen":1673602544130,"defaultSendRequestWhen":"immediate"}`

### What did you expect to happen?

I expect to get a `pressed` state in `action` property when I press the button.
I expect to get a more realistic value in the `battery` property.

### Zigbee2MQTT version

1.29.1

### Adapter firmware version

20210708

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus

### Debug log

```
Zigbee2MQTT:debug 2023-01-13 11:10:23: Received Zigbee message from 'Sonnette', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32776}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-01-13 11:10:23: MQTT publish: topic 'zigbee2mqtt/Sonnette', payload '{"battery":0,"battery_low":true,"linkquality":108,"tamper":false}'
```

2t0m, 2023-01-14:
Today the `zonestatus` has changed from to `32776` to `32768`.

```
Zigbee2MQTT:debug 2023-01-14 09:31:29: Received Zigbee message from 'Sonnette', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32768}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-01-14 09:31:29: MQTT publish: topic 'zigbee2mqtt/Sonnette', payload '{"action":"pressed","battery":100,"battery_low":false,"last_seen":"2023-01-14T08:31:29.566Z","linkquality":134,"tamper":false,"voltage":3000}'
Zigbee2MQTT:info  2023-01-14 09:31:29: MQTT publish: topic 'zigbee2mqtt/Sonnette', payload '{"action":"","battery":100,"battery_low":false,"last_seen":"2023-01-14T08:31:29.566Z","linkquality":134,"tamper":false,"voltage":3000}'
Zigbee2MQTT:info  2023-01-14 09:31:29: MQTT publish: topic 'zigbee2mqtt/Sonnette/action', payload 'pressed' 
```

The button click now induces the `action` property change and the `battery` property looks better.

closed 2023-01-14 by 2t0m

OggyParx, 2024-10-03:
Hi @2t0m 

How do you fix this Problem ? How can I change the zonestatus ?