Koenkk/zigbee2mqtt #16142: Elko Thermostat ESH Plus Super TR RF PH

t112013, 2023-01-13:
### What happened?

Hello, iam quite sure there is something wrong with the support of this unit, i believe there are two versions of it, most likely a new revision, i have 4 at home, 3 "old" in which one of those doesnt want to pair at all (so iam actually getting a new one from elko)  and one where the PCB even has another color. 
The thing is that i have been strugling with network stability and i activated the last seen attribute and then i can see that the "new" version is repporting verything 1.2 or even 3 minuttes where as the "old" ones are hammering every second.
This is acutally the reason why i didnt use Home Asssitants Zigbee integration becasue there the "old" thermostats cannot change the target temperature value. 

### Zigbee2MQTT version

1.29.1-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220810164645-if00

arnerek, 2023-01-27:
I also had this issue that I got reporting every second. It stopped when I commented the reporting of all the Elko attributes (from line 92 to 170) but I don't know why:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/84edb807202652de80071437fcefa33c6b205fee/devices/elko.js#L92 

Could you post a screenshot of the reporting tab? There is something odd here, as most attributes appeared as numbers and not as name, and I were not able to change the reporting interval.

tore1980, 2023-01-28:
I have the same report interval (Model 4523430), every second or faster and making the network unstable.
If I make changes or just press "apply" I get an error:

023-01-28 21:08:16Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x000d6f000d978bc0/1 hvacThermostat([{"attribute":1025,"minimumReportInterval":"1","maximumReportInterval":3600,"reportableChange":"1"}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Don't know value type for 'undefined')' 
[image]
[image]
 
[image]

arnerek, 2023-01-29:
I think something is incorrect in the code that configures the reporting as they are not identified by name only by number. 

And also configuring fails as it does not know what kind of data type these attributes are. I have looked at the code many times, but I have not fully understood the Z2M Converter terminology yet.

closed 2023-03-08 by github-actions[bot]

t112013, 2023-04-15:
could anyone help ? there is clearly something wrong