Koenkk/zigbee2mqtt #16145: xiaomi/Aqara T1 2 gang smart wall switch (no neutral wire) QBKG18LM

richard-x, 2023-01-13:
### Link

https://www.aqara.com/cn/d31_spec

### Database entry

{"id":11,"type":"EndDevice","ieeeAddr":"0x54ef44100060da3d","nwkAddr":23381,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.switch.b2lacn01","epList":[1,2,41,42,51,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2,3,4,5,6,18,64704],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"dateCode":"04-13-2020","appVersion":32}},"aqaraOpple":{"attributes":{"247":{"type":"Buffer","data":[100,16,0,101,16,0,3,40,14,5,33,1,0,154,32,0,8,33,32,1,10,33,0,0,11,32,0]}}},"genOnOff":{"attributes":{"245":56317184,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6,18,64704],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"245":117440602,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"41":{"profId":260,"epId":41,"devId":256,"inClusterList":[18],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"42":{"profId":260,"epId":42,"devId":256,"inClusterList":[18],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"51":{"profId":260,"epId":51,"devId":256,"inClusterList":[18],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":32,"stackVersion":2,"hwVersion":1,"dateCode":"04-13-2020","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1209552899},"lastSeen":1673623299346,"defaultSendRequestWhen":"immediate"}

### Comments

In the web interface, only the left button can be controlled, and the right button cannot be controlled. thanks.

### External converter

```shell
{
    zigbeeModel: ['lumi.switch.b2lacn01'],
    model: 'QBKG18LM',
    vendor: 'Aqara',
    description: '双键开关',
    fromZigbee: [fz.on_off, fz.xiaomi_on_off_action, fz.xiaomi_on_off_ignore_endpoint_4_5_6, fz.xiaomi_operation_mode_basic],
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right'), e.action(['single_left', 'single_right', 'single_both'])],    
    toZigbee: [tz.on_off, tz.xiaomi_power],
    meta: {multiEndpoint: true, configureKey: 1},
    endpoint: (device) => {
            return {'system': 1, 'left': 2, 'right': 3};
        },
    configure: async (device, coordinatorEndpoint, logger) => {
    	   device.type = 'EndDevice';
        device.save();
    },
},
```

Koenkk, 2023-01-14:
What if you change `return {'system': 1, 'left': 2, 'right': 3};` to `return {'left': 1, 'right': 2};`

richard-x, 2023-01-15:
> 如果你`return {'system': 1, 'left': 2, 'right': 3};`改成`return {'left': 1, 'right': 2};`
Thank you very much for your kind help. My problem has been solved. How to learn the meaning of these parameters? Where can I get relevant information? Thanks again.

richard-x, 2023-01-15:
> > 如果你`return {'system': 1, 'left': 2, 'right': 3};`改成`return {'left': 1, 'right': 2};`
> > 非常感谢你的帮助。我的问题已经解决了。如何学习这些参数的含义呢？我在哪里可以获得相关信息？再次感谢。

richard-x, 2023-01-15:
> What if you change to `return {'system': 1, 'left': 2, 'right': 3};``return {'left': 1, 'right': 2};`

Thank you very much for your kind help. My problem has been solved. How to learn the meaning of these parameters? Where can I get relevant information? Thanks again.

Koenkk, 2023-01-16:
From the `data/database.db` entry you see that it doesn't have and endpoint 3.

richard-x, 2023-01-18:
> From the `data/database.db` entry you see that it doesn't have and endpoint 3.

Oh, I see where you are pointing. How do I understand what other parameters mean? How to draw inferences from one instance and write configuration files for other devices? For example, I recently bought the curtain motor (TERNCY-CM01) of Xiaoyan, which is not in the support list. I want to learn how to read the meaning of parameters in database entries.

closed 2023-02-25 by github-actions[bot]