Koenkk/zigbee2mqtt #16147: [Feature request]: GP-WOU019BBDWG power_on_behavior

Ulrar, 2023-01-13:
### Is your feature request related to a problem? Please describe

Hi,

Just migrated from ZHA to Z2M, and I noticed that the samsung smartthings plugs (GP-WOU019BBDWG) don't expose the power_on_behavior like they do in ZHA.

### Describe the solution you'd like

Not sure what the default behavior is, but it'd be nice to have that configurable.
In ZHA it shows up as a drop down under configuration for the device, with the choice of on / off / previous.

### Describe alternatives you've considered

-

### Additional context

-

Koenkk, 2023-01-14:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-14 by Koenkk

Ulrar, 2023-01-14:
Awesome, thanks a lot for the quick add !