Koenkk/zigbee2mqtt #16148: Hue Dimmer Switch updating to the same version

Ulrar, 2023-01-13:
### What happened?

Hi,

I have two hue dimmer switch (model [324131092621](https://www.zigbee2mqtt.io/devices/324131092621.html#philips-324131092621)) which reporting they have an update available.
When I run that update they seem to go to 100% and succeed, but they stay on the same version.

I have a feeling I already reported this, or found another issue about it, a couple of years ago but I can't find it now. Apologies if I missed another existing issue.

### What did you expect to happen?

Looking at other switches of the same model I have, it seems the latest firmware version for these is 6.1.1.28573.

I would expect the firmware version to go from 5.45.1.17846 to 6.1.1.28573 after the update.

### How to reproduce it (minimal and precise)

Not sure, only two of my switches are doing this, the other ones all updated fine.

### Zigbee2MQTT version

1.29.1 commit: 7d67ffc

### Adapter firmware version

20220219

### Adapter

zStack3x0 - tubeszb-cc2652 PoE model over TCP. I did have the same issue a couple of years ago using a ZZH

Ulrar, 2023-02-13:
Not stale

Ulrar, 2023-03-16:
Not stale

Ulrar, 2023-04-16:
Not stale

Ulrar, 2023-05-18:
Not stale

Ulrar, 2023-06-18:
Not stale

nriendeau, 2023-06-24:
@Ulrar, the firmware most likely updated, it is just that some of Zigbee2MQTT config files were not updated and show you the old version number.

Your problem is actually already included in https://github.com/Koenkk/zigbee2mqtt/issues/10139 .

If you try the possible fixes suggested there (removing and repairing, editing database.db to remove your devices and repair, etc...) one of the will most likely work and you will see that your dimmers are already updated.

For one of my dimmers simply removing and repairing fixed it but for the other I had to remove and blocklist it, remove it from the blocklist and repair...

I assume that putting it in the blocklist resulted in the device being removed from database.db, cleaned up the information in state.json and other similar files. My guess is stopping Zigbee2MQTT and editing database.db to remove the dimmer (make a backup first) and restarting Zigbee2MQTT would have given the same result.

Good luck and have a nice day!

Nicolas

Ulrar, 2023-07-03:
Sorry for the delay, finally got around to it.

So I first just removed it from database.db and re-paired it, but no change.
I then did what you suggested and blocklisted it, then unblocked it and re-paired it and indeed it does seem to have done the trick. I'll keep an eye on it to make sure it stays 'up to date' but that looks good at first glance, thanks for the tip !

closed 2023-07-03 by Ulrar