Koenkk/zigbee2mqtt #16151: TuYa TS130F wrong direction

JakubMartenka, 2023-01-13:
### What happened?

When I physically click close the blind closes. 
But when I click on HA the blind opens.

Zigbee2MQTT - 1.29.1-1

### Zigbee2MQTT version

1.29.1-1

### Adapter firmware version

20230104.0

### Adapter

TS130F

JakubMartenka, 2023-01-14:
When motor_reversal is ON from HA works perfect but when I click on physically button the opposite turns on

Video:
https://user-images.githubusercontent.com/97041743/212461932-18358ea0-1e0f-479b-8397-2c2cfd1b5dee.MOV

https://vimeo.com/789223128

Koenkk, 2023-01-14:
Can you provide the data/database.db entry of this device?

JakubMartenka, 2023-01-14:
@Koenkk no problem, but idk how do it :D  where is this file?

JakubMartenka, 2023-01-14:
@Koenkk this is it? `{"id":6,"type":"Router","ieeeAddr":"0xa4c1385e84caab90","nwkAddr":6943,"manufId":4417,"manufName":"_TZ3000_wptayaqr","powerSource":"Mains (single phase)","modelId":"TS130F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[4,5,6,258,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"tuyaBacklightSwitch":1,"tuyaBacklightMode":1}},"closuresWindowCovering":{"attributes":{"32768":0,"currentPositionLiftPercentage":18,"tuyaMovingState":1,"tuyaCalibration":1,"tuyaMotorReversal":1,"moesCalibrationTime":122}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1673720380138,"defaultSendRequestWhen":"immediate"}
`

Koenkk, 2023-01-15:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition =  {
    zigbeeModel: ['TS130F'],
    model: 'TS130F',
    vendor: 'TuYa',
    description: 'Curtain/blind switch',
    fromZigbee: [fz.cover_position_tilt, tuya.fz.backlight_mode_low_medium_high, fz.tuya_cover_options],
    toZigbee: [tz.cover_state, tz.cover_position_tilt, tz.tuya_cover_calibration, tz.tuya_cover_reversal,
        tuya.tz.backlight_indicator_mode],
    whiteLabel: [{vendor: 'LoraTap', model: 'SC400'}],
    exposes: [e.cover_position(), exposes.enum('moving', ea.STATE, ['UP', 'STOP', 'DOWN']),
        exposes.binary('calibration', ea.ALL, 'ON', 'OFF'), exposes.binary('motor_reversal', ea.ALL, 'ON', 'OFF'),
        exposes.enum('backlight_mode', ea.ALL, ['LOW', 'MEDIUM', 'HIGH']),
        exposes.numeric('calibration_time', ea.STATE).withUnit('S').withDescription('Calibration time')],
},;

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

JakubMartenka, 2023-01-15:
@Koenkk the external converter did not fix it. 
I changed the description to make sure it was added.
[image]

JakubMartenka, 2023-01-16:
@Koenkk i try to understand how it works and i have one idea. 
Maybe we need here change when motor_reversal to {0: 'DOWN', 1: 'STOP', 2: 'UP'}

[image]

JakubMartenka, 2023-01-16:
@Koenkk  No, the same problem with external_converters too. When i i have disabled motor_reversal on physical buttons it works fine but in HA close = open and open = close and when i enable motor_reversal I still have this problem https://vimeo.com/789223128

Koenkk, 2023-01-17:
Can you provide the debug log when opening the cover via HA and it closes instead?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

JakubMartenka, 2023-01-17:
Open and stop with motor_reversal = ON from HA
[image]

Close and stop with motor_reversal = ON from HA
[image]

Open and stop motor_reversal = ON from button
[image]

Close and stop motor_reversal = ON from button
[image]

Open and stop with motor_reversal = OFF from HA
[image]

Close and stop with motor_reversal = OFF from HA
[image]

Open and stop motor_reversal = ON from button
[image]

Close and stop motor_reversal = OFF from button
[image]

closed 2023-02-25 by github-actions[bot]