Koenkk/zigbee2mqtt #16153: [FUT036Z] power_on_behavioir shown but unsupported

emandtf, 2023-01-13:
### What happened?

Recently for Miboxer FUT036Z devices Z2M started to shown "power_on_behaviour" feature but when used it returns immediately an NOT_SUPPORTED erro as red toast on Z2M's frontend.

I suppose this feature was added by mistake.

### What did you expect to happen?

Not shown an UNSUPPORTED feature on Z2M frontend.

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

0x26780700 (May 2022)

### Adapter

Conbee 2

Koenkk, 2023-01-14:
Seems this device does not support it, removed it!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-14 by Koenkk

JakeMannen, 2023-02-23:
This also seems to be the case for the **FUT035Z**. The power_on_behavior is not working.

Koenkk, 2023-02-25:
@JakeMannen disabled!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)