Koenkk/zigbee2mqtt #16160: New (unsupported) Bosch Devices (version II) cannot be interviewed

g-jaeger, 2023-01-14:
### What happened?

I tried to pair a (not) yet supported Bosch Door/Window Contact II device, but it failed at the _interview_ step.
Seems this applies to other Bosch (version II) devices.
See discussion [here](https://github.com/Koenkk/zigbee2mqtt/discussions/15966#discussion-4726771).

I'm willing to help debugging this issue. If necessary, I could provide such a device.

TIA,
 Gerhard

### What did you expect to happen?

 Successful pairing.

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

20220219

### Adapter

ZigStar GW

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10417908/log.txt)
[interview-failure.log](https://github.com/Koenkk/zigbee2mqtt/files/10417911/interview-failure.log)

noderat, 2023-01-14:
Possibly related to https://github.com/Koenkk/zigbee2mqtt/issues/13520

g-jaeger, 2023-01-15:
Thanks for that reference. According to that thread some stuff needs to be done, so I
* took the latest-dev docker image
* added the proposed install.js file (and changed ieeeaddr to the one of my device)

But then noticed the thingy with the network key (installcode)  - I don't have one :(
Also noticed that there's no QR-Code for that device - so maybe no installcode needed for it?

Any hint about how I could proceed?

TIA

g-jaeger, 2023-01-16:
First of all: Sorry for the noise. I was not patient enough. Finally I could find the QR code (inside the battery slot) and could extract the install-code. By adding that to the install.js file, I could get the interview phase work.

Will now continue with the converter stuff...

g-jaeger, 2023-01-16:
Close ticket as pilot error. Sorry for the noise. Pairing/interviewing works. Maybe lack of documentation could be blamed.

closed 2023-01-16 by g-jaeger