Koenkk/zigbee2mqtt #16161: TS0601 thermostat only first 2 items of the sent payload are set

ericvb, 2023-01-14:
### What happened?

Was trying to set the schedules in the programming mode, but the default workdays_schedule was never changed after clicking on apply. The week and holidays_schedule were adapted.
[image]

### What did you expect to happen?

In the logfile of Z2M I saw this, a payload with 3 items, but only 2 sets?:
```
Zigbee2MQTT:debug 2023-01-14 15:38:22: Received MQTT message on 'zigbee2mqtt/Badkamer venster/set' with data '{"holidays_schedule":"07:00/22°C 08:00/22°C 11:30/22°C 12:30/22°C 14:00/22°C 23:00/16°C","week":"5+2","workdays_schedule":"06:30/24°C 08:00/16°C 11:30/16°C 12:30/16°C 16:30/22°C 22:00/16°C"}'
Zigbee2MQTT:debug 2023-01-14 15:38:22: Publishing 'set' 'holidays_schedule' to 'Badkamer venster'
Zigbee2MQTT:debug 2023-01-14 15:38:23: Publishing 'set' 'week' to 'Badkamer venster'
```
I expected a third line: Zigbee2MQTT:debug 2023-01-14 15:38:22: Publishing 'set' 'workdays_schedule' to 'Badkamer venster'
But this was not generated.

So I did a manual test via the option to publish manually a packet. I reconstructed the same payload, but now in a different order.
```
topic:
zigbee2mqtt/Badkamer venster/set
payload:
{
        "week": "5+2",
        "workdays_schedule": "06:30/27°C 08:00/16°C 11:30/16°C 12:30/16°C 16:30/22°C 22:00/16°C",
        "holidays_schedule": "07:00/28°C 08:00/22°C 11:30/22°C 12:30/22°C 14:00/22°C 23:00/16°C"
}
```
The result was this:
```
Zigbee2MQTT:debug 2023-01-14 15:19:03: Received MQTT message on 'zigbee2mqtt/Badkamer venster/set' with data '{
        "week": "5+2",
        "workdays_schedule": "06:30/27°C 08:00/16°C 11:30/16°C 12:30/16°C 16:30/22°C 22:00/16°C",
        "holidays_schedule": "07:00/28°C 08:00/22°C 11:30/22°C 12:30/22°C 14:00/22°C 23:00/16°C"
}'
Zigbee2MQTT:debug 2023-01-14 15:19:03: Publishing 'set' 'week' to 'Badkamer venster'
Zigbee2MQTT:debug 2023-01-14 15:19:05: Publishing 'set' 'workdays_schedule' to 'Badkamer venster'
```
Now the holidays_schedule was not updated.
I played a little with the order and always only the first 2 items were set.

### How to reproduce it (minimal and precise)

Construct manually a topic and payload and publish it.
```
topic:
zigbee2mqtt/Badkamer venster/set
payload:
{
        "week": "5+2",
        "workdays_schedule": "06:30/22°C 08:00/16°C 11:30/16°C 12:30/16°C 16:30/22°C 22:00/16°C",
        "holidays_schedule": "07:00/22°C 08:00/22°C 11:30/22°C 12:30/22°C 14:00/22°C 23:00/16°C"
}
```

### Zigbee2MQTT version

1.29.2-1

### Adapter firmware version

20220219

### Adapter

CC2652P2 TCP ZigBee Coordinator V0.2 cod.m

### Debug log

```
debug 2023-01-14 16:37:11: Received MQTT message on 'zigbee2mqtt/Badkamer venster/set' with data '{
[... 8 lines cut ...]
info  2023-01-14 16:37:16: MQTT publish: topic 'zigbee2mqtt/Badkamer venster', payload '{"auto_lock":"MANUAL","away_mode":"OFF","away_preset_days":1,"away_preset_temperature":16,"battery_low":false,"boost_time":300,"child_lock":"UNLOCK","comfort_temperature":22,"current_heating_setpoint":22,"eco_temperature":16,"force":"normal","holidays":[{"hour":7,"minute":0,"temperature":22},{"hour":8,"minute":0,"temperature":22},{"hour":11,"minute":30,"temperature":22},{"hour":12,"minute":30,"temperature":22},{"hour":14,"minute":0,"temperature":22},{"hour":23,"minute":0,"temperature":16}],"holidays_schedule":"07:00/22°C 08:00/22°C 11:30/22°C 12:30/22°C 14:00/22°C 23:00/16°C","linkquality":72,"local_temperature":20,"local_temperature_calibration":-3,"max_temperature":35,"min_temperature":10,"position":0,"preset":"manual","running_state":"idle","system_mode":"auto","update":{"installed_version":87,"latest_version":87,"state":"idle"},"update_available":false,"valve_detection":null,"week":"5+2","window_detection":"OFF","window_detection_params":{"minutes":10,"temperature":5},"window_open":false,"workdays":[{"hour":6,"minute":30,"temperature":22},{"hour":8,"minute":0,"temperature":16},{"hour":11,"minute":30,"temperature":16},{"hour":12,"minute":30,"temperature":16},{"hour":16,"minute":30,"temperature":22},{"hour":22,"minute":0,"temperature":16}],"workdays_schedule":"06:30/22°C 08:00/16°C 11:30/16°C 12:30/16°C 16:30/22°C 22:00/16°C"}'
debug 2023-01-14 16:37:16: Received Zigbee message from 'Badkamer venster', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[7,0,25,8,0,22,11,30,22,12,30,22,14,0,22,23,0,16],"type":"Buffer"},"datatype":0,"dp":113}],"seq":3}' from endpoint 1 with groupID 0
info  2023-01-14 16:37:16: MQTT publish: topic 'zigbee2mqtt/Badkamer venster', payload '{"auto_lock":"MANUAL","away_mode":"OFF","away_preset_days":1,"away_preset_temperature":16,"battery_low":false,"boost_time":300,"child_lock":"UNLOCK","comfort_temperature":22,"current_heating_setpoint":22,"eco_temperature":16,"force":"normal","holidays":[{"hour":7,"minute":0,"temperature":25},{"hour":8,"minute":0,"temperature":22},{"hour":11,"minute":30,"temperature":22},{"hour":12,"minute":30,"temperature":22},{"hour":14,"minute":0,"temperature":22},{"hour":23,"minute":0,"temperature":16}],"holidays_schedule":"07:00/25°C 08:00/22°C 11:30/22°C 12:30/22°C 14:00/22°C 23:00/16°C","linkquality":72,"local_temperature":20,"local_temperature_calibration":-3,"max_temperature":35,"min_temperature":10,"position":0,"preset":"manual","running_state":"idle","system_mode":"auto","update":{"installed_version":87,"latest_version":87,"state":"idle"},"update_available":false,"valve_detection":null,"week":"5+2","window_detection":"OFF","window_detection_params":{"minutes":10,"temperature":5},"window_open":false,"workdays":[{"hour":6,"minute":30,"temperature":22},{"hour":8,"minute":0,"temperature":16},{"hour":11,"minute":30,"temperature":16},{"hour":12,"minute":30,"temperature":16},{"hour":16,"minute":30,"temperature":22},{"hour":22,"minute":0,"temperature":16}],"workdays_schedule":"06:30/22°C 08:00/16°C 11:30/16°C 12:30/16°C 16:30/22°C 22:00/16°C"}'
debug 2023-01-14 16:37:21: Received Zigbee message from 'Badkamer venster', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,220],"type":"Buffer"},"datatype":2,"dp":2}],"seq":19200}' from endpoint 1 with groupID 0
info  2023-01-14 16:37:21: MQTT publish: topic 'zigbee2mqtt/Badkamer venster', payload '{"auto_lock":"MANUAL","away_mode":"OFF","away_preset_days":1,"away_preset_temperature":16,"battery_low":false,"boost_time":300,"child_lock":"UNLOCK","comfort_temperature":22,"current_heating_setpoint":22,"eco_temperature":16,"force":"normal","holidays":[{"hour":7,"minute":0,"temperature":25},{"hour":8,"minute":0,"temperature":22},{"hour":11,"minute":30,"temperature":22},{"hour":12,"minute":30,"temperature":22},{"hour":14,"minute":0,"temperature":22},{"hour":23,"minute":0,"temperature":16}],"holidays_schedule":"07:00/25°C 08:00/22°C 11:30/22°C 12:30/22°C 14:00/22°C 23:00/16°C","linkquality":72,"local_temperature":20,"local_temperature_calibration":-3,"max_temperature":35,"min_temperature":10,"position":0,"preset":"manual","running_state":"idle","system_mode":"auto","update":{"installed_version":87,"latest_version":87,"state":"idle"},"update_available":false,"valve_detection":null,"week":"5+2","window_detection":"OFF","window_detection_params":{"minutes":10,"temperature":5},"window_open":false,"workdays":[{"hour":6,"minute":30,"temperature":22},{"hour":8,"minute":0,"temperature":16},{"hour":11,"minute":30,"temperature":16},{"hour":12,"minute":30,"temperature":16},{"hour":16,"minute":30,"temperature":22},{"hour":22,"minute":0,"temperature":16}],"workdays_schedule":"06:30/22°C 08:00/16°C 11:30/16°C 12:30/16°C 16:30/22°C 22:00/16°C"}'
debug 2023-01-14 16:38:12: Saving state to file /config/zigbee2mqtt/state.json
```

```
debug 2023-01-14 16:40:18: Received MQTT message on 'zigbee2mqtt/Badkamer venster/set' with data '{
        "week": "5+2",
        "workdays_schedule": "06:30/27°C 08:00/16°C 11:30/16°C 12:30/16°C 16:30/22°C 22:00/16°C",
        "holidays_schedule": "07:00/28°C 08:00/22°C 11:30/22°C 12:30/22°C 14:00/22°C 23:00/16°C"
}'
debug 2023-01-14 16:40:18: Publishing 'set' 'week' to 'Badkamer venster'
debug 2023-01-14 16:40:18: Publishing 'set' 'workdays_schedule' to 'Badkamer venster'
debug 2023-01-14 16:40:18: Received Zigbee message from 'Badkamer venster', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":111}],"seq":4}' from endpoint 1 with groupID 0
[... cut ...] lable":false,"valve_detection":null,"week":"5+2","window_detection":"OFF","window_detection_params":{"minutes":10,"temperature":5},"window_open":false,"workdays":[{"hour":6,"minute":30,"temperature":27},{"hour":8,"minute":0,"temperature":16},{"hour":11,"minute":30,"temperature":16},{"hour":12,"minute":30,"temperature":16},{"hour":16,"minute":30,"temperature":22},{"hour":22,"minute":0,"temperature":16}],"workdays_schedule":"06:30/27°C 08:00/16°C 11:30/16°C 12:30/16°C 16:30/22°C 22:00/16°C"}'

```

didier-guerin85, 2023-01-14:
Hello,
I noticed the same problem and I modified the composite. I made 2 composites and it works. It's not pretty, I know. and since version 127.1 the fields are no longer filled in by default. It was convenient....
[image]

ericvb, 2023-01-14:
#15306 
I think this issue is about the same problem...

leblitzdick, 2023-01-23:
> Hello, I noticed the same problem and I modified the composite. I made 2 composites and it works. It's not pretty, I know. and since version 127.1 the fields are no longer filled in by default. It was convenient.... [image]

Hi,
i have the same problem with my TS0601, can you explain how you made these "2 composites"? My zigbee skills are very low an i only want to solve this problem, thankx

didier-guerin85, 2023-01-23:
**/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices $ sudo nano tuya.js** 

_find:_
zigbeeModel: ['kud7u2l'],
        fingerprint: [
            {modelID: 'TS0601', manufacturerName: '_TZE200_ckud7u2l'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_ywdxldoj'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_cwnjrr72'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_pvvbommb'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_9sfg7gm0'}, // HomeCloud
            {modelID: 'TS0601', manufacturerName: '_TZE200_2atgpdho'}, // HY367
            {modelID: 'TS0601', manufacturerName: '_TZE200_cpmgn2cf'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_4eeyebrt'}, // Immax 07732B
        ],
        model: 'TS0601_thermostat',
        vendor: 'TuYa',
        description: 'Radiator valve with thermostat',

and replace EXPOSE by this

exposes: [
            e.child_lock(), e.window_detection(),
            exposes.binary('window_open', ea.STATE).withDescription('Window open?'),
            e.battery_low(), e.valve_detection(), e.position(),
            exposes.climate().withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withLocalTemperature(ea.STATE).withSystemMode(['heat', 'auto', 'off'], ea.STATE_SET,
                    'Mode of this device, in the `heat` mode the TS0601 will remain continuously heating, i.e. it does not regulate ' +
                    'to the desired temperature. If you want TRV to properly regulate the temperature you need to use mode `auto` ' +
                    'instead setting the desired temperature.')
                .withLocalTemperatureCalibration(-9, 9, 1, ea.STATE_SET)
                .withPreset(['schedule', 'manual', 'boost', 'complex', 'comfort', 'eco'])
                .withRunningState(['idle', 'heat'], ea.STATE),
            e.auto_lock(), e.away_mode(), e.away_preset_days(), e.boost_time(), e.comfort_temperature(), e.eco_temperature(), e.force(),
            e.max_temperature(), e.min_temperature(), e.away_preset_temperature(),
            **exposes.composite('programming_mode').withDescription('Schedule MODE ⏱ - In this mode, ' +
                    'the device executes a preset week programming temperature time and temperature.')
                .withFeature(e.week())
                .withFeature(exposes.text('workdays_schedule', ea.STATE_SET)),
            exposes.composite('programming_mode').withDescription('Schedule MODE ⏱ - In this mode, ' +
                    'the device executes a preset week programming temperature time and temperature.')
                .withFeature(e.week())
                .withFeature(exposes.text('holidays_schedule', ea.STATE_SET))**
        ],
 

and reload MQTT

 It's not pretty..... but .....it's ok

leblitzdick, 2023-01-23:
thank you! 
so my zigbee2mqtt runs in a docker container by homeassistent and it's quite hard for me to manipulate that image without destroying everything...
So i will wait for someone smarter when me to fix this :-)

leblitzdick, 2023-01-23:
So if it's helping someone. I get this error message in homeassistent then i try to set the time/temperature pairs. Only in the v1.29.2-1 edge version:

`No converter available for 'programming_mode' ({"holidays_schedule":"08:30/22°C 10:30/18°C 13:00/18°C 18:00/18°C 21:00/18°C 23:00/16°C","week":"5+2","workdays_schedule":"06:30/22°C 08:30/18°C 13:00/18°C 18:00/18°C 21:00/18°C 23:00/16°C"})`

the stable version reports no error an do also set no  "workdays_schedule" only "holidays_schedule"

ericvb, 2023-01-24:
There is some regression apparently in the latest version from the DEV branch.
If this goes unnoticed into the next release, we have a problem for this device and maybe other devices also, if the problem resides in basic code.
@Koenkk, what can we do to help?

didier-guerin85, 2023-01-24:
hello, 

I agree, do not make changes in the sources.

Here is the list of problems.

* A display problem for the Week/Holiday/workday fields. It works until version 1.27.0 and not in 1.27.1 at now. (info present in "STATUS").
* A problem updating Week/Holiday/workday instructions at the same time. Only 2 infos are correctly done. only 2 at a time. That's why I changed the COMPOSITE and it works like on SmartLife.
* A problem updating the current temperature. So I set up a cron to send an order to anyone and it responds by sending all the info it has at its disposal. (a bit bad... I know :-)).
* And one last remark. I replaced the thermal probe with an external probe with 3 meters of wire and I have the temperature of the room and not that of the radiator. And it works well. But the regulation is bizarre. I ask for a setpoint of 16 and I always have a regulation that runs between 17 and 17.5. But when we know it, we deal with it.
In general, I now have a thermostat control system ---> radiator valve. It works wonderfully but it has been working for 15 years and I am looking for an equivalent and reliable solution that does not consume too many batteries for its future replacement because this equipment no longer exists (FHT80B and FHEM).
It's a bit long and I speak the language of Molière so I use a google translator :-)
Good day to all. DG

ericvb, 2023-01-25:
Another issue #16328 related to the same problem?

mrwogu, 2023-02-03:
Same here. `No converter available for 'programming_mode'`

leblitzdick, 2023-02-09:
> * A display problem for the Week/Holiday/workday fields. It works until version 1.27.0 and not in 1.27.1 at now. (info present in "STATUS").

I tested it with a 1.27.0 proxy version and i could not reproduce it to work with the current (2023.2.3) homeassistent version.

pakagi, 2023-03-12:
I would say this is still valid, I still get the "No converter available".

remi00, 2023-03-15:
It looks like it's 100% reproducibility with TS0601 - confirming I got the same behavior with latest version. Basically, it prevents from the possibility of editting the schedule. Anybody who decided to use HA/Zigbe2mqtt instead of crappy native Tuya APP will not be able to customize the schedule on the TRV. Kinda severe issue from the functionality standpoint.

neovinter, 2023-03-20:
> It looks like it's 100% reproducibility with TS0601 - confirming I got the same behavior with latest Zigbee2mqtt version. Basically, it prevents from the possibility of editting the schedule. Anybody who decided to use HA/Zigbee2mqtt instead of crappy native Tuya APP will not be able to customize the schedule on the TRV. Kinda severe issue from the functionality standpoint.

Yes, that's right. So I still have to use version 1.28.2-1 Schedule doesn't work in later versions.

geigervibe, 2023-03-22:
I reported a few more problems here:
https://github.com/Koenkk/zigbee2mqtt/issues/15903#issue-1516836975 (first post).
The other problems I reported there are not z2m related...

pakagi, 2023-04-10:
This bug seems to be still present in 1.30.3 :-(

mrwogu, 2023-05-11:
Still present in 1.30.4

mrwogu, 2023-06-11:
Still present in 1.31.2-1

ericvb, 2023-06-12:
@mrwogu, thank you for testing this each time. An annoying bug...

@Koenkk, is this unwanted behavior globally for the TS0601 type thermostat or just for a particular brand?
The problem appears not be in basic code, I don't see other issues describing the same problem, so it must be more specific to the type of hardware.
Can we do something to help to debug this issue?

closed 2023-07-21 by github-actions[bot]

ericvb, 2023-08-07:
The problem is still present in 1.32.2-1

kornelremete, 2023-09-26:
same at me

ericvb, 2023-09-26:
The winter is coming again for the northern earth side, so I will take the time to test this again :-)
Probably I will recreate a new ticket linked to this one, because I can't open this one who was closed by this da... bot.
Maybe there is already a new one based on this one?

JurajSK, 2023-10-17:
I was at the same point when I realize it needs different topic, different from the other commands
correct is "zigbee2mqtt/friendly_name/set/schedule"

then I use service in HA, MQTT publish

yaml code:

service: mqtt.publish
data:
  topic: zigbee2mqtt/TRV spalna/set/schedule
  payload: >-
    {     "workdays":[         {"hour":6,"minute":0,"temperature":15},        
    {"hour":8,"minute":0,"temperature":15},        
    {"hour":17,"minute":30,"temperature":17},        
    {"hour":20,"minute":30,"temperature":20},        
    {"hour":22,"minute":0,"temperature":20},        
    {"hour":23,"minute":30,"temperature":20}     ] }

note: you can change time and temperature in each time slot, but leave all 6, don´t delete even one

ericvb, 2023-10-28:
probably here  : https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts (line 2544)
and here : https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/legacy.ts (line 7563)
but have no experience here...

ericvb, 2024-01-02:
A follow up.
After installing the new release of the Zigbee2MQTT 1.35.0-1, I see a correct representation of the schedules.
Probably due to this correction: https://github.com/nurikk/zigbee2mqtt-frontend/pull/1851

But now I don't see the Week options, they are gone : 5+2, 6+1 and 7 options
Before:
[image]
After installing latest version:
[image]

stockno, 2024-01-06:
After updating zigbee2mqtt from version 1.24.0.1 to version 1.35.0.1. the setting of the TUYA TS0601 thermostatic head has changed. Current temperature, desired temperature and % valve opening are not shown.
Do you have any solution? Thank you
[image]
[image]

ericvb, 2024-01-08:
@stockno, are you running the latest HA version?
The printscreens are the old HVAC cards.

I have installed the latest HA version and then you get this:
[image]

stockno, 2024-01-16:
I had everything up to date. But I tried again, but this time in a different order. I updated HA first and then zigbee2mqtt. And now everything works for me. The problem was in the order of the updates