Koenkk/zigbee2mqtt #16162: Energy not updating OWON PC321

tpawork, 2023-01-14:
### What happened?

After upgrading from 1.28.2 to 1.29.1 (and 1.29.2) the energy of the OWON PC321 is not updated any more. 
energy_l1, energy_l2 and energy_l3 are still updated.

It seems like after a restart of zigbee2mqtt it's updated once.

### What did you expect to happen?

energy is updated 

### How to reproduce it (minimal and precise)

Upgrade to 1.29.1

### Zigbee2MQTT version

1.29.1 and 1.29.2

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0

Koenkk, 2023-01-15:
energy is now provided per phase instead of just one.

tpawork, 2023-01-16:
Hi @Koenkk, the energy per phase were already existing before. Does this mean the total energy is now removed?
[image]

Koenkk, 2023-01-16:
@Phil-ibert  should `energy` still be updated?

lermit4, 2023-02-11:
I have this problem with version 1.30.1-1. the phase Energy values per phase are updated but not the total Energy.

[image]
[image]
[image]

HakunaMatmata, 2023-02-26:
Hello,
Same problem here with another energy meter :
Casaia CCB432 : https://github.com/Koenkk/zigbee2mqtt/issues/16291

Koenkk, 2023-03-07:
The device does not report the total energy, you have to create a sensor yourself in HA to do this computation (sum of all energy per phase)

tpawork, 2023-03-07:
So this was something that has been done by z2m before 1.29.1 but has been removed now and has to be done manually in HA?

kalemena, 2023-03-08:
Hi,
I am having similar problem with OWON PC321.
Starting with 1.28.3, it seems Total energy only reports once upon zigbee2mqtt start.
1.28.2 works as expected.

Koenkk, 2023-03-09:
In 1.28.2 the energy was also reported per phase (no sum). Link to code used in 1.28.2: https://github.com/Koenkk/zigbee-herdsman-converters/blob/6deaf294bc8fc99f8b39cf3989fef7506e1590ee/devices/owon.js#L210

kalemena, 2023-03-10:
OK, this means what I thought a version issue might be a use case issue.
Let me summarise scenario in few more details.

Pre-requisits:
- using docker version of zigbee2mqtt
- adapter is Conbee II
- registered OWON PC321 through Zigbee2MQTT UI

Scenario:
- starting zigbee2mqtt with docker version 1.28.2
- connecting to UI, on the exposed panel of PC321, the top level information "energy" is getting updated

Here is what Zigbee2MQTT expose panel look like:

[image]

- stoping the docker container
- starting any higher version > 1.28.2 of same container
- connecting to UI, on the exposed panel of PC321, the top level information "energy" is updated only ONCE after zigbee2mqtt starts

- going back to 1.28.2
- the energy updates again at regular pace

Note: MQTT channel return both total "energy" and "power" fields. And both are no longer updated with version higher than 1.28.2

I hope this helps.
Let me know if there is something unusual done in scenario, or if there are specific tests I can perform to troubleshoot.

Koenkk, 2023-03-11:
- I see the issue now, could you provide the debug logging when running z2m 1.30.2 when you would expect `energy` updates? See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

- do you have any options set for this device in the `configuration.yaml`? (like `debounce`)

HakunaMatmata, 2023-03-11:
Hello both, I understand that the problem is the same with owon and casaia device.
@Koenkk we were also discussing this issue here https://github.com/Koenkk/zigbee2mqtt/issues/16291

If it can help you to understand both.
I also have the conbee 2 as an adapter.

kalemena, 2023-03-11:
Hi @Koenkk 
I have no specific options for this device.

See debug logs attached for when device sends energy.
[OWON-PC321-debug.log](https://github.com/Koenkk/zigbee2mqtt/files/10948550/OWON-PC321-debug.log)

The line below shows what seems the expected energy total value (few time after the restart):
"currentSummDelivered":[0,2964566],

But right after the parsing, MQTT publish still show an older value (from z2m startup I guess):
"energy":2964.52,

As a side note, 
I now see I have similar power/energy issue as @HakunaMatmata, in my case with devices "Nous AZ1 (newer version)".

Koenkk, 2023-03-12:
@kalemena 

Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    PC321_metering: {
        cluster: 'seMetering',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const payload = fz.metering.convert(model, msg, publish, options, meta);
            if (msg.data.hasOwnProperty('owonL1Energy')) {
                payload.energy_l1 = msg.data['owonL1Energy'][1] / 1000.0;
            }
            if (msg.data.hasOwnProperty('owonL2Energy')) {
                payload.energy_l2 = msg.data['owonL2Energy'][1] / 1000.0;
            }
            if (msg.data.hasOwnProperty('owonL3Energy')) {
                payload.energy_l3 = msg.data['owonL3Energy'][1] / 1000.0;
            }
            if (msg.data.hasOwnProperty('owonL1ReactiveEnergy')) {
                payload.reactive_energy_l1 = msg.data['owonL1ReactiveEnergy'][1] / 1000.0;
            }
            if (msg.data.hasOwnProperty('owonL2ReactiveEnergy')) {
                payload.reactive_energy_l2 = msg.data['owonL2ReactiveEnergy'][1] / 1000.0;
            }
            if (msg.data.hasOwnProperty('owonL3ReactiveEnergy')) {
                payload.reactive_energy_l3 = msg.data['owonL3ReactiveEnergy'][1] / 1000.0;
            }
            if (msg.data.hasOwnProperty('owonL1PhasePower')) {
                payload.power_l1 = msg.data['owonL1PhasePower'];
            }
            if (msg.data.hasOwnProperty('owonL2PhasePower')) {
                payload.power_l2 = msg.data['owonL2PhasePower'];
            }
            if (msg.data.hasOwnProperty('owonL3PhasePower')) {
                payload.power_l3 = msg.data['owonL3PhasePower'];
            }
            if (msg.data.hasOwnProperty('owonL1PhaseReactivePower')) {
                payload.reactive_power_l1 = msg.data['owonL1PhaseReactivePower'];
            }
            if (msg.data.hasOwnProperty('owonL2PhaseReactivePower')) {
                payload.reactive_power_l2 = msg.data['owonL2PhaseReactivePower'];
            }
            if (msg.data.hasOwnProperty('owonL3PhaseReactivePower')) {
                payload.reactive_power_l3 = msg.data['owonL3PhaseReactivePower'];
            }
            if (msg.data.hasOwnProperty('owonL1PhaseVoltage')) {
                payload.voltage_l1 = msg.data['owonL1PhaseVoltage'] / 10.0;
            }
            if (msg.data.hasOwnProperty('owonL2PhaseVoltage')) {
                payload.voltage_l2 = msg.data['owonL2PhaseVoltage'] / 10.0;
            }
            if (msg.data.hasOwnProperty('owonL3PhaseVoltage')) {
                payload.voltage_l3 = msg.data['owonL3PhaseVoltage'] / 10.0;
            }
            if (msg.data.hasOwnProperty('owonL1PhaseCurrent')) {
                payload.current_l1 = msg.data['owonL1PhaseCurrent'] / 1000.0;
[... 41 lines cut ...]
        exposes.numeric('energy_l1', ea.STATE).withUnit('kWh').withDescription('Phase 1 energy'),
        exposes.numeric('energy_l2', ea.STATE).withUnit('kWh').withDescription('Phase 2 energy'),
        exposes.numeric('energy_l3', ea.STATE).withUnit('kWh').withDescription('Phase 3 energy'),
        exposes.numeric('reactive_energy_l1', ea.STATE).withUnit('kVArh').withDescription('Phase 1 reactive energy'),
        exposes.numeric('reactive_energy_l2', ea.STATE).withUnit('kVArh').withDescription('Phase 2 reactive energy'),
        exposes.numeric('reactive_energy_l3', ea.STATE).withUnit('kVArh').withDescription('Phase 3 reactive energy'),
        exposes.numeric('power_l1', ea.STATE).withUnit('W').withDescription('Phase 1 power'),
        exposes.numeric('power_l2', ea.STATE).withUnit('W').withDescription('Phase 2 power'),
        exposes.numeric('power_l3', ea.STATE).withUnit('W').withDescription('Phase 3 power'),
        exposes.numeric('reactive_power_l1', ea.STATE).withUnit('VAr').withDescription('Phase 1 reactive power'),
        exposes.numeric('reactive_power_l2', ea.STATE).withUnit('VAr').withDescription('Phase 2 reactive power'),
        exposes.numeric('reactive_power_l3', ea.STATE).withUnit('VAr').withDescription('Phase 3 reactive power'),
    ],
};

module.exports = definition;

```
- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

kalemena, 2023-03-12:
@Koenkk 

Below, the "currentSummDelivered" line seems correct, but then there is parsing issue, and no MQTT publish occurs.

```
Zigbee2MQTT:debug 2023-03-12 15:59:26: Received Zigbee message from 'C01~Garage~PowerMeter', type 'attributeReport', cluster 'seMetering', data '{"12547":170,"16643":[0,127806],"8451":39,"currentSummDelivered":[0,2999378],"instantaneousDemand":7,"status":0}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2023-03-12 15:59:27: Received Zigbee message from 'C01~Garage~PowerMeter', type 'attributeReport', cluster 'seMetering', data '{"16644":0,"owonL1Energy":[0,1002361],"owonL1PhaseCurrent":0,"owonL1PhasePower":0,"owonL1PhaseReactivePower":0,"owonL1PhaseVoltage":2346,"owonL1ReactiveEnergy":[0,198]}' from endpoint 1 with groupID null
Zigbee2MQTT:error 2023-03-12 15:59:27: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'energy_l1')}
Zigbee2MQTT:debug 2023-03-12 15:59:27: TypeError: Cannot set properties of undefined (setting 'energy_l1')
    at Object.convert (/app/data/extension/externally-loaded.js:20:35)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:143:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)
    at DeconzAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)
    at DeconzAdapter.emit (node:events:513:28)
```

Koenkk, 2023-03-13:
Updated https://github.com/Koenkk/zigbee2mqtt/issues/16162#issuecomment-1465178777

kalemena, 2023-03-13:
@Koenkk 

With proposal, there are no exception anymore, but no better luck at publishing MQTT upon "Received Zigbee message".
The metering.convert seems returns empty.

Tweaking a bit with below helps for getting our energy updates, but this defeats purpose of convert code:

```
...
convert: (model, msg, publish, options, meta) => {
            const payload = {...fz.metering.convert(model, msg, publish, options, meta)};
            if (msg.data.hasOwnProperty('currentSummDelivered')) {
                payload.energy = msg.data['currentSummDelivered'][1] / 1000.0;
            }
            if (msg.data.hasOwnProperty('owonL1Energy')) {
                payload.energy_l1 = msg.data['owonL1Energy'][1] / 1000.0;
            }
            ...
```
The hasAlreadyProcessedMessage at start of metering.convert seems true on my side, hence returning and preventing to read values.
But is that because of custom external converter or would that be same using owon.js ?

 ```
convert: (model, msg, publish, options, meta) => {
            if (utils.hasAlreadyProcessedMessage(msg, model)) return;
            const payload = {};
```

Koenkk, 2023-03-14:
Updated https://github.com/Koenkk/zigbee2mqtt/issues/16162#issuecomment-1465178777 again.

kalemena, 2023-03-14:
Perfect. 
Both energy and power are now published as expected.
Tell me if I can help testing to bring this in a next release.
Thanks

Koenkk, 2023-03-15:
Pushed the fix for both OWON PC321 and CASAIA CCB432.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-03-15 by Koenkk