Koenkk/zigbee2mqtt #16166: TS110E can't detect physical clicks after update

marcin-adamczewski, 2023-01-14:
### What happened?

Automations for TS110E (1-gang) devices stopped working. It happened after updating to 1.29.1-1 from a much older version.
As I can see it is because the device exposes now two different states:
"state_1" - which is "on" when tapped on a physical button
"state" - which is "on" when changed the state programmatically (through UI/app). When changed programmatically it also makes the "state_1" on. 

So now it's hard to create simple HomeAssitant automation as there are two states and only one is recognized as state - "state" (programmatically state changes only). In other words, there is no easy possibility to detect physical clicks.

When setting a trigger on entity state there is a "State 1" attribute available but no "State" attribute. And it turns out that "State 1" trigger is actually reacting to changing "state" (the programmatic attribute I mentioned). So the naming seems to be wrong here.
There is also no "brightness" attribute when creating automation. Screenshots below.

[image]

[image]

Some logs:
1. When clicked physical button:
```
 2023-01-14 19:39:04Received Zigbee message from 'Light Bathroom', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
2023-01-14 19:39:04MQTT publish: topic 'zigbee2mqtt/Light Bathroom', payload '{"brightness":120,"linkquality":69,"power_on_behavior":null,"state":"OFF","state_1":"ON","switch_type":null}'
```

2. When tapped in the app:
```
2023-01-14 19:41:36Received MQTT message on 'zigbee2mqtt/Light Bathroom/set' with data '{"state":"ON"}'
2023-01-14 19:41:36Publishing 'set' 'state' to 'Light Bathroom'
2023-01-14 19:41:36Received Zigbee message from 'Light Bathroom', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
2023-01-14 19:41:36MQTT publish: topic 'zigbee2mqtt/Light Bathroom', payload '{"brightness":120,"linkquality":116,"power_on_behavior":null,"state":"OFF","state_1":"ON","switch_type":null}'
2023-01-14 19:41:36MQTT publish: topic 'zigbee2mqtt/Light Bathroom', payload '{"brightness":120,"linkquality":116,"power_on_behavior":null,"state":"ON","state_1":"ON","switch_type":null}'
```

### Zigbee2MQTT version

1.29.1-1

### Adapter firmware version

20220219

### Adapter

Sonoff Dongle P

Koenkk, 2023-01-15:
Can you provide the data/database.db entry of this device?

marcin-adamczewski, 2023-01-15:
Here you go
```
{"id":17,"type":"Router","ieeeAddr":"0xa4c1383b49d8392e","nwkAddr":8153,"manufId":4417,"manufName":"_TZ3210_zxbtub8r","powerSource":"Mains (single phase)","modelId":"TS110E","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"de
vId":257,"inClusterList":[4,5,3,6,8,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"E\u001es)\u0013Q\u001es)\u0016\u001f\u0001\u0000\u0000\u0012\u0018\u0000\u0000\u0000\u0012","65506":37,"65508":0,"655
34":0,"modelId":"TS110E","manufacturerName":"_TZ3210_zxbtub8r","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"61440":0,"onOff":0,"onTime":0,"moesStartUpOnOff":2
}},"genLevelCtrl":{"attributes":{"61440":797,"64514":0,"64515":11,"64516":1000,"currentLevel":120}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0014da382b","endpointID":1},{"cluster":8,"type":"endpoint","devi
ceIeeeAddress":"0x00124b0014da382b","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{
}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":756186969},"lastSeen":1673813767452,"defaultSendRequestWhen":"immediate"}
```

Koenkk, 2023-01-16:
Could it be that you are using an external converter? (check your `configuration.yaml` for a `external_converters` section)

marcin-adamczewski, 2023-01-16:
Nope, the array is empty.

Koenkk, 2023-01-16:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS110E', ['_TZ3210_zxbtub8r', '_TZ3210_k1msuvg6']),
    model: 'TS110E_1gang_1',
    vendor: 'TuYa',
    description: '1 channel dimmer',
    fromZigbee: extend.light_onoff_brightness({disablePowerOnBehavior: true, disableMoveStep: true, disableTransition: true})
        .fromZigbee.concat([tuya.fz.power_on_behavior]),
    toZigbee: extend.light_onoff_brightness({disablePowerOnBehavior: true, disableMoveStep: true, disableTransition: true})
        .toZigbee.concat([tuya.tz.power_on_behavior]),
    exposes: [e.light_brightness(), e.power_on_behavior(), tuya.exposes.switchType()],
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

marcin-adamczewski, 2023-01-16:
Awesome! It's working fine now. Both physical and programmatic clicks are detected and stored in the "state" field. Looks like "state_1" is not being changed at all and is always "OFF". 
I'm just wondering...previously it was possible to distinguish physical from programmatic clicks. Maybe it would be useful for some people to have it?

There is one more thing. The brightness is still not visible in the HomeAssitant automations creator. I've noticed that this is the case for 2-gang switch as well. I've never used brightness when reading the state in HA automations so maybe it was always like this. Just letting you know, not something that I need. It looks like this now:
[image]

Koenkk, 2023-01-17:
This code should be the same as in z2m 1.29.2 so I'm confused. Can you try with the latest dev and without the external converter? (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

marcin-adamczewski, 2023-01-17:
Ok, so this is what I did.
1. Removed external converter
2. Updated to 1.29.2, restarted z2m and still didn't work
3. Stopped z2m add-on
4. Installed and started the Edge add-on
**5. Works fine on the Edge version.**
6. Disabled Edge and started z2m 1.29.2 - doesn't work

Koenkk, 2023-01-18:
I see, then this will be fixed in the next z2m version (1 February).

closed 2023-01-18 by Koenkk

enricodeleo, 2023-04-18:
Maybe this issue is related to mine: on all the devices I have, the physical button stops working after a while until I turn on the light again via software (e.g. Alexa command, virtual button, brightness adjustment etc). If I use the physical button right after lights are turned on it works as expected (and switches off too). I cannot get why this is happening or how long it takes to stop working again (let's say when I come from work no lights are responsive to physical buttons). 

State looks the same on all devices and this is an example:
```
{
    "brightness": 0,
    "linkquality": 56,
    "max_brightness": 255,
    "min_brightness": 10,
    "power_on_behavior": "previous",
    "state": "OFF",
    "switch_type": "momentary"
}
```

Any ideas? I'm using the Home Assistant add on version 1.30.3-1