Koenkk/zigbee2mqtt #16167: Devices falling out from network after update to 1.9.2

kizio1000, 2023-01-14:
### What happened?

.After morning update to 1.9.2 devices started to loosing connection to network.  Both routers and battery devices. Since update at the morning four devices left network (went offline). Three aquara movement sensors and one 230V relay switch.  

### What did you expect to happen?

That devices will stay connected as they were before update

### Zigbee2MQTT version

1.9.2-1

### Adapter firmware version

6.10.3.0 build 29

### Adapter

ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Debug log

Nothing in logs, devices becoming offline

tristangrichard, 2023-01-14:
I also have Sonoff SNZB-0X devices dropping out of the network again

Electronlibre2012, 2023-01-17:
hello, i lost 2 aqara pir sensors, they dont detect movement anymore...i repair and now no battery, voltage indication and they go  unavailable every ten minutes...

i revert will try to go back to 1.9.1....

closed 2023-02-24 by github-actions[bot]

tristangrichard, 2023-04-09:
Changing the battery of the device and re-pair helped