Koenkk/zigbee2mqtt #16170: [New device support]: Legrand WNACB40 (4-Button Scene Controller)

brantmacga, 2023-01-14:
### Link

https://www.legrand.us/wiring-devices/adorne-collection/switches/adorne-wireless-smart-scene-switch-with-netatmo-white/p/wnacb40w1

### Database entry

0x646266fffe62c80a

### Comments

I have the device paired, but I'm unable to figure out how to get access to the 4 scene buttons. I'd like to be able to use them to trigger events/scenes in Home Assistant. I also don't see an option to bind any of the 4 actions. 

This is one of the 4-button scene controllers for the north american market. The other version is shown here --> https://www.legrand.us/wiring-devices/radiant-collection/switches/radiant-wireless-smart-scene-controller-with-netatmo-white/p/wnrcb40wh

### External converter

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const ota = require('zigbee-herdsman-converters/lib/ota');

const readInitialBatteryState = async (type, data, device) => {
    if (['deviceAnnounce'].includes(type)) {
        const endpoint = device.getEndpoint(1);
        const options = {manufacturerCode: 0x1021, disableDefaultResponse: true};
        await endpoint.read('genPowerCfg', ['batteryVoltage'], options);
    }
};

const tzLocal = {
    auto_mode: {
        key: ['auto_mode'],
        convertSet: async (entity, key, value, meta) => {
            const mode = {'off': 0x00, 'auto': 0x02, 'on_override': 0x03};
            const payload = {data: Buffer.from([mode[value]])};
            await entity.command('manuSpecificLegrandDevices3', 'command0', payload);
            return {state: {'auto_mode': value}};
        },
    },
};

module.exports = [
    {
        zigbeeModel: ['Wireless Scenes Command'],
        model: 'WNACB40',
        vendor: 'Legrand',
        // led blink RED when battery is low
        description: 'Wireless Smart Scene Controller',
        fromZigbee: [fz.identify, fz.on_off, fz.command_toggle],
        exposes: [e.battery(), e.action(['press_1', 'press_2', 'press_3', 'press_4'])],
        toZigbee: [],
        meta: {battery: {voltageToPercentage: '3V_2500'}, publishDuplicateTransaction: true},
        onEvent: readInitialBatteryState,
        configure: async (device, coordinatorEndpoint, logger) => {
            await extend.configure(device, coordinatorEndpoint, logger);
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genIdentify']);

        },
    },
];
```

brantmacga, 2023-01-17:
I am using a different converter now, still not getting the response I need. 

I need to be able to see each button in HA when trying to use it to toggle scenes. Right now when I try to configure it in HA automations, I have "when battery level changes" and "default action". 

This is my converter.....

### External converter

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const ota = require('zigbee-herdsman-converters/lib/ota');

const readInitialBatteryState = async (type, data, device) => {
    if (['deviceAnnounce'].includes(type)) {
        const endpoint = device.getEndpoint(1);
        const options = {manufacturerCode: 0x1021, disableDefaultResponse: true};
        await endpoint.read('genPowerCfg', ['batteryVoltage'], options);
    }
};

const tzLocal = {
    auto_mode: {
        key: ['auto_mode'],
        convertSet: async (entity, key, value, meta) => {
            const mode = {'off': 0x00, 'auto': 0x02, 'on_override': 0x03};
            const payload = {data: Buffer.from([mode[value]])};
            await entity.command('manuSpecificLegrandDevices3', 'command0', payload);
            return {state: {'auto_mode': value}};
        },
    },
};

module.exports = [
    {
        zigbeeModel: ['Wireless Scenes Command'],
        model: 'WNACB4, WNRCB4',
        vendor: 'Legrand',
        description: 'Wireless Smart Scene Controller',
        fromZigbee: [fz.legrand_scenes, fz.battery, fz.ignore_poll_ctrl, fz.legrand_master_switch_center],
        toZigbee: [],
        exposes: [e.battery(), e.action(['Scene_1', 'Scene_2', 'Scene_3', 'Scene_4', 'center'])],
        meta: {battery: {voltageToPercentage: '3V_2500'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genIdentify', 'genPowerCfg']);
        },
     },
];
' ' '

jiquem, 2023-02-01:
You could have a look to https://github.com/Koenkk/zigbee-herdsman-converters/pull/5359

brantmacga, 2023-02-01:
> You could have a look to [Koenkk/zigbee-herdsman-converters#5359](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5359)

thank you. Is your device a Green Power device?

jiquem, 2023-02-01:
It's not a "green power" device : it has a battery

brantmacga, 2023-02-01:
> It's not a "green power" device : it has a battery

https://zigbee.blakadder.com/Legrand_ZLGP15.html

The “GP” is for green power. Is this the device you have ?

brantmacga, 2023-02-02:
I've added fz.command_recall. 

Now when I try to create an automation, I have "recall_1 action" as a trigger. This is the only trigger, and it works for all 4 buttons. Now trying to get it where I have a trigger option for each of the four buttons so they can control different devices/scenese. 

' ' '
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const ota = require('zigbee-herdsman-converters/lib/ota');

const readInitialBatteryState = async (type, data, device) => {
    if (['deviceAnnounce'].includes(type)) {

    }
};

const tzLocal = {
    auto_mode: {
        key: ['auto_mode'],
        convertSet: async (entity, key, value, meta) => {
            const mode = {'off': 0x00, 'auto': 0x02, 'on_override': 0x03};
            const payload = {data: Buffer.from([mode[value]])};
            await entity.command('manuSpecificLegrandDevices3', 'command0', payload);
            return {state: {'auto_mode': value}};
        },
    },
};

module.exports = [
    {
        zigbeeModel: ['Wireless Scenes Command'],
        model: 'WNACB4, WNRCB4',
        vendor: 'Legrand',
        description: 'Wireless Smart Scene Controller',
        fromZigbee: [fz.legrand_scenes, fz.battery, fz.ignore_poll_ctrl, fz.legrand_master_switch_center, fz.command_recall],
        toZigbee: [],
        exposes: [e.battery(), e.action(['Scene_1', 'Scene_2', 'Scene_3', 'Scene_4', 'center'])],
        meta: {battery: {voltageToPercentage: '3V_2500'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genIdentify', 'genPowerCfg']);
        },
     },
];
' ' '

brantmacga, 2023-02-02:
Tried another converter setup in attempt to have access to all four buttons.... Still only have the one global "recall_1 action"

### External converter

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const ota = require('zigbee-herdsman-converters/lib/ota');

const readInitialBatteryState = async (type, data, device) => {
    if (['deviceAnnounce'].includes(type)) {

    }
};

const tzLocal = {
    auto_mode: {
        key: ['auto_mode'],
        convertSet: async (entity, key, value, meta) => {
            const mode = {'off': 0x00, 'auto': 0x02, 'on_override': 0x03};
            const payload = {data: Buffer.from([mode[value]])};
            await entity.command('manuSpecificLegrandDevices3', 'command0', payload);
            return {state: {'auto_mode': value}};
        },
    },
};

module.exports = [
    {
        zigbeeModel: ['Wireless Scenes Command'],
        model: 'WNACB4, WNRCB4',
        vendor: 'Legrand',
        description: 'Wireless Smart Scene Controller',
        fromZigbee: [fz.legrand_scenes, fz.battery, fz.ignore_poll_ctrl, fz.legrand_master_switch_center, fz.command_Scene1_recall, fz.command_Scene2_recall, fz.command_Scene3_recall, fz.command_Scene4_recall],
        toZigbee: [],
        exposes: [e.battery(), e.action(['Scene1', 'Scene2', 'Scene3', 'Scene4', 'center'])],
        meta: {battery: {voltageToPercentage: '3V_2500'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genIdentify', 'genPowerCfg']);
        },
     },
];
```

brantmacga, 2023-02-02:
Last attempt for today; still didn't work. 

### External converter

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const ota = require('zigbee-herdsman-converters/lib/ota');

const readInitialBatteryState = async (type, data, device) => {
    if (['deviceAnnounce'].includes(type)) {

    }
};

const tzLocal = {
    auto_mode: {
        key: ['auto_mode'],
        convertSet: async (entity, key, value, meta) => {
            const mode = {'off': 0x00, 'auto': 0x02, 'on_override': 0x03};
            const payload = {data: Buffer.from([mode[value]])};
            await entity.command('manuSpecificLegrandDevices3', 'command0', payload);
            return {state: {'auto_mode': value}};
        },
    },
};

module.exports = [
    {
        zigbeeModel: ['Wireless Scenes Command'],
        model: 'WNACB4, WNRCB4',
        vendor: 'Legrand',
        description: 'Wireless Smart Scene Controller',
        fromZigbee: [fz.legrand_scenes, fz.battery, fz.ignore_poll_ctrl, fz.legrand_master_switch_center, fz.command_recall],
        toZigbee: [],
        exposes: [e.battery(), e.action(['press_1', 'press_2', 'press_3', 'press_4', 'center'])],
        meta: {battery: {voltageToPercentage: '3V_2500'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genIdentify', 'genPowerCfg']);
        },
     },
];
```

brantmacga, 2023-02-03:
@Koenkk any advice with scene control converters?

Koenkk, 2023-02-04:
@brantmacga could you provide me the debug log when pressing the buttons?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

brantmacga, 2023-02-04:
This is a single press of each of the 4 buttons. Thank you for taking a look at this with me!

2023-02-04 10:48:41Received Zigbee message from '0x646266fffe62c80a', type 'commandRecall', cluster 'genScenes', data '{"groupid":65515,"sceneid":1}' from endpoint 1 with groupID 65515

2023-02-04 10:48:46Received Zigbee message from '0x646266fffe62c80a', type 'commandRecall', cluster 'genScenes', data '{"groupid":65517,"sceneid":1}' from endpoint 1 with groupID 65517

2023-02-04 10:48:48Received Zigbee message from '0x646266fffe62c80a', type 'commandRecall', cluster 'genScenes', data '{"groupid":65516,"sceneid":1}' from endpoint 1 with groupID 65516

2023-02-04 10:48:50Received Zigbee message from '0x646266fffe62c80a', type 'commandRecall', cluster 'genScenes', data '{"groupid":65514,"sceneid":1}' from endpoint 1 with groupID 65514

brantmacga, 2023-02-04:
2023-02-04 11:01:22Received Zigbee message from '0x646266fffe62c80a', type 'commandCheckIn', cluster 'genPollCtrl', data '{}' from endpoint 1 with groupID 0

2023-02-04 11:01:22Received Zigbee message from '0x646266fffe62c80a', type 'commandRecall', cluster 'genScenes', data '{"groupid":65517,"sceneid":1}' from endpoint 1 with groupID 65517

2023-02-04 11:01:24Received Zigbee message from '0x646266fffe62c80a', type 'readResponse', cluster 'genBasic', data '{"appVersion":0}' from endpoint 1 with groupID 0

2023-02-04 11:01:24Received Zigbee message from '0x646266fffe62c80a', type 'readResponse', cluster 'genBasic', data '{"hwVersion":0}' from endpoint 1 with groupID 0

2023-02-04 11:01:25Received Zigbee message from '0x646266fffe62c80a', type 'readResponse', cluster 'genPollCtrl', data '{"checkinInterval":345600}' from endpoint 1 with groupID 0

Koenkk, 2023-02-05:
The `groupid` varies, along with the `action` you will also find an `action_group` in the published payload.

m4sis, 2023-02-08:
I also have a Legrand button with 4 scenes, like you, I get action_1 for the four buttons.
I have no idea how to successfully assign the groupid to an automation

brantmacga, 2023-02-08:
I haven't yet figured out how to implement what Koenkk is telling me. I understand there are different group ID's, I'm just not sure yet how that should be written in a way that HA understands for using in automations.

jiquem, 2023-02-08:
I am like you, on hold on this issue ;(

m4sis, 2023-02-16:
breakthrough ?

brantmacga, 2023-02-17:
> breakthrough ?

no. I haven't yet had time to work on it again.

Hellou1337, 2023-11-21:
hi, i have the same legrand switch and only one button si working as have the guys before me.  Also i have legrand 067774 and also only one action is received. So basically the same problem :) 

Is there any way to distinguish between pushed buttons? 
Thanks in advance for any reply

jiquem, 2023-11-24:
Hello, 

You can proceed like this to trigger the automation. 
For instance, this one will be triggered when the button 1 is used
[image]

Hellou1337, 2023-11-24:
Hi,

thanks for reply - will use.

however for 067775 only one button is sending signals, others just dont show up as any action in Z2M or any logs. I guess some kind of external converter shoud do the trick? The one button that works sends action_group":65516 so that i can work with. But if no message shows up in Z2M what can i do?

jiquem, 2023-11-25:
Try this one

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const ota = require('zigbee-herdsman-converters/lib/ota');

const readInitialBatteryState = async (type, data, device) => {
    if (['deviceAnnounce'].includes(type)) {
        const endpoint = device.getEndpoint(1);
        const options = {manufacturerCode: 0x1021, disableDefaultResponse: true};
        await endpoint.read('genPowerCfg', ['batteryVoltage'], options);
    }
};

const tzLocal = {
    auto_mode: {
        key: ['auto_mode'],
        convertSet: async (entity, key, value, meta) => {
            const mode = {'off': 0x00, 'auto': 0x02, 'on_override': 0x03};
            const payload = {data: Buffer.from([mode[value]])};
            await entity.command('manuSpecificLegrandDevices3', 'command0', payload);
            return {state: {'auto_mode': value}};
        },
    },
};

const definition = {
//        zigbeeModel: [' Wireless Scenes Command'],
        fingerprint: [{modelID: ' Wireless Scenes Command\u0000\u0000\u0000\u0000\u0000\u0000\u0000'}],

	//0x000474000092b308
	//        fingerprint: [{modelID: 'GreenPower_254', ieeeAddr: /^0x00000000005.....$/}],
	//Wireless Scenes Command
		//zigbeeModel: [' Wireless Scenes Command\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'+
            //'\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'],
        model: '067755',
        vendor: 'Legrand',
        description: 'Wireless Scenes Command',
        fromZigbee: [fz.identify, fz.battery, fz.command_recall],
        exposes: [e.battery(), e.action(['identify', 'recall_1_1'])],
        toZigbee: [],
        meta: {multiEndpoint: true, battery: {voltageToPercentage: '3V_2500'}, publishDuplicateTransaction: true},
        onEvent: readInitialBatteryState,
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genOnOff', 'genLevelCtrl']);
        },

};

module.exports = definition;
```

9611919, 2023-12-23:
Sorry for dummy question...)
Where did you out the code?

jiquem, 2023-12-24:
Have a look to https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-adding-your-device

jiquem, 2023-12-25:
It’s normal : z2m doesn’t allow firmware upgrade for this device

9611919, 2023-12-26:
> It’s normal : z2m doesn’t allow firmware upgrade for this device

Simply IMHO. Can it happen that I use Bticino and update is for Legrand?

jiquem, 2023-12-26:
No

9611919, 2023-12-26:
> No

https://github.com/Koenkk/zigbee-OTA/issues/328

I did it!

jiquem, 2023-12-27:
You did it with a powered device.

9611919, 2023-12-27:
> You did it with a powered device.

Yes. With bticino k4411c