Koenkk/zigbee2mqtt #16173: Switching from conbee2 to sonoff stick

cryd-s, 2023-01-15:
### What happened?

Want to switch from a working conbee 2 config to a sonoff stick.

### What did you expect to happen?

Want to switch from a working conbee 2 config to a sonoff stick.

### Zigbee2MQTT version

1.29.2-1

### Adapter firmware version

cannot enter the frontend

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus,TI CC2652P + CP2102(N

### Debug log

info  2023-01-15 09:27:42: Logging to console and directory: '/config/zigbee2mqtt-test/log/2023-01-15.09-27-42' filename: log.txt
info  2023-01-15 09:27:42: Starting Zigbee2MQTT version 1.29.2 (commit #unknown)
info  2023-01-15 09:27:42: Starting zigbee-herdsman (0.14.83-hotfix.0)
error 2023-01-15 09:28:47: Error while starting zigbee-herdsman
error 2023-01-15 09:28:47: Failed to start zigbee
error 2023-01-15 09:28:47: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-01-15 09:28:47: Exiting...
error 2023-01-15 09:28:47: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:347:23)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

cryd-s, 2023-01-15:
Here some additional infos:

- change panid:

```advanced:
  pan_id: 7000 (was before changing somthing with 6892 i think=´)
  ext_pan_id:
    - 228 (was before changing 221)
    - 228
    - 228
    - 228
    - 228
    - 228
    - 228
    - 228
  channel: 3

buhito81, 2023-01-23:
Exactly the same issue. Stick works fine in ZHA

cryd-s, 2023-01-27:
is no one able to fix this ?

buhito81, 2023-01-28:
Apperently not... 😭 I'm having the same. What would help is a debug log showing what network is has detected, if any.

cryd-s, 2023-01-28:
`2023-01-28T16:18:15.477Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-28T16:18:15.477Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":99}
2023-01-28T16:18:15.478Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,99,0,83]
2023-01-28T16:18:15.481Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,19,1,0,113]
2023-01-28T16:18:15.481Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,19,1,0,113]
2023-01-28T16:18:15.481Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 19 - [1,0] - 113
2023-01-28T16:18:15.481Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvLength - {"length":1}
2023-01-28T16:18:15.481Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-28T16:18:15.481Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvReadExt - {"id":99,"offset":0}
2023-01-28T16:18:15.481Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,33,28,99,0,0,0,90]
2023-01-28T16:18:15.487Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,28,0,1,0,127]
2023-01-28T16:18:15.487Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,28,0,1,0,127]
2023-01-28T16:18:15.487Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 28 - [0,1,0] - 127
2023-01-28T16:18:15.487Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvReadExt - {"status":0,"len":1,"value":{"type":"Buffer","data":[0]}}
2023-01-28T16:18:15.487Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-28T16:18:15.487Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":98}
2023-01-28T16:18:15.487Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,98,0,82]
2023-01-28T16:18:15.490Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,19,16,0,96]
2023-01-28T16:18:15.490Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,19,16,0,96]
2023-01-28T16:18:15.490Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 19 - [16,0] - 96
2023-01-28T16:18:15.491Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvLength - {"length":16}
2023-01-28T16:18:15.491Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-28T16:18:15.491Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvReadExt - {"id":98,"offset":0}
2023-01-28T16:18:15.491Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,33,28,98,0,0,0,91]
2023-01-28T16:18:15.497Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,18,97,28,0,16,154,54,12,55,73,205,10,216,109,190,130,46,82,204,232,129,54]
2023-01-28T16:18:15.497Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,18,97,28,0,16,154,54,12,55,73,205,10,216,109,190,130,46,82,204,232,129,54]
2023-01-28T16:18:15.497Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 18 - 3 - 1 - 28 - [0,16,154,54,12,55,73,205,10,216,109,190,130,46,82,204,232,129] - 54
2023-01-28T16:18:15.497Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvReadExt - {"status":0,"len":16,"value":{"type":"Buffer","data":[154,54,12,55,73,205,10,216,109,190,130,46,82,204,232,129]}}
2023-01-28T16:18:15.497Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-28T16:18:15.497Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":98}
2023-01-28T16:18:15.497Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,98,0,82]
2023-01-28T16:18:15.501Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,19,16,0,96]
2023-01-28T16:18:15.501Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,19,16,0,96]
2023-01-28T16:18:15.501Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 19 - [16,0] - 96
2023-01-28T16:18:15.501Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvLength - {"length":16}
2023-01-28T16:18:15.501Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-28T16:18:15.501Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvWriteExt - {"id":98,"offset":0,"len":16,"value":{"type":"Buffer","data":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13]}}
2023-01-28T16:18:15.501Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,22,33,29,98,0,0,0,16,0,1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13,91]
2023-01-28T16:18:15.508Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,97,29,0,125]
2023-01-28T16:18:15.508Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,97,29,0,125]
2023-01-28T16:18:15.508Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 1 - 29 - [0] - 125
2023-01-28T16:18:15.508Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvWriteExt - {"status":0}
2023-01-28T16:18:15.508Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-28T16:18:15.508Z zigbee-herdsman:adapter:zStack:startup:commissioning beginning network commissioning
2023-01-28T16:18:15.508Z zigbee-herdsman:adapter:zStack:znp:SREQ --> APP_CNF - bdbSetChannel - {"isPrimary":1,"channel":2048}
2023-01-28T16:18:15.508Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,47,8,1,0,8,0,0,43]
2023-01-28T16:18:15.511Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,111,8,0,102]
2023-01-28T16:18:15.511Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,111,8,0,102]
2023-01-28T16:18:15.511Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 15 - 8 - [0] - 102
2023-01-28T16:18:15.511Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- APP_CNF - bdbSetChannel - {"status":0}
[... cut ...]
2023-01-28T16:18:19.452Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,79,128,8,2,0,198]
2023-01-28T16:18:19.452Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,79,128,8,2,0,198]
2023-01-28T16:18:19.452Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 15 - 128 - [8,2,0] - 198
2023-01-28T16:18:19.452Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- APP_CNF - bdbComissioningNotifcation - {"status":8}
2023-01-28T16:18:19.452Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []`

cryd-s, 2023-01-28:
here a herdsman log

```2023-01-28T16:18:15.477Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-28T16:18:15.477Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":99}
2023-01-28T16:18:15.478Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,99,0,83]
2023-01-28T16:18:15.481Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,19,1,0,113]
2023-01-28T16:18:15.481Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,19,1,0,113]
2023-01-28T16:18:15.481Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 19 - [1,0] - 113
2023-01-28T16:18:15.481Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvLength - {"length":1}
2023-01-28T16:18:15.481Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-28T16:18:15.481Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvReadExt - {"id":99,"offset":0}
2023-01-28T16:18:15.481Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,33,28,99,0,0,0,90]
2023-01-28T16:18:15.487Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,28,0,1,0,127]
2023-01-28T16:18:15.487Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,28,0,1,0,127]
2023-01-28T16:18:15.487Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 28 - [0,1,0] - 127
2023-01-28T16:18:15.487Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvReadExt - {"status":0,"len":1,"value":{"type":"Buffer","data":[0]}}
2023-01-28T16:18:15.487Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-28T16:18:15.487Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":98}
2023-01-28T16:18:15.487Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,98,0,82]
2023-01-28T16:18:15.490Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,19,16,0,96]
2023-01-28T16:18:15.490Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,19,16,0,96]
2023-01-28T16:18:15.490Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 19 - [16,0] - 96
2023-01-28T16:18:15.491Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvLength - {"length":16}
2023-01-28T16:18:15.491Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-28T16:18:15.491Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvReadExt - {"id":98,"offset":0}
2023-01-28T16:18:15.491Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,33,28,98,0,0,0,91]
2023-01-28T16:18:15.497Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,18,97,28,0,16,154,54,12,55,73,205,10,216,109,190,130,46,82,204,232,129,54]
2023-01-28T16:18:15.497Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,18,97,28,0,16,154,54,12,55,73,205,10,216,109,190,130,46,82,204,232,129,54]
2023-01-28T16:18:15.497Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 18 - 3 - 1 - 28 - [0,16,154,54,12,55,73,205,10,216,109,190,130,46,82,204,232,129] - 54
2023-01-28T16:18:15.497Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvReadExt - {"status":0,"len":16,"value":{"type":"Buffer","data":[154,54,12,55,73,205,10,216,109,190,130,46,82,204,232,129]}}
2023-01-28T16:18:15.497Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-28T16:18:15.497Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":98}
2023-01-28T16:18:15.497Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,98,0,82]
2023-01-28T16:18:15.501Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,19,16,0,96]
2023-01-28T16:18:15.501Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,19,16,0,96]
2023-01-28T16:18:15.501Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 19 - [16,0] - 96
2023-01-28T16:18:15.501Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvLength - {"length":16}
2023-01-28T16:18:15.501Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-28T16:18:15.501Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvWriteExt - {"id":98,"offset":0,"len":16,"value":{"type":"Buffer","data":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13]}}
2023-01-28T16:18:15.501Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,22,33,29,98,0,0,0,16,0,1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13,91]
2023-01-28T16:18:15.508Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,97,29,0,125]
2023-01-28T16:18:15.508Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,97,29,0,125]
2023-01-28T16:18:15.508Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 1 - 29 - [0] - 125
2023-01-28T16:18:15.508Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvWriteExt - {"status":0}
2023-01-28T16:18:15.508Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-28T16:18:15.508Z zigbee-herdsman:adapter:zStack:startup:commissioning beginning network commissioning
2023-01-28T16:18:15.508Z zigbee-herdsman:adapter:zStack:znp:SREQ --> APP_CNF - bdbSetChannel - {"isPrimary":1,"channel":2048}
2023-01-28T16:18:15.508Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,47,8,1,0,8,0,0,43]
2023-01-28T16:18:15.511Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,111,8,0,102]
2023-01-28T16:18:15.511Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,111,8,0,102]
2023-01-28T16:18:15.511Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 15 - 8 - [0] - 102
[... cut ...]
2023-01-28T16:18:19.452Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,79,128,8,2,0,198]
2023-01-28T16:18:19.452Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,79,128,8,2,0,198]
2023-01-28T16:18:19.452Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 15 - 128 - [8,2,0] - 198
2023-01-28T16:18:19.452Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- APP_CNF - bdbComissioningNotifcation - {"status":8}
2023-01-28T16:18:19.452Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []```

cryd-s, 2023-01-28:
and here a debug log

```debug 2023-01-28 17:20:28: Loaded state from file /config/z2m-sonoff/state.json
info  2023-01-28 17:20:28: Logging to console and directory: '/config/z2m-sonoff/log/2023-01-28.17-20-28' filename: log.txt
info  2023-01-28 17:20:28: Starting Zigbee2MQTT version 1.29.2 (commit #unknown)
info  2023-01-28 17:20:28: Starting zigbee-herdsman (0.14.83-hotfix.0)
debug 2023-01-28 17:20:28: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/z2m-sonoff/coordinator_backup.json","databaseBackupPath":"/config/z2m-sonoff/database.db.backup","databasePath":"/config/z2m-sonoff/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB0"}}'
error 2023-01-28 17:21:33: Error while starting zigbee-herdsman
error 2023-01-28 17:21:33: Failed to start zigbee
error 2023-01-28 17:21:33: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-01-28 17:21:33: Exiting...
error 2023-01-28 17:21:33: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:347:23)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)```

Koenkk, 2023-01-29:
Make sure to power off you old network completely OR specify a different channel in the configuration.yaml (you have to change this in the z2m configuration.yaml file, not the HA addon page.

cryd-s, 2023-01-29:
Tried this several times but nothing worked - deleted the Complet z2m folder and reinstall it too.

Koenkk, 2023-01-29:
In your log you see that the old settings are still used `"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221]`, note that 3 is not a valid channel, it has to be between 11 and 26.

cryd-s, 2023-01-29:
Ok so i can put Channel to 15 and panID to 222?

cryd-s, 2023-01-30:
ok perfect it works thanks a lot!!!

closed 2023-01-30 by cryd-s