Koenkk/zigbee2mqtt #16175: Devices can no longer be paired [CC2531][Docker]

TyroTechSoft, 2023-01-15:
### What happened?

I am currently moving from a Raspberry Pi 4 to an ODroid H3.
Unfortunately, not all devices can be paired.
On the old system, these devices worked.

Affected devices:
- MCCGQ11LM
- DJT11LM

Do you need more information?

Thanks, Greetings
Tyro

### Zigbee2MQTT version

Old Host: 1.28.4 // New Host: 1.29.2

### Adapter firmware version

CC2531_DEFAULT_20211115

### Adapter

CC2531

kizio1000, 2023-01-15:
Same here, back to dev version and no problem with pairing. Nothing in log, devices are just not pairing, nothing shows up and device stays in pairing mode. 

Sonoff zogbee E
EZSP v8
6.10.3.0 build 297

TyroTechSoft, 2023-01-15:
Dev Version of Zigbee2MQTT?
`koenkk/zigbee2mqtt:latest-dev` ???

TyroTechSoft, 2023-01-15:
Ok, now only 3 devices won't pair.
And 2 devices have problems with the interview.

TyroTechSoft, 2023-01-15:
@kizio1000
Can you learn all your devices?
Unfortunately not for me.

TyroTechSoft, 2023-01-17:
I bought a new stick.
Now it works again.
Stick => SONOFF Zigbee 3.0 USB Dongle Plus,TI CC2652P + CP2102(N)

closed 2023-01-17 by TyroTechSoft