Koenkk/zigbee2mqtt #16176: TuYa TS011F_plug_1 add measurement_poll_interval to spec information on webui

shirou93, 2023-01-15:
### What happened?

When i edited in yaml file measurement_poll_interval option working like a charm.

[image]

Propably coverter not expose measurement_poll_interval to webui.

I add issue https://github.com/nurikk/zigbee2mqtt-frontend/issues/1651, but its propably not frontend issue?

### Zigbee2MQTT version

Z2M 1.29.2

### Adapter firmware version

20220219

### Adapter

Sonoff 3 USB

thomasloven, 2023-02-18:
Still an issue.

closed 2023-03-28 by github-actions[bot]

vahaldor, 2023-05-29:
> ~Still an issue.~
> 
> Edit: Sorry, I was mistaken. A device reporting as `TS011F_plug_1` is not supposed to expose `measurement_poll_interval`.

Thomas, I have the plug_1 version and it stopped reporting 0W power. Is it possible to manually force it to do it? Is it possible to configure measurement_poll_interval for this plug as well? Where do you edit the yaml in z2m? thank you