Koenkk/zigbee2mqtt #16178: Legrand by Netatmo 067776A not well recognized

cyrilphoenix71, 2023-01-15:
### What happened?

I've bought a Legrand by Netatmo 067726A version
Normally is a 26 is for white version and A is for postion get/set support
But when I include it to z2m it's recognise like a BTicino K4027C_L4027C_N4027C_NT4027C, why ?
More position support not works well, only open/close/stop
More again after include, front LED still green blinking.
Here the state.json

### Zigbee2MQTT version

1.29.1

### Adapter firmware version

0x26780700

### Adapter

dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE1996382-if00

### Debug log

[state.json (1).zip](https://github.com/Koenkk/zigbee2mqtt/files/10420276/state.json.1.zip)

imfreddie, 2023-01-18:
I can add that i have a lot of difficulties to pair my bticino shutters as well. I have 4 and i manage to only pair one so far. i tried every trick i found on internet to pair but im not able to pair them properly. It says everything get interview correctly but i cannot use the percentage for opening the shutters.

cyrilphoenix71, 2023-02-04:
any updates ? comments ?

dlepretre, 2023-02-25:
Hi all
I've just bought the same device (`067726A` white version (french)) and I have strange behavior too.
The device is successfully paired and identified as `BTicino Shutter SW with level control (K4027C/L4027C/N4027C/NT4027C)`

when I use the device button (`up`, `down`, `stop`), it works as expected. When I use  HA/zigbee2mqtt (close, open) it has strange behavior:
- the state is always `CLOSE` even if level is `0` or `100`
- if I set level to a value between `0` and `100`, the shutter stop in the middle and device stop to work (need to reinstall previous switch)

If you have any advices that could help me to analyze, you're welcome.

closed 2023-04-05 by github-actions[bot]