Koenkk/zigbee2mqtt #16180: [New device support]: power_on_behavior with ADEO LDSENK02F

Blastmun, 2023-01-15:
### Link

https://www.zigbee2mqtt.io/devices/LDSENK02F.html#adeo-ldsenk02f

### Database entry

{"id":16,"type":"Router","ieeeAddr":"0x0c4314fffe5bf377","nwkAddr":3348,"manufId":4727,"manufName":"ADEO","powerSource":"Mains (single phase)","modelId":"LDSENK02F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,1794,2820,64642],"outClusterList":[3,10,25],"clusters":{"genBasic":{"attributes":{"modelId":"LDSENK02F","manufacturerName":"ADEO","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20200217-586","swBuildId":"2.6"}},"seMetering":{"attributes":{"multiplier":1,"divisor":100,"currentSummDelivered":[0,0]}},"genOnOff":{"attributes":{"onOff":1}},"haElectricalMeasurement":{"attributes":{"activePower":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00258da6d6","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b00258da6d6","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b00258da6d6","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":1},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{"scenes":{"0_0":{"name":"Scene 0","state":{"state":"ON"}}}}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20200217-586","swBuildId":"2.6","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-414661485},"lastSeen":1673796879995,"defaultSendRe

### Comments

Hello, 
The "power_on_behavior" function is not supported by the socket. On the 10A ([LDSENK01F]  it works perfectly. 
How can I solve this problem? 
I'm on HomeAssistant OS and Zigbee2mqtt Edge

### External converter

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const e = exposes.presets;

const fzLocal = {
    LDSENK08: {
        cluster: 'ssIasZone',
        type: 'commandStatusChangeNotification',
        convert: (model, msg, publish, options, meta) => {
            const zoneStatus = msg.data.zonestatus;
            return {
                contact: !((zoneStatus & 1) > 0),
                vibration: (zoneStatus & 1<<1) > 0,
                tamper: (zoneStatus & 1<<2) > 0,
                battery_low: (zoneStatus & 1<<3) > 0,
            };
        },
    },
};

module.exports = [
    {
        zigbeeModel: ['LDSENK08'],
        model: 'LDSENK08',
        vendor: 'ADEO',
        description: 'ENKI LEXMAN wireless smart door window sensor with vibration',
        fromZigbee: [fzLocal.LDSENK08, fz.battery],
        toZigbee: [],
        exposes: [e.battery_low(), e.contact(), e.vibration(), e.tamper(), e.battery()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint);
        },
    },
    {
        zigbeeModel: ['LDSENK09'],
        model: 'LDSENK09',
        vendor: 'ADEO',
        description: 'Security system key fob',
        fromZigbee: [fz.command_arm, fz.command_panic],
        toZigbee: [],
        exposes: [e.action(['panic', 'disarm', 'arm_partial_zones', 'arm_all_zones'])],
        onEvent: async (type, data, device) => {
            // Since arm command has a response zigbee-herdsman doesn't send a default response.
            // This causes the remote to repeat the arm command, so send a default response here.
            if (data.type === 'commandArm' && data.cluster === 'ssIasAce') {
                await data.endpoint.defaultResponse(0, 0, 1281, data.meta.zclTransactionSequenceNumber);
            }
[... 182 lines cut ...]
        description: '10A/16A EU smart plug',
        vendor: 'ADEO',
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_genLevelCtrl_report],
        toZigbee: [tz.on_off],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.onOff(endpoint);
            await reporting.activePower(endpoint);
            await reporting.currentSummDelivered(endpoint);
            await reporting.readMeteringMultiplierDivisor(endpoint);
        },
        exposes: [e.power(), e.switch(), e.energy()],
    },
    {
        zigbeeModel: ['LDSENK10'],
        model: 'LDSENK10',
        vendor: 'ADEO',
        description: 'ENKI LEXMAN motion sensor',
        fromZigbee: [fz.ias_occupancy_alarm_1],
        toZigbee: [],
        exposes: [e.occupancy(), e.battery_low(), e.tamper()],
    },
];
```

Koenkk, 2023-01-16:
Added!

closed 2023-01-16 by Koenkk

Blastmun, 2023-01-16:
Ty =)

fjmarcelo, 2023-05-11:
Hello, I bought an ADEO LDSENK02S smart plug last week and zigbee2mqtt says it is **NOT** compatible. Is there any way to add it? Thank you very much in advance.