Koenkk/zigbee2mqtt #16181: [New device support]: Relabeled Sunricher SR-ZG9092A (ROB_200-040-0)

MischaDing, 2023-01-15:
### Link

https://www.robbshop.nl/robb-smarrt-thermostaat-elektrische-vloerverwarming-zigbee

### Database entry

{"id":9,"type":"Router","ieeeAddr":"0x1fff00030000004f","nwkAddr":15343,"manufId":4644,"manufName":"Sunricher","powerSource":"Mains (single phase)","modelId":"HK-LN-HEATER-A","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,4,5,9,10,513,516,1794,2820],"outClusterList":[1030],"clusters":{"genBasic":{"attributes":{"modelId":"HK-LN-HEATER-A","manufacturerName":"Sunricher","powerSource":1,"zclVersion":3,"appVersion":20,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"6.9.1.0_r4"}},"genDeviceTempCfg":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":7,"inClusterList":[0,3],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"stackVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":20,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"6.9.1.0_r4","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1673796618090,"defaultSendRequestWhen":"immediate"}

### Comments

This is sold by ROBB shop as item ROB_200-040-0. The device actually shows that it is the Sunricher SR-ZG9092A. This Sunricher device is already supported, only not the relabeled version.

[SR-ZG9092A_instruction.pdf](https://github.com/Koenkk/zigbee2mqtt/files/10420632/SR-ZG9092A_instruction.pdf)
[ROB_200-040-0_download_1.pdf](https://github.com/Koenkk/zigbee2mqtt/files/10420633/ROB_200-040-0_download_1.pdf)

MischaDing, 2023-01-15:
In [sunricher.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/sunricher.js) converter renamed         `zigbeeModel: ['ZG9092'],` to `zigbeeModel: ['HK-LN-HEATER-A'],` and found partially working.

pitkaha, 2023-01-19:
Hi, having the same problem as you. Take a look at [https://github.com/Koenkk/zigbee2mqtt/issues/15972](url), maybe we can get it to work

MischaDing, 2023-02-08:
> Hi, having the same problem as you. Take a look at [https://github.com/Koenkk/zigbee2mqtt/issues/15972](url), maybe we can get it to work

Hi, I will close this case since your case is progressed much more.

closed 2023-02-08 by MischaDing