Koenkk/zigbee2mqtt #16186: Lonsonho QS-Zigbee-S04-2C-LN - No network route (205)

Alex2zl, 2023-01-15:
### What happened?

Hello. I have a similar issue (https://github.com/Koenkk/zigbee2mqtt/issues/13623)
   There are 2 identical relays from Lonsonho (QS-Zigbee-S04-2C-LN) which are not correctly defined in z2mqtt (defined as TS0002). And the worst thing is that after some time (usually after a day), these devices stop responding to commands, they cannot be controlled, an error appears: "No network route (205)". After a while, the connection can recover itself again, and the relay can be controlled again, but then it is lost again, and so on in a circle.
With an external adapter, the devices are detected correctly as lonsohno, but this does not solve the "No network route (205)" error.
The devices are close to the coordinator (SONOFF ZIGBEE 3.0 USB DONGLE PLUS), there are other devices in the room (including routers) that work correctly.
What can I do to try and fix this error?

### What did you expect to happen?

Correct detection of the device model in z2mqtt, and stable operation of the device, without error No network route (205)

### Zigbee2MQTT version

1.29.2-1

### Adapter firmware version

zStack3x0 (20220219)

### Adapter

SONOFF ZIGBEE 3.0 USB DONGLE PLUS

timderspieler, 2023-01-18:
This is the same issue as: https://github.com/Koenkk/zigbee2mqtt/issues/12774

Same Zigbee Stick, Same situation, but another device from Lonsonho. We need a sniff from a network to fix this. ( -> https://github.com/Koenkk/zigbee2mqtt/issues/12774#issuecomment-1177633964 )

Alex2zl, 2023-01-19:
Unfortunately, I do not have a CC2531 stick, but the ezsp stick should come soon, I'll try to get data from it

closed 2023-02-26 by github-actions[bot]