Koenkk/zigbee2mqtt #16188: [Feature request]: Network Key and Pan ID auto generated during install to minimize insecure installs

bharvey88, 2023-01-15:
### Is your feature request related to a problem? Please describe

While learning more about zigbee and z2m in the home assistant discord server I have been made aware that my install is far from secure. Per: https://www.zigbee2mqtt.io/advanced/zigbee/03_secure_network.html#change-zigbee-network-encryption-key it says the install uses a default encryption key and pan ID. It also states we are unable to generate the key ourselves when using the addon inside home assistant OS. I find this default install behavior to be very worrying and I request that there be a way to automatically generate both a random network key and pan ID to make this a noob-friendly installation experience.

### Describe the solution you'd like

Automatically generate randomized network key and pan ID and extended pan ID during installation process.

### Describe alternatives you've considered

Alternatively, a required set of options for both network key and pan ID and extended pan ID in the configuration tab with accompanying documentation in the documentation tab of the addon.

### Additional context

No additional context to give, let me know if you want any added information.

Stale bot closed this issue and I was unable to reopen so I made a new one: https://github.com/Koenkk/zigbee2mqtt/issues/14868

eikowagenknecht, 2023-01-29:
I found that while the docs say it can't be done when using the HA addon, it's not that hard (or I'm missing something).

Warning: You need to repair all devices after you do that!!

- Shutdown zigbee2mqtt addon
- Open \\<homeassistant_ip>\config\zigbee2mqtt\configuration.yaml
- set network_key and ext_pan_id like so:
  ```
  advanced:
      network_key:
      - 1
      - 2
      - 3
      - 4
      - 5
      - 6
      - 7
      - 8
      - 9
      - 10
      - 11
      - 12
      - 13
      - 14
      - 15
      - 220
    ext_pan_id:
      - 1
      - 2
      - 3
      - 4
      - 5
      - 6
      - 7
      - 221
  ```
  Obviously use other values for each entry (betweed 1 and 221 should be safe).
- Remove coordinator_backup.json
- Restart and repair all your devices

bharvey88, 2023-01-30:
> I found that while the docs say it can't be done when using the HA addon, it's not that hard (or I'm missing something).
>
> Warning: You need to repair all devices after you do that!!
>
>     * Shutdown [...]

This is useful but I created this issue with the hopes of changing the default install behavior of z2m to be a secure install not using known keys, not just the ability to change settings after install.

eikowagenknecht, 2023-01-30:
Sure, that would be even better 👍

bharvey88, 2023-03-02:
Still needs to be resolved

junosuarez, 2023-03-02:
I agree this is an important issue. Insecure defaults are one of the most common category of security vulnerabilities in systems design. Most people setting up a zigbee network are looking to get started as quickly as possible and have little knowledge at the beginning. However, by the time they learn about this problem, they may have invested time and effort in setting up a large network with many devices - having to re-pair each of them creates a lot of friction which may lead to continuing with the insecure default configuration.

The proposed solution of random key generation at installation time is a good one. Let's please keep in mind the various ways that people first install or run zigbee2mqtt. In particular, Docker or other declarative, minimal configuration setups. 

Further, I think this warrants showing a warning in the zigbee2mqtt web UI with a link to instructions on how to improve network security.

Are there any constraints, considerations, or concerns from the project maintainers?

HelpITru, 2023-03-02:
This is a huge security hole. And a lot of time spent looking for problems. I have three neighbors with default settings and devices walk between networks. I'm the only smart one who will change the panid soon. And they don't know and will suffer until I tell them what to do.

HelpITru, 2023-03-02:
after all , after changing the PanID , you still need to repaired all devices !!! and at the first initial installation, there are no linked devices yet. Thanks!

bharvey88, 2023-04-02:
bump before the bot gets here.. still hoping to have this looked at by @Koenkk

bharvey88, 2023-05-19:
bad bot, not stale.

bharvey88, 2023-06-06:
dont even think about it bot. still open!

sjorge, 2023-06-06:
There is actually already support to generate a random network key by setting it to `GENERATE`. This was requested a few times over the years and I believe the main hurdle is that by setting it to in `data/configuration.yaml` in the repository it would break most (all?) existing installs.

I vaguely remember there being talk about moving `data/configuration.yaml` to `data/configuration.yaml.example` and copying it to `data/configuration.yaml` if it's missing... but I don't remember what happened to it as I can't find the ticket. It was at least a few years ago.

lyricnz, 2023-06-21:
Yes, please implement random default;  using a preconfigured static value is dumb.

bharvey88, 2023-06-30:
continuing to hope for a reply from the maintainer here

Koenkk, 2023-07-01:
I'm fine with doing this, but the thing holding me back is that the people who run a `git clone` installation will get a merge conflict on their `configuration.yaml`.

bharvey88, 2023-07-05:
> I'm fine with doing this, but the thing holding me back is that the people who run a `git clone` installation will get a merge conflict on their `configuration.yaml`.

I understand your hesitation but if this is increasing security for all new installs I believe it warrants a breaking change for the minority of people using git clone.

Koenkk, 2023-07-05:
I will check if I can implement something non-breaking.

attila-lendvai, 2023-07-15:
IIRC, it's possible to add a file to `.gitignore`, and also have it committed to the repo using `git add -f` -- if you want to accommodate a usecase where the read-write data storage file is also added to the git repo, that is. i wouldn't. this is just one example of the many ways it can lead to headaches.

also, people who run from a git checkout (me included) are expected to be savvy enough to deal with breaking changes.

but now i'll need to disassemble a couple of wall sockets to properly secure my network, which is certainly more headache than a dirty `git diff` in a checkout.

Koenkk, 2023-07-18:
I've made a PR to fix this issue (without introducing a breaking change). Please let me know if this is what you expect: https://github.com/Koenkk/zigbee2mqtt/pull/18357

spudwebb, 2023-07-18:
your PR generates a random `network_key`, but what about `pan_id `and `ext_pan_id `don't we need those to be randomly generated as well?

MattWestb, 2023-07-19:
Not needed **if** only having one 25.4 network in radio range but if not they must being unique or you is getting conflicts and strange problems in some cases.
I think the best is also randomizing this 2 for getting it working without problems for all user cases.

bharvey88, 2023-07-19:
> your PR generates a random `network_key`, but what about `pan_id `and `ext_pan_id `don't we need those to be randomly generated as well?

Yes I agree, I opened this issue to solve the problems of an insecure install due to all three of these (network_key, pan_id, ext_pan_id) not just network_key. This came to light after realizing zha does randomize all of these things during first install. I will also note that zha does an initial energy scan and chooses the best channel for you, which frankly I also think is a great QoL feature.

Koenkk, 2023-07-19:
pan_id/ext_pan_id has nothing to do with security (see https://github.com/Koenkk/zigbee2mqtt/pull/18357#issuecomment-1642477934)

MattWestb, 2023-07-19:
> pan_id/ext_pan_id has nothing to do with security (see [#18357 (comment)](https://github.com/Koenkk/zigbee2mqtt/pull/18357#issuecomment-1642477934))

So you can also see it but in in Zigbee and other 802.15.4 network its one part of the security configuration.
Also you is getting mush problems then users is setting up more Z2M systems in radio range and getting strange problems in there systems then all is badly configuration from start.

Koenkk, 2023-08-12:
Closing this now as `network_key`, `ext_pan_id` an `pan_id` are now generated for new installations.

closed 2023-08-12 by Koenkk

louis-lau, 2024-10-04:
@Koenkk docker install instructions told me to download https://raw.githubusercontent.com/Koenkk/zigbee2mqtt/master/data/configuration.yaml, which does not have the values set to GENERATE. Does that mean no secure defaults for new docker installations?

Impact123, 2024-10-04:
There's also an example here: https://github.com/Koenkk/zigbee2mqtt/blob/master/data/configuration.example.yaml

louis-lau, 2024-10-04:
Perhaps the instructions need updating in that case

Koenkk, 2024-10-04:
@louis-lau would you mind making a PR to update the instructions?

louis-lau, 2024-12-15:
Took me much longer than it should have (life), but see #3329 :)