Koenkk/zigbee2mqtt #16189: z2m crashes with the same strange error.

komw, 2023-01-15:
### What happened?

Z2m sometimes constantly crashes. Sometimes there is a few days without any problem, but sometimes it can happen every few minutes, always showing the same error.  I'm attaching the debug log. From the logs, only the last line repeats after every crash:

```
Error: Write 0x90fd9ffffede3c52/1 genLevelCtrl({"onLevel":255}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 9001 - 1 - 4 - 8 - 4 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)
```
Its strange error, because the 0x90fd9ffffede3c52/1 device its a Ikea Tradfrii bulb, which is located about 3meters from coordinator.

### What did you expect to happen?

I expect that this error should be handled in z2m

### How to reproduce it (minimal and precise)

There is no way to reproduce, it randomly happens.

### Zigbee2MQTT version

1.29.2 commit: unknown

### Adapter firmware version

20220219

### Adapter

sonoff zigbee 3.0 usb dongle plus (ZBDongle-P)

### Debug log

Zigbee2MQTT:debug 2023-01-15 21:25:43: Received MQTT message on 'homeassistant/sensor/0x680ae2fffe3e4896/last_seen/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/z-hall-table/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x680ae2fffe3e4896"],"manufacturer":"Lidl","model":"Silvercrest 3 gang switch, with 4 USB (EU, FR, CZ, BS) (HG06338)","name":"z-hall-table"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:clock","name":"z-hall-table last seen","state_topic":"zigbee2mqtt/z-hall-table","unique_id":"0x680ae2fffe3e4896_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'
Zigbee2MQTT:debug 2023-01-15 21:25:43: Received MQTT message on 'homeassistant/sensor/0x00158d00054dbd56/battery/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/z-bathroom-sensor-1/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00158d00054dbd56"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"z-bathroom-sensor-1","sw_version":"3000-0001"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"z-bathroom-sensor-1 battery","state_class":"measurement","state_topic":"zigbee2mqtt/z-bathroom-sensor-1","unique_id":"0x00158d00054dbd56_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Zigbee2MQTT:debug 2023-01-15 21:25:43: Received MQTT message on 'homeassistant/sensor/0x00158d00054dbd56/temperature/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/z-bathroom-sensor-1/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00158d00054dbd56"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"z-bathroom-sensor-1","sw_version":"3000-0001"},"device_class":"temperature","enabled_by_default":true,"name":"z-bathroom-sensor-1 temperature","state_class":"measurement","state_topic":"zigbee2mqtt/z-bathroom-sensor-1","unique_id":"0x00158d00054dbd56_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
Zigbee2MQTT:debug 2023-01-15 21:25:43: Received MQTT message on 'homeassistant/sensor/0x00158d00054dbd56/humidity/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/z-bathroom-sensor-1/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00158d00054dbd56"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"z-bathroom-sensor-1","sw_version":"3000-0001"},"device_class":"humidity","enabled_by_default":true,"name":"z-bathroom-sensor-1 humidity","state_class":"measurement","state_topic":"zigbee2mqtt/z-bathroom-sensor-1","unique_id":"0x00158d00054dbd56_humidity_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.humidity }}"}'
Zigbee2MQTT:debug 2023-01-15 21:25:43: Received MQTT message on 'homeassistant/sensor/0x00158d00054dbd56/pressure/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/z-bathroom-sensor-1/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00158d00054dbd56"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"z-bathroom-sensor-1","sw_version":"3000-0001"},"device_class":"pressure","enabled_by_default":true,"name":"z-bathroom-sensor-1 pressure","state_class":"measurement","state_topic":"zigbee2mqtt/z-bathroom-sensor-1","unique_id":"0x00158d00054dbd56_pressure_zigbee2mqtt","unit_of_measurement":"hPa","value_template":"{{ value_json.pressure }}"}'
Zigbee2MQTT:debug 2023-01-15 21:25:43: Received MQTT message on 'homeassistant/sensor/0x00158d00054dbd56/voltage/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/z-bathroom-sensor-1/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00158d00054dbd56"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"z-bathroom-sensor-1","sw_version":"3000-0001"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","name":"z-bathroom-sensor-1 voltage","state_class":"measurement","state_topic":"zigbee2mqtt/z-bathroom-sensor-1","unique_id":"0x00158d00054dbd56_voltage_zigbee2mqtt","unit_of_measurement":"mV","value_template":"{{ value_json.voltage }}"}'
Zigbee2MQTT:debug 2023-01-15 21:25:43: Received MQTT message on 'homeassistant/sensor/0x00158d00054dbd56/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/z-bathroom-sensor-1/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00158d00054dbd56"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"z-bathroom-sensor-1","sw_version":"3000-0001"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"z-bathroom-sensor-1 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/z-bathroom-sensor-1","unique_id":"0x00158d00054dbd56_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:debug 2023-01-15 21:25:43: Received MQTT message on 'homeassistant/sensor/0x00158d00054dbd56/last_seen/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/z-bathroom-sensor-1/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00158d00054dbd56"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"z-bathroom-sensor-1","sw_version":"3000-0001"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:clock","name":"z-bathroom-sensor-1 last seen","state_topic":"zigbee2mqtt/z-bathroom-sensor-1","unique_id":"0x00158d00054dbd56_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'
Zigbee2MQTT:debug 2023-01-15 21:25:43: Received MQTT message on 'homeassistant/sensor/0x00158d00042d1bc3/battery/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/z-kitchen-window-sensor-1/availability","value_template":"{ [... cut ...]
Error: Write 0x90fd9ffffede3c52/1 genLevelCtrl({"onLevel":255}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 9001 - 1 - 4 - 8 - 4 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)

canedje, 2023-01-19:
I do have the same problem only with sonoff E version dongle

p6sfrx725z28knfy, 2023-02-03:
I also experience Z2M (1.30) crashing with a similar error message as the last log entry, the device is an Ikea Tradfro [LED1536G5](https://www.zigbee2mqtt.io/devices/LED1536G5.html#ikea-led1536g5) with the CC2652P based Sonoff ZBDongle-P running the 20221226 coordinator firmware:

> Error: Write 0xXXXXXXXXXXXXXXXX/1 genLevelCtrl({"onLevel":255}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

canedje, 2023-03-06:
The problem is still there

closed 2023-04-15 by github-actions[bot]

sekot, 2023-04-15:
still there

canedje, 2023-04-15:
Yep here also

schattenphoenix, 2023-05-26:
Also here

imthefrizzlefry, 2023-06-04:
I just started running into this

m33ts4k0z, 2023-06-26:
Same issue with the Skyconnect dongle

SJensen721, 2023-08-20:
I also have this issue. Z2M, Sonoff Dongle P, did not flash (I read it came pre-flashed).

smulle48, 2023-09-11:
Same issue here, both with slae.sh and so off adapter

Koenkk, 2023-09-11:
This is very likely a firmware issue, I suggest to try 20230901 from this thread (provide feedback there): https://github.com/Koenkk/Z-Stack-firmware/discussions/474