Koenkk/zigbee2mqtt #16195: Devices keep leaving the network

evlo, 2023-01-16:
### What happened?

For whatever reason, usually when Z2M crashes for some reason for a while, devices some leave the network

### What did you expect to happen?

devices to never leave network until I explicitly request that

### How to reproduce it (minimal and precise)

Have (tuya) devices
crash/kill Z2M
after maybe a day start it up

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

HA yellow

### Debug log

```
info  2023-01-14 13:50:35: MQTT publish: topic 'zigbee2mqtt-ttyAMA1/ZM2 Tuya LivingRoom Switch main', payload '{"last_seen":"2023-01-14T12:50:35.820Z","linkquality":208,"state_center":"ON","state_left":"OFF","state_right":"OFF"}'
warn  2023-01-14 13:50:36: Device 'ZM2 Tuya Lidl Office button lampicka' left the network
info  2023-01-14 13:50:37: MQTT publish: topic 'zigbee2mqtt-ttyAMA1/bridge/event', payload '{"data":{"friendly_name":"ZM2 Tuya Lidl Office button lampicka","ieee_address":"0x2c1165fffeca9120"},"type":"device_leave"}'
warn  2023-01-14 13:50:37: Device 'ZM2 Tuya Living room TH lower ' left the network
info  2023-01-14 13:50:37: MQTT publish: topic 'zigbee2mqtt-ttyAMA1/bridge/event', payload '{"data":{"friendly_name":"ZM2 Tuya Living room TH lower ","ieee_address":"0xa4c138b278f1443b"},"type":"device_leave"}'
warn  2023-01-14 13:50:37: Device 'ZM2 Tuya Entryway Switch bathroom' left the network
info  2023-01-14 13:50:38: MQTT publish: topic 'zigbee2mqtt-ttyAMA1/bridge/event', payload '{"data":{"friendly_name":"ZM2 Tuya Entryway Switch bathroom","ieee_address":"0x804b50fffe0992e8"},"type":"device_leave"}'
```
I have enabled debug logging and will update next time this happens

kizio1000, 2023-01-16:
Same here. Already after couple hours from install latest update to 1.9.2 I lost four devices, one day more another two. This did not happen before update. Nothing else changed in my setup. Additionally I can not bind them back. They do not bind, just no reaction from z2m. If I change to Edge version I am able to bind them back with no problem and then they are available again in 1.9.2 too for a time being but in the meantime other are falling out the network. Before 1.8.4 my z2m of Sonoff-E was rock solid, now it's not reliable and this freaks me out as I have central heating and like 50% of lights on it...

shackrat, 2023-01-22:
You are not alone.  I'v been having devices drop off the network daily.  Everything previously ran stable in ZHA for years.

mvivaldi, 2023-02-23:
I have the same problem with version 1.30.1.
After a crash a device leave the network
```
info  2023-02-23 19:45:13: MQTT publish: topic 'z2monfalcone/bridge/event', payload '{"data":{"friendly_name":"termostato-monfalcone","ieee_address":"0x000d6f00112e4666"},"type":"device_leave"}'
```
this happend 3 or 4 times in the last few days.

maxime1992, 2023-03-26:
Up

bartzz, 2023-04-06:
I have the same issue with `koenkk/zigbee2mqtt:1.28.2` (tried other versions as well)

```
Zigbee2MQTT:info  2023-04-06 09:35:23: MQTT publish: topic 'zigbee2mqtt/korytarz', payload '{"battery":63,"humidity":59.77,"last_seen":"2023-04-06T07:35:23.343Z","linkquality":108,"power_outage_count":2078,"pressure":1010.6,"temperature":19.81,"voltage":2945}'

Zigbee2MQTT:warn  2023-04-06 09:39:38: Device 'korytarz' left the network

Zigbee2MQTT:info  2023-04-06 09:39:38: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"korytarz","ieee_address":"0x00158d0008025b89"},"type":"device_leave"}'
```

Devices just leave randomly, they'll stay in Z2M for a few hours after repairing, then they get lost again. Any solutions to this?

DannyFeliz, 2023-05-07:
This still happens.

DannyFeliz, 2023-06-07:
I ended up going back to the Conbee II since I couldn't fix this problem.

closed 2023-07-16 by github-actions[bot]

Stooovie, 2023-08-28:
Same, i have two devices disappear from Z2M just in the last 24 hours. Not disconnected or unavailable. Actually gone from the devices list.

boekabart, 2023-09-21:
Same here, using the SkyConnect stick. It was a POWERED (not battery) Ikea device this time, very near the stick (2 meters max), it basically factory reset itself it seems (started blinking), and by enabling joining, it quickly came back in the device list. Weird.

muspelheim, 2023-12-15:
same for me