Koenkk/zigbee2mqtt #16198: IKEA Tradfri E1743 can be bound to 7 devices max

damarges, 2023-01-16:
### What happened?

After upgrading E1743 to Firmware 2.3.079 I know that remotes can be bound to devices only. Not to groups anymore.

But it seems like I can not bind it to more than seven devices. Z2M says it binds it and also lists the 8th (and more) binding. But upon pressing a Button. Only the First seven Lights are switched.

### What did you expect to happen?

Bind more than 7 devices or at least give an Error in the binding Process if there really is a limit of 7 devices per E1743 remote.

### How to reproduce it (minimal and precise)

Reset and repair Tradfri E1743 with latest Firmware 2.3.079

Then try to bind more than 7 Ikea tradfri GU10 light bulbs with that remote

### Zigbee2MQTT version

1.29.2 commit: unknown

### Adapter firmware version

zStack3x0

### Adapter

Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001

evlo, 2023-01-16:
_super sorry for the offtopic going here, here, but can you please share screenshot of the binding settings, I still can't figure out how to use it, even after reading the wiki, example woulde be great_

damarges, 2023-01-16:
[image]

damarges, 2023-01-28:
Anyone any ideas? Exactly 7 Lights seems to be the Maximum when Binding via z2m. Pairing the Switch manually to more than 7 Lights (directly holding it Close to the bulb) is working

closed 2023-03-07 by github-actions[bot]