Koenkk/zigbee2mqtt #16202: HEIMAN HS1CG-E (Gas sensor) always reports no gas leak (gas:false)

noirrak, 2023-01-16:
### What happened?

Gas sensor always reports no gas leak.

### What did you expect to happen?

When leak detected/test button pressed, status should change to "gas:true".

### How to reproduce it (minimal and precise)

Press test button on sensor. Status will report "gas:false".

### Zigbee2MQTT version

1.29.2-1

### Adapter firmware version

20220219

### Adapter

CC2652P

### Debug log

```
Debug 2023-01-16 15:15:18Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"warn"}},"transaction":"xaok3-1"}'
Info 2023-01-16 15:15:20Succesfully changed options
Info 2023-01-16 15:15:20MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"xaok3-2"}'
Debug 2023-01-16 15:16:02Received Zigbee message from '0x000d6f001539fdba', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":34}' from endpoint 1 with groupID 0
Info 2023-01-16 15:16:02MQTT publish: topic 'zigbee2mqtt/0x000d6f001539fdba', payload '{"battery_low":false,"gas":false,"linkquality":80,"tamper":false}'
Debug 2023-01-16 15:16:03Received Zigbee message from '0x000d6f001539fdba', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0
Info 2023-01-16 15:16:03MQTT publish: topic 'zigbee2mqtt/0x000d6f001539fdba', payload '{"battery_low":false,"gas":false,"linkquality":109,"tamper":false}'
Debug 2023-01-16 15:16:46Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"info"}},"transaction":"xaok3-3"}'
Info 2023-01-16 15:16:46Succesfully changed options
Info 2023-01-16 15:16:46MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"xaok3-3"}'
Info 2023-01-16 15:16:52Succesfully changed options
Info 2023-01-16 15:16:52MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"xaok3-6"}'
```

Koenkk, 2023-01-16:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['GASSensor-EM', '358e4e3e03c644709905034dae81433e'],
    model: 'HS1CG-E',
    vendor: 'HEIMAN',
    description: 'Combustible gas sensor',
    fromZigbee: [fz.ias_gas_alarm_2],
    toZigbee: [],
    whiteLabel: [{vendor: 'Piri', model: 'HSIO18008'}],
    exposes: [e.gas(), e.battery_low(), e.tamper()],
};

module.exports = definition;

```
- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

noirrak, 2023-01-16:
! It works now! Thank you for your help!

**Debug:**
```
Debug 2023-01-16 19:42:51Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"info"}},"transaction":"xd5cr-1"}'
Info 2023-01-16 19:42:51Succesfully changed options
Info 2023-01-16 19:42:51MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"xd5cr-1"}'
Info 2023-01-16 19:43:05MQTT publish: topic 'zigbee2mqtt/0x000d6f001539fdba', payload '{"battery_low":false,"gas":true,"linkquality":102,"tamper":false}'
Info 2023-01-16 19:43:06MQTT publish: topic 'zigbee2mqtt/0x000d6f001539fdba', payload '{"battery_low":false,"gas":false,"linkquality":98,"tamper":false}'
Info 2023-01-16 19:43:25Succesfully changed options
Info 2023-01-16 19:43:25MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"xd5cr-2"}'

```

Koenkk, 2023-01-17:
@AntorFr does this device work fine in z2m 1.29.2 for you?

AntorFr, 2023-01-18:
Thanks @Koenkk,

I don't think my changes was shipped with Z2MQTT v1.29.2 (ie. if a remove my external converter my gas sensor is displayed as not supported)

But i do test the proposed change (inside my converter). it's not working (for '358e4e3e03c644709905034dae81433e' device).

I tested with :
`fromZigbee: [fz.ias_gas_alarm_2]`
which **is not working** for me (Gas stays "False" all the time)
and 
`fromZigbee: [fz.ias_gas_alarm_1]`
which **is working** for me (Gas moves to "True" when triggering test mode)

so we probably have to create 2 different definitions ?
Let me know if you want my contribution on it.

Best,

Koenkk, 2023-01-19:
@noirrak @AntorFr can you both provide the data/database.db entry of your device?

AntorFr, 2023-01-21:
@Koenkk sure.

`{"id":16,"type":"Router","ieeeAddr":"0x000d6f0014f33fcc","nwkAddr":62267,"manufId":4619,"manufName":"HEIMAN","powerSource":"Mains (single phase)","modelId":"358e4e3e03c644709905034dae81433e","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,9,1280],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"zclVersion":1}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0024c0a885","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":19,"stackVersion":2,"hwVersion":16,"dateCode":"2019.9.12","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1674288124043,"defaultSendRequestWhen":"immediate"}
`

noirrak, 2023-01-21:
@Koenkk 
```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0024bfd672","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281,10],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null,"defaultSendRequestWhen":"immediate"}
{"id":2,"type":"Router","ieeeAddr":"0x000d6f001539fdba","nwkAddr":19047,"manufId":4619,"manufName":"HEIMAN","powerSource":"Mains (single phase)","modelId":"GASSensor-EM","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,9,1280],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0024bfd672","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":21,"stackVersion":2,"hwVersion":16,"dateCode":"2019.9.30","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1674335614250,"defaultSendRequestWhen":"immediate"}
{"id":3,"type":"EndDevice","ieeeAddr":"0x000d6f0011d5eaf7","nwkAddr":26213,"manufId":4619,"manufName":"HEIMAN","powerSource":"Battery","modelId":"COSensor-EM","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"COSensor-EM","manufacturerName":"HEIMAN","powerSource":3,"zclVersion":1,"appVersion":18,"stackVersion":2,"hwVersion":16,"dateCode":"2018.3.9","swBuildId":"1.1.1"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0024bfd672","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0024bfd672","endpointID":242}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":18,"stackVersion":2,"hwVersion":16,"dateCode":"2018.3.9","swBuildId":"1.1.1","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1674334650243,"defaultSendRequestWhen":"immediate"}
```

Koenkk, 2023-01-22:
Does it work for both of you with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    ias_gas_alarm: {
        cluster: 'ssIasZone',
        type: 'commandStatusChangeNotification',
        convert: (model, msg, publish, options, meta) => {
            const zoneStatus = msg.data.zonestatus;
            return {
                gas: ((zoneStatus & 1) > 0) || (zoneStatus & 1<<1) > 0,
                tamper: (zoneStatus & 1<<2) > 0,
                battery_low: (zoneStatus & 1<<3) > 0,
            };
        },
    },
}

const definition = {
    zigbeeModel: ['GASSensor-EM', '358e4e3e03c644709905034dae81433e'],
    model: 'HS1CG-E',
    vendor: 'HEIMAN',
    description: 'Combustible gas sensor',
    fromZigbee: [fzLocal.ias_gas_alarm],
    toZigbee: [],
    whiteLabel: [{vendor: 'Piri', model: 'HSIO18008'}],
    exposes: [e.gas(), e.battery_low(), e.tamper()],
}

module.exports = definition;

```

AntorFr, 2023-01-27:
@Koenkk thanks for the upgrade.

tested on my side, working good !

Best

noirrak, 2023-03-05:
@Koenkk , Thanks.

closed 2023-03-06 by Koenkk

noirrak, 2024-06-01:
@Koenkk, the problem occurred again.