Koenkk/zigbee2mqtt #16207: Critical Error: Failed to call 'Frontend' 'stop'

stkr22, 2023-01-16:
### What happened?

Started Zigbee2MQTT with a fresh install. But It keeps crashing tried 1.29.2 and latest-dev. I get the following output:
```
Using '/data' as data directory
Zigbee2MQTT:info  2023-01-16 15:21:48: Logging to console and directory: '/data/log/2023-01-16.15-21-48' filename: log.txt
Zigbee2MQTT:info  2023-01-16 15:21:49: Starting Zigbee2MQTT version 1.29.0 (commit #dddce57)
Zigbee2MQTT:info  2023-01-16 15:21:49: Starting zigbee-herdsman (0.14.83)
Zigbee2MQTT:info  2023-01-16 15:22:02: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2023-01-16 15:22:02: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":114,"product":0,"revision":"0x26720700","transportrev":0},"type":"ConBee2/RaspBee2"}'
Zigbee2MQTT:info  2023-01-16 15:22:05: Currently 0 devices are joined:
Zigbee2MQTT:warn  2023-01-16 15:22:05: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2023-01-16 15:22:05: Allowing new devices to join.
Zigbee2MQTT:warn  2023-01-16 15:22:05: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2023-01-16 15:22:05: Zigbee: allowing new devices to join.
Zigbee2MQTT:error 2023-01-16 15:23:27: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of null (reading 'clients')
at Frontend.stop (/app/lib/extension/frontend.ts:83:39)
at Controller.callExtensions (/app/lib/controller.ts:316:40)
at Controller.stop (/app/lib/controller.ts:193:9)
at stop (/app/index.js:111:5)
at process.handleQuit (/app/index.js:117:9))
Zigbee2MQTT:error 2023-01-16 15:23:27: Not connected to MQTT server!
Zigbee2MQTT:error 2023-01-16 15:23:27: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
Zigbee2MQTT:info  2023-01-16 15:23:27: Disconnecting from MQTT server
TypeError: Cannot read properties of undefined (reading 'end')
at MQTT.disconnect (/app/lib/mqtt.ts:129:21)
at Controller.stop (/app/lib/controller.ts:198:9)
at stop (/app/index.js:111:5)
at process.handleQuit (/app/index.js:117:9)
```

### What did you expect to happen?

Zigbee2mqtt run normally and start frontend.

### How to reproduce it (minimal and precise)

Deploy Zigbee2MQTT as Deployment in Kubernetes

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

0x26720700

### Adapter

ConbeeII

### Debug log

Using '/data' as data directory
Zigbee2MQTT:info  2023-01-16 15:21:48: Logging to console and directory: '/data/log/2023-01-16.15-21-48' filename: log.txt
Zigbee2MQTT:info  2023-01-16 15:21:49: Starting Zigbee2MQTT version 1.29.0 (commit #dddce57)
Zigbee2MQTT:info  2023-01-16 15:21:49: Starting zigbee-herdsman (0.14.83)
Zigbee2MQTT:info  2023-01-16 15:22:02: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2023-01-16 15:22:02: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":114,"product":0,"revision":"0x26720700","transportrev":0},"type":"ConBee2/RaspBee2"}'
Zigbee2MQTT:info  2023-01-16 15:22:05: Currently 0 devices are joined:
Zigbee2MQTT:warn  2023-01-16 15:22:05: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2023-01-16 15:22:05: Allowing new devices to join.
Zigbee2MQTT:warn  2023-01-16 15:22:05: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2023-01-16 15:22:05: Zigbee: allowing new devices to join.
Zigbee2MQTT:error 2023-01-16 15:23:27: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of null (reading 'clients')
at Frontend.stop (/app/lib/extension/frontend.ts:83:39)
at Controller.callExtensions (/app/lib/controller.ts:316:40)
at Controller.stop (/app/lib/controller.ts:193:9)
at stop (/app/index.js:111:5)
at process.handleQuit (/app/index.js:117:9))
Zigbee2MQTT:error 2023-01-16 15:23:27: Not connected to MQTT server!
Zigbee2MQTT:error 2023-01-16 15:23:27: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
Zigbee2MQTT:info  2023-01-16 15:23:27: Disconnecting from MQTT server
TypeError: Cannot read properties of undefined (reading 'end')
at MQTT.disconnect (/app/lib/mqtt.ts:129:21)
at Controller.stop (/app/lib/controller.ts:198:9)
at stop (/app/index.js:111:5)
at process.handleQuit (/app/index.js:117:9)

stkr22, 2023-01-16:
Just figured it out: The actual issues is the failing mqtt connection. The error message is a bit misleading here(at least for me).

However, did this change somehow? Because mosquitto is in the same namespace as Zigbee2MQTT and it used to work by just setting it to "mqtt://mosquitto"(name of the service). Now it only works if I set the full name "mqtt://mosquitto.home-control.svc.cluster.local".

Koenkk, 2023-01-16:
Fixed the error!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

stkr22, 2023-01-16:
That was fast 😀

Thanks for the quick feedback! 👍

closed 2023-01-17 by Koenkk