Koenkk/zigbee2mqtt #16208: IKEA E1525/E1745 silently disappears from the network and needs repairing

marcor78, 2023-01-16:
### What happened?

Hello,
I'm new to Zigbee2mqtt (recently migrated from Deconz) and running on Sonoff Dongle Version E (EFR32MG21).
I have a strange behaviour from one of my Tradfri Motion Sensor (E1525/E1745): it works fine for hours and then suddenly, without any advice or notification, simply disappears from the network, both devices list and map, like it never joined the network.
I notice this just because the automation based on this sensor (a light bulb) doesn't trigger.
After repairing the device, everything ok for a few hours, then again disappears.
Battery is 100% full and I have a couple of other similar devices that never leave the network.

How can I debug this? Or at least have a notice when the device leaves the network?
Home Assistant tends to see the device as off until I restart Zigbee2Mqtt or reload the integration.

### What did you expect to happen?

Device shouldn't leave the network all of a sudden and apparently without reason

### How to reproduce it (minimal and precise)

Pairing IKEA E1525/E1745 motion sensor correctly, interview and configuration all ok.
Running smoothly for a few hours
Device leave the network and doesn't reconnect anymore until repairing it

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

6.10.3.0

### Adapter

Sonoff Dongle Version E (EFR32MG21)

### Debug log

nothing in the log, simply the device doesn't exist in the network anymore

evlo, 2023-01-16:
similar
https://github.com/Koenkk/zigbee2mqtt/issues/16195
https://github.com/Koenkk/zigbee2mqtt/issues/16167

marcor78, 2023-01-16:
> similar #16195 #16167

yes I saw those issues too, but it seems somehow they happened after a major update. In my case I did absolutely nothing to the network, the device just disappear and needs to be repaired.
Could it be something specific for the device (Ikea Tradfri E1525/E1745)? The strangest thing is that I have 3 of them in my network but only 1 of them has this strange behaviour...

Xeevis, 2023-01-17:
I'm pretty convinced this is related to **EFR32MG21** chip, all the linked issues have it in common (HA yellow uses it too) and its support is marked as "experimental" in Z2M after all. When I moved some devices from my rock-solid ConBee II setup to ZBDongle-E, TRV started dropping from network while other devices hold fine. Nothing ever dropped on CB2.

So probably not random or general problem, just device specific on this particular chip.

evlo, 2023-01-17:
My other network with TI chip as coordinator did not suffer this, but that one is also composed of non tuya devices (some based on EFR32MG21) but anecdotal evidence supports that hypothesis

SucubusJenny, 2023-01-20:
I have the same ore a similar issue, after a while the sensor will drop off the network but then will reappear and either reconnect properly or it will show "Interview completed False" but still fully work.

closed 2023-02-27 by github-actions[bot]