Koenkk/zigbee2mqtt #16211: manual starting with npm works fine, systemctl does not

ltersteg, 2023-01-16:
### What happened?

I followed this tutorial:
https://www.zigbee2mqtt.io/guide/installation/01_linux.html

everything works fine. But when i came to the systemctl section:
https://www.zigbee2mqtt.io/guide/installation/01_linux.html#optional-running-as-a-daemon-with-systemctl

i followed the tutorial, used this configuration file:

### What did you expect to happen?

i expected the status zigbee2mqtt to stay online

### How to reproduce it (minimal and precise)

i exactly followed the tutorial and used this configuration in my file:

### Zigbee2MQTT version

 1.29.1 commit: 7d67ffc

### Adapter firmware version

i dont know

### Adapter

Texas Instruments CC2531

### Debug log

login as: q
q@192.168.188.25's password:
Linux openhabian 5.15.84-v7l+ #1613 SMP Thu Jan 5 12:01:26 GMT 2023 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Jan 16 15:42:25 2023 from 192.168.188.24
q@openhabian:~ $ systemctl status zigbee2mqtt.service
● zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor p>
     Active: active (running) since Mon 2023-01-16 15:45:36 CET; 4s ago
   Main PID: 2034 (npm start)
      Tasks: 19 (limit: 4915)
        CPU: 5.852s
     CGroup: /system.slice/zigbee2mqtt.service
             ├─2034 npm start
             ├─2045 sh -c -- node index.js
             └─2046 node index.js

Jan 16 15:45:36 openhabian systemd[1]: Started zigbee2mqtt.
Jan 16 15:45:37 openhabian npm[2034]: > zigbee2mqtt@1.29.1 start
Jan 16 15:45:37 openhabian npm[2034]: > node index.js

q@openhabian:~ $ systemctl status zigbee2mqtt.service
● zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor p>
     Active: activating (auto-restart) (Result: exit-code) since Mon 2023-01-16>
    Process: 2660 ExecStart=/usr/bin/npm start (code=exited, status=1/FAILURE)
   Main PID: 2660 (code=exited, status=1/FAILURE)
        CPU: 6.481s

Jan 16 15:50:46 openhabian systemd[1]: zigbee2mqtt.service: Failed with result >
Jan 16 15:50:46 openhabian systemd[1]: zigbee2mqtt.service: Consumed 6.481s CPU>
lines 1-9/9 (END)...skipping...
● zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Mon 2023-01-16 15:50:46 CET; 7s ago
    Process: 2660 ExecStart=/usr/bin/npm start (code=exited, status=1/FAILURE)
   Main PID: 2660 (code=exited, status=1/FAILURE)
        CPU: 6.481s

Jan 16 15:50:46 openhabian systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jan 16 15:50:46 openhabian systemd[1]: zigbee2mqtt.service: Consumed 6.481s CPU time.
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~

q@openhabian:~ $ sudo journalctl -u zigbee2mqtt.service -f
[sudo] password for q:
-- Journal begins at Fri 2023-01-13 12:53:47 CET. --
Jan 16 15:52:02 openhabian npm[2823]:     at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:197:12)
Jan 16 15:52:02 openhabian npm[2823]:     at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:197:12)
Jan 16 15:52:02 openhabian npm[2823]:     at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:197:12)
Jan 16 15:52:02 openhabian npm[2823]:     at Object.mkdirSync (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:82:10)
Jan 16 15:52:02 openhabian npm[2823]:     at Object.init (/opt/zigbee2mqtt/lib/util/logger.ts:43:12)
Jan 16 15:52:02 openhabian npm[2823]:     at new Controller (/opt/zigbee2mqtt/lib/controller.ts:53:16)
Jan 16 15:52:02 openhabian npm[2823]:     at start (/opt/zigbee2mqtt/index.js:106:18)
Jan 16 15:52:02 openhabian systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Jan 16 15:52:02 openhabian systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jan 16 15:52:02 openhabian systemd[1]: zigbee2mqtt.service: Consumed 6.453s CPU time.
Jan 16 15:52:12 openhabian systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 33.
Jan 16 15:52:12 openhabian systemd[1]: Stopped zigbee2mqtt.
Jan 16 15:52:12 openhabian systemd[1]: zigbee2mqtt.service: Consumed 6.453s CPU time.
Jan 16 15:52:12 openhabian systemd[1]: Started zigbee2mqtt.
Jan 16 15:52:13 openhabian npm[2844]: > zigbee2mqtt@1.29.1 start
Jan 16 15:52:13 openhabian npm[2844]: > node index.js
Jan 16 15:52:17 openhabian npm[2856]: Error: EACCES: permission denied, mkdir '/opt/zigbee2mqtt/data/log/2023-01-16.15-52-17'
Jan 16 15:52:17 openhabian npm[2856]:     at Object.mkdirSync (node:fs:1382:3)
Jan 16 15:52:17 openhabian npm[2856]:     at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:199:16)
Jan 16 15:52:17 openhabian npm[2856]:     at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:197:12)
Jan 16 15:52:17 openhabian npm[2856]:     at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:197:12)
Jan 16 15:52:17 openhabian npm[2856]:     at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:197:12)
Jan 16 15:52:17 openhabian npm[2856]:     at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:197:12)
Jan 16 15:52:17 openhabian npm[2856]:     at Object.mkdirSync (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:82:10)
Jan 16 15:52:17 openhabian npm[2856]:     at Object.init (/opt/zigbee2mqtt/lib/util/logger.ts:43:12)
Jan 16 15:52:17 openhabian npm[2856]:     at new Controller (/opt/zigbee2mqtt/lib/controller.ts:53:16)
Jan 16 15:52:17 openhabian npm[2856]:     at start (/opt/zigbee2mqtt/index.js:106:18)
Jan 16 15:52:17 openhabian systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Jan 16 15:52:17 openhabian systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jan 16 15:52:17 openhabian systemd[1]: zigbee2mqtt.service: Consumed 6.353s CPU time.
Jan 16 15:52:27 openhabian systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 34.
Jan 16 15:52:27 openhabian systemd[1]: Stopped zigbee2mqtt.
Jan 16 15:52:27 openhabian systemd[1]: zigbee2mqtt.service: Consumed 6.353s CPU time.
Jan 16 15:52:27 openhabian systemd[1]: Started zigbee2mqtt.
Jan 16 15:52:28 openhabian npm[2870]: > zigbee2mqtt@1.29.1 start
Jan 16 15:52:28 openhabian npm[2870]: > node index.js

troyfung, 2023-01-17:
Check the /etc/systemd/system/zigbee2mqtt.service file.
Do you have a "User" specified and do they have permission to create directories in the log location?

You could try and and remove "User" from the systemd file.

ltersteg, 2023-01-20:
yeah that was it, thanks a lot

closed 2023-02-27 by github-actions[bot]