Koenkk/zigbee2mqtt #16212: VZM31-SN Missing Homeassistant discovery topic for "action"

mdallaire, 2023-01-16:
### What happened?

I installed several VZM31-SN in my house in the last few days and for all of them I have no "action" sensor in Home-Assistant.

- I see the action in z2m (ex: up_double), 
- I see the topic (action) and payload (up_double) under the zigbee2mqtt topic of the switch.
- I don't see a action sensor in Home-Assitant
- I don't see a Home-Assistant discovery topic for the action sensor under the homeassistant topic.

### What did you expect to happen?

I expected to see a action sensor in Home-Assistant.

### How to reproduce it (minimal and precise)

Add VZM31-SN to z2m, check home-assistant for action sensor.

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

20220219

### Adapter

zzh

### Debug log

[zigbee.log](https://github.com/Koenkk/zigbee2mqtt/files/10427958/zigbee.log)

hexonMD, 2023-01-21:
Experiencing the same issue. Tried repairing the switch multiple times with no success. I don't however see any actions within Z2M. The section just shows N/A.

InovelliUSA, 2023-01-28:
What settings do you have configured for your HA integration in z2m?

[image]

@hexonMD which coordinator do you have?

mdallaire, 2023-01-28:
I do not have the legacy trigger checked.

I was able to trigger automations by using the device based triggers in Home-Assistant.

Does that mean the device based triggers are the "not legacy" triggers and that the action sensor is considered a legacy trigger?

hexonMD, 2023-01-28:
I’m running the Electrolama zig-a-zig-ah!

InovelliUSA, 2023-01-31:
> I was able to trigger automations by using the device based triggers in Home-Assistant.
> 
> Does that mean the device based triggers are the "not legacy" triggers and that the action sensor is considered a legacy trigger?

Yes, I believe so. I recommend using the device based triggers, but I think enabling the legacy option will enable the other.

@hexonMD I think yours is a different issue and I think I am working with you on that in the support system. The switch is binding to endpoint 0 on your coordinator and the commands are either not being sent or are not being parsed in z2m.

InovelliUSA, 2023-02-03:
@hexonMD I am trying to force the converter to bind to endpoint 1 of the coordinator, but I am not having success. I am wondering if anyone else has experience doing this? I have searched through a lot of sample code and cannot find anything. This is what I am using and it seems to be binding to the lowest available ep on your coordinator (coordinatorEndpoint - which is 0).

```
configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint ,  [
                'seMetering',
                'haElectricalMeasurement',
                'genOnOff',
                'genLevelCtrl',
            ]);
            await reporting.onOff(endpoint);

            await endpoint.read('haElectricalMeasurement', [
                'acPowerMultiplier',
                'acPowerDivisor',
            ]);
            await reporting.readMeteringMultiplierDivisor(endpoint);
        }
```

closed 2023-03-14 by github-actions[bot]