Koenkk/zigbee2mqtt #16223: [New device support]: Legrand Hospitality On Off Switch (no converter available for state_1 problem)

brantmacga, 2023-01-16:
### Link

https://www.legrand.us/wiring-devices/designer-switches-and-outlets/adorne-smart-switch-with-netatmo-white/p/wnal10w1

### Database entry

 0x646266fffe622372

### Comments

I have the device paired, but when I try to turn it on or off, I get the error "no converter available for state_1". 

When I physically turn the switch on or off, I see that reflected correctly within Z2M. Also the LED on/off function of the device are working correctly. 

When I go to the "exposes" tab in Z2M, I see this....
**Endpoint 1:
      state_1:
      On/off state of the switch             ON ---- OFF**
      
And when I select ON I get the 'No converter available for state_1" error message. 

Any ideas? 

### External converter

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const ota = require('zigbee-herdsman-converters/lib/ota');

const readInitialBatteryState = async (type, data, device) => {
    if (['deviceAnnounce'].includes(type)) {
        const endpoint = device.getEndpoint(1);
        const options = {manufacturerCode: 0x1021, disableDefaultResponse: true};
        await endpoint.read('genPowerCfg', ['batteryVoltage'], options);
    }
};

const tzLocal = {
    auto_mode: {
        key: ['auto_mode'],
        convertSet: async (entity, key, value, meta) => {
            const mode = {'off': 0x00, 'auto': 0x02, 'on_override': 0x03};
            const payload = {data: Buffer.from([mode[value]])};
            await entity.command('manuSpecificLegrandDevices3', 'command0', payload);
            return {state: {'auto_mode': value}};
        },
    },
};

module.exports = [

    {
        zigbeeModel: ['Hospitality on off switch'],
        model: 'WNAL10',
        vendor: 'Legrand',
        description: 'Adorne Smart Switch with Netatmo',
        fromZigbee: [fz.on_off, fz.legrand_binary_input_on_off, fz.legrand_cluster_fc01],
        toZigbee: [tz.on_off, tz.legrand_settingEnableLedInDark, tz.legrand_settingEnableLedIfOn],
        exposes: [e.switch().withEndpoint(1),
            exposes.binary('led_in_dark', ea.ALL, 'ON', 'OFF').withDescription(`Enables the LED when the light is turned off, allowing to` +
                ` see the switch in the dark`),
            exposes.binary('led_if_on', ea.ALL, 'ON', 'OFF').withDescription('Enables the LED when the light is turned on')],
        meta: {multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(endpoint);
        },
    },
];
```

Danit2, 2023-01-17:
Delete the

meta: {multiEndpoint: true},

and the .withEndpoint(1)

brantmacga, 2023-01-17:
That was it, thanks! I thought I’d already removed the meta: line but completely missed .withEndpoint(1).