Koenkk/zigbee2mqtt #16225: [Feature request]: Allow to overwrite simulated_brightness value of dimmers/remotes

ASNNetworks, 2023-01-17:
### Is your feature request related to a problem? Please describe

**I would really like the option to be able to overwrite/set the simulated_brightness option of devices (like dimmers etc that support that feature).** 

Hi, so I don't know if this is possible, but I think since it's all simulated anyway, it should be possible. It's a very niche problem and I haven't seen anyone else ask about this. But I am convinced this will be a fantastic feature that really helps automations and dimming lights. I apologize in advance for the extensive explanation, but I hope it's clear what I'm going for.

### Describe the solution you'd like

I have a few Philips Hue Tap Dials and messed around with the simulated_brightness feature and created automations with Node-RED with it. It works really well I must say. I have it now set up that when I use the dial, it uses the brightness of the dimmer (simulated brightness) as the payload to set the brightness of a light. This works very consistent, especially when using different lights inside a Zigbee group.

Here's my issue however: as long as you only use the Tap Dial, the brightness of the light and the Tap Dial are always in sync. So you know what to expect when using the dial to dim/brighten the light. However, when I change the brightness of the light using other ways (like service calls through automations or Lovelace) then the light will have a different brightness than the simulated_brightess value of the Tap Dial. This causes a messy experience.

**Example:**

I use the Tap Dial and keep dialing till it's simulated_brightness value hits 50/255. All payloads till that point get sent to the light I want to dim through Node-RED. So the lights dims as I keep rotating all the way to 50/255 (great!). Then I turn off my light till morning. The next morning my light turns on at max brightness through an automation. So that sits at 255/255. When I then use the Tap Dial to dim my light, it sends a brightnes payload lower than its last value (50/255) (I use a delta of 25, so would go to 25/255). So what happens, is that my light then dims from 255/255 all the way to 25/255, since that's where the Tap Dial value sits.

I tried to send an mqtt command to set the brightness of the Tap Dial, but that gives out an error in zigbee2mqtt that the converter doesn't exist. So at the moment, that value is read only. My solution would be to make it writable as well, just like normal lights. So then I can incorporate an mqtt message in the morning that sets the Tap Dial brightness to 255/255, basically syncing up the value with the light. That way it feels like an actual dimmer that is tied to the light.

### Describe alternatives you've considered

I tried other ways by using global values in Node-RED to store brightness there and use as a syncing pont, but nothing really works. As long as the brightness value of the Tap Dial cannot be set by the user, it will not be possible to truly use it as a dimmer with consistent experience.

### Additional context

I hope this can get implemented :)

Thyraz, 2023-01-24:
+1 for this.

I would like to use this feature with dfferent devices on one tap dial.
E.g. Button 1 switches to volume control, Button 2 to brightness,  ...

So if I dim up the lights to 100% and then switch to volume and start to turn down, I would send volume comands like 95%, 90%, ... to my AV-Receiver, which might kill all ppl in the room (at least their hearing). :P 

So I would also like to send the current volume / brightness when swithing modes. 
That way, rotating the dial would start from the current point and allow to use the whole value range (regardless of the value that was used before).

ASNNetworks, 2023-01-24:
> +1 for this.
>
> I would like to use this feature with dfferent devices on one tap dial. E.g. Button 1 switches to volume control, Button 2 to brightness, ...
>
> So if I dim up the lights to 100% and [...]

Yes, also great use case for this. Currently I created automations and using the rotate_dial payloads to send out volume_set and brightness_step commands. This works fine, but I'd rather tie it to the simulated_brightness since it feels more direct.

ASNNetworks, 2023-02-28:
still relevant

closed 2023-04-09 by github-actions[bot]