Koenkk/zigbee2mqtt #16228: Moes BHT-002-GCLZB "homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: State max length is 255 characters."

adorobis, 2023-01-17:
### What happened?

Since a while (not sure when exactly, but probably since one of the recent latest-dev releases my HA is flooded with the following error message for one of the devices: Moes BHT-002-GCLZB Apart from that in the logs the device and the integration seems to work ok. Full error message:
```
Logger: homeassistant
Source: core.py:1140
First occurred: 12:00:27 (2393 occurrences)
Last logged: 13:11:34

Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...x7f9357c36420>)
Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...x7f9357e8fa70>)
Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...x7f9357e8f6f0>)
Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...x7f9357e8c7b0>)
Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...x7f9357df8dd0>)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/client.py", line 696, in _mqtt_handle_message
    self._mqtt_data.state_write_requests.process_write_state_requests()
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 267, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 556, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 673, in _async_write_ha_state
    self.hass.states.async_set(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1480, in async_set
    state = State(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1140, in __init__
    raise InvalidStateError(
homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.termostat_szary_program. State max length is 255 characters.
```
Not sure if it is z2m or ha core issue.

### What did you expect to happen?

No errors in the log

### How to reproduce it (minimal and precise)

Just add the Moes BHT-002-GCLZB to zigbee2mqtt and observe HA logs.

### Zigbee2MQTT version

1.29.2-dev commit: e56dff4

### Adapter firmware version

0x26780700

### Adapter

Conbee II

### Debug log

[_zigbee_logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/10434906/_zigbee_logs.txt)

adorobis, 2023-01-17:
I have also opened an issue on ha core as I'm not sure which component is causing it:
https://github.com/home-assistant/core/issues/86081

adorobis, 2023-01-17:
Based on the feedback from HA Core team the limit is for a long time 255 characters for entity state. I've noticed that it is now trying to publish a new entity `sensor.[name]_program` and it is `Unknown` due to this limitation. I believe it should be either removed or split into multiple entities to comply with this limits on HA side.

ondrejbaranek, 2023-01-30:
There is 255 char limit for state in HA, it needs to be attribute, or to be splitted to comply with that limit...

cyberdie, 2023-02-01:
Same problem here AFTER updating Z2M to 1.30...
No error before that....

olympia, 2023-02-01:
if you don't need the program as a sensor this issue can be workarounded by disabling the program sensor in HA :)

adorobis, 2023-02-01:
Wow, cool. How come I have not figured this out :) Thanks!

AlekseiPun, 2023-02-10:
> если вам не нужна программа как датчик, то эту проблему можно обойти, отключив программный датчик в HA :)

it is not entirely clear how to do this, please tell me for beginners how to fix this error point by point

Strish, 2023-02-10:
> it is not entirely clear how to do this, please tell me for beginners how to fix this error point by point

In the properties of this device in HA, find the sensor "Program" and deactivate it

AlekseiPun, 2023-02-10:
> > не совсем понятно как это сделать, подскажите пожалуйста для новичков как исправить эту ошибку по пунктам
> 
> В свойствах этого устройства в HA найдите датчик "Программа" и деактивируйте его

Great! Thank u)

osmeest, 2023-02-26:
I recently installed Z2M in my HA setup (next to ZHA), the better support for the Moes BHT002 was one of the reason for this.
Although I can configure the thermostat through HA climate entity, I see that the status in HA is not always in sync with the one reported by Z2M. I enabled logging and then found that very same message. Could it be the reason for the desynchronization of states ?

marcusg, 2023-03-29:
> I recently installed Z2M in my HA setup (next to ZHA), the better support for the Moes BHT002 was one of the reason for this. Although I can configure the thermostat through HA climate entity, I see [...]

@osmeest Yes, at least for me - I'm using the instructions from @Strish as a workaround. 

> In the properties of this device in HA, find the sensor "Program" and deactivate it

Koenkk, 2023-05-07:
Fixed now: https://github.com/Koenkk/zigbee2mqtt/issues/16460#issuecomment-1537341464

closed 2023-05-07 by Koenkk