Koenkk/zigbee2mqtt #16229: ZB-DL01 issues with power_on_behavior property

aalmazanarbs, 2023-01-17:
### What happened?

The property `power_on_behavior` is not working as expected. When it is configured to `off` and the device is turned off two times, the light stays on.

because of the power is off, the first time the light is off, but if the power is off again and turn on, the second time the light is on.

It is weird, since the second turn on, is not notified as event.

### What did you expect to happen?

The light must always be switched off regardless of the number of times the power is switched on or off.

### How to reproduce it (minimal and precise)

- Configure `power_on_behavior` to `off`
- Turn off power
- Turn on power (light is off)
- Turn off power 
- Turn on power (the light is on :-1: )

### Zigbee2MQTT version

1.29.2-1 

### Adapter firmware version

20220219

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

aalmazanarbs, 2023-02-17:
Can anyone help?

c0dr1ver, 2023-03-24:
> Can anyone help?

could you publish database.db entry of this device?

aalmazanarbs, 2023-03-24:
> > Can anyone help?
> 
> could you publish database.db entry of this device?

Of course, here are you are:

```
{
    "id": 5,
    "type": "Router",
    "ieeeAddr": "0xa4c1381986443a13",
    "nwkAddr": 5697,
    "manufId": 4107,
    "manufName": "eWeLight",
    "powerSource": "Mains (single phase)",
    "modelId": "ZB-DL01",
    "epList": [
        1
    ],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 257,
            "inClusterList": [
                0,
                3,
                4,
                5,
                6,
                8,
                4096
            ],
            "outClusterList": [],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "hwVersion": 0
                    }
                },
                "genOnOff": {
                    "attributes": {
                        "startUpOnOff": 255
                    }
                }
            },
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        }
    },
    "appVersion": 6,
    "stackVersion": 2,
    "hwVersion": 0,
    "dateCode": "20220518",
    "swBuildId": "2.0.1",
    "zclVersion": 3,
    "interviewCompleted": true,
    "meta": {
        "configured": 88764544
    },
    "lastSeen": 1679483391483,
    "defaultSendRequestWhen": "immediate"
}
```

closed 2023-05-02 by github-actions[bot]

aalmazanarbs, 2023-05-02:
The issues is not fixed.

BenJamesAndo, 2024-01-29:
It might be because this device doesn't actually support power on behavior. I have the same device and same problem. Looking at the screenshots of the device on Aliexpress it seems the only options available are brightness and power.