Koenkk/zigbee2mqtt #16236: [New device support]: Innr LED RS 230C

Timmmy001, 2023-01-17:
### Link

https://www.innr.com/en/product/smart-spot-colour/

### Database entry

{"id":3,"type":"Unknown","ieeeAddr":"0x943469fffe3af020","nwkAddr":27742,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1673979394701,"defaultSendRequestWhen":"immediate"}

### Comments

New led spot is displayed as unknown.

    Light colour 16 million colours
    Colour temperature 1800K - 6500K
    Light quality (CRI) >80
    Lumen 350 lm
    Beam angle 36°
    Rated lifespan 50.000 h
    Switch-on time < 0.5 sec
    Dimming range 5% - 100%

    Power consumption 6W
    Voltage/Frequency 220-240V/50-60Hz
    Protocol Zigbee 3.0
    Replacement wattage 50W

    Art number RS 230 C
    EAN code 8718781552039
    Art number RS 230 C-2
    EAN code 8718781552046

### Supported color modes

 Light colour 16 million colours

### Color temperature range

Colour temperature 1800K - 6500K

Timmmy001, 2023-01-18:
I have just found out that the bulb allready there:
https://www.zigbee2mqtt.io/devices/RS_230_C.html

i don't understand why it is not recognized then.

Koenkk, 2023-01-18:
The device did not fully pair, try pairing it again closer to the coordinator.

Timmmy001, 2023-01-19:
Thanks for the tip, I tried again and had a little more patience. After about 3 minutes it was then fully recognized.
With other devices felt it went faster without having a real experience with zigbee.

closed 2023-01-19 by Timmmy001