Koenkk/zigbee2mqtt #16239: Aqara Pet Feeder can't be triggered from Home Assistant

kimmeld, 2023-01-18:
### What happened?

The Aqara Pet Feeder (ZNCWWSQ01LM) has its 'feed' attribute exposed to Home Assistant as a select entity with one option, 'START'.  The problem is that, because there is only one option, Home Assistant is unable to trigger feeding as there is no way in the UI to make the state change.

### What did you expect to happen?

I expected to be able to trigger a feeding cycle.

### How to reproduce it (minimal and precise)

- Pair an Aqara C1 Pet Feeder.
- Look at the entity in Home Assistant.
- Change the Feed entity to 'START' (you _may_ be able to do this once after first adding the device)
- Observe that the state is now stuck at 'START' and impossible to change.

### Zigbee2MQTT version

1.29.1 commit: 7d67ffc

### Adapter firmware version

20220219

### Adapter

SLZB-06

kimmeld, 2023-01-18:
I've created a PR in zigbee-herdsman-converters which has the fix I've applied to my installation to make this work.

Ideally I'd like to see the 'feed' attribute appear as a button in Home Assistant, but I haven't figured out how to make that happen yet.  This is a quick interim fix.

addrum, 2023-02-01:
Portion weight and serving size input numbers also show up as unavailable and no way to set them from HA for me too but I can from zigbee2mqtt. 

Does the physical feed button work for you? It doesn't for me but not sure if that's related to zigbee2mqtt or a defect.

kimmeld, 2023-02-02:
Once my zigbee-herdsman-converters change (https://github.com/Koenkk/zigbee-herdsman-converters/pull/5360) makes it in to zigbee2mqtt you should be able to use it from HA.  I've applied that change by hand to my installation and it works well.

kimmeld, 2023-02-19:
I just upgraded zigbee2mqtt and my change to zigbee-herdsman-converters is in there and working, so closing this issue.

closed 2023-02-19 by kimmeld