Koenkk/zigbee2mqtt #16242: IKEA E1524/E1810 will not update

brhahlen, 2023-01-18:
### What happened?

I have three IKEA Tradfri E1524/E1810 Remotes, of which two will, for some reason, not update to latest Firmware Version and keep indicating that a new firmware is available.

[image]

When I run the OTA update, it will proceed to 100%, without any errors, but then it will report that the old version is still installed.

This has been occurring over the last few firmwares. Mostly I can't really be bothered, but now HA is showing the 2 devices as well that they need updating, so that triggers me :)

### What did you expect to happen?

I exepected that the firmwares would be updated.

### How to reproduce it (minimal and precise)

Get a E1524/E1810 and try to update it, I guess?

### Zigbee2MQTT version

1.29.2 commit: bb3e8f6

### Adapter firmware version

20211217

### Adapter

zzh! CC2652R

ikarisan, 2023-01-19:
Strange, my two E1524/E1810 are on

[2.3.014] (2019-04-01)
[2.3.014] (2019-07-08)

but no updates are found. 

Did you restart Zigbee2Mqtt?

brhahlen, 2023-01-24:
I've restarted Z2M multiple times.
I _think_ it's probably something in the Firmware that's wrong for those devices, but not sure.

nulano, 2023-01-26:
IIRC my one E1810 updated on the second try. The first time it got to 100% percent and kept the old version, on the second attempt it updated correctly. I thought it was related either to the battery or pressing the toggle button at the right time, but I can't remember anymore whether I did anything different the second time.

brhahlen, 2023-02-03:
Yeah, that doesn't work for me, unfortunately...
I have three updates pending now, that I cannot complete in any way :)

cturra, 2023-02-03:
for what it’s worth, i also so had 1 stubborn 5 button remote that would not update. i ended up fixing it by popping out the battery, then putting it back in. after ‘resetting’ it, i kept pressing one of the buttons every second and initiated the OTA update. that worked on the first try.

ChLah, 2023-02-07:
> for what it’s worth, i also so had 1 stubborn 5 button remote that would not update. i ended up fixing it by popping out the battery, then putting it back in. after ‘resetting’ it, i kept pressing one of the buttons every second and initiated the OTA update. that worked on the first try.

That solved it for me. I've had the same problem for a couple of weeks now. Removing the battery for a bit and then trying the OTA update again whilst pushing the arrow right button approx. every second helped.
Unfortunatelly the zigbee2mqtt page for the device (https://www.zigbee2mqtt.io/devices/E1524_E1810.html) does not mention this approach for troubleshooting.
Thank you very much, @cturra, you're a life saver. The update message in home assistant was really bothering me!

oxivanisher, 2023-02-08:
> for what it’s worth, i also so had 1 stubborn 5 button remote that would not update. i ended up fixing it by popping out the battery, then putting it back in. after ‘resetting’ it, i kept pressing one of the buttons every second and initiated the OTA update. that worked on the first try.

This also solved it for me. Although I had to be cautious to no press the button too fast, since the update seems to stop if you press a button during it (or at least at the beginning). Every 2 to 3 seconds worked great. Also the button pressed to keep the device alive does not matter, I tested most of them with my four E1524s.

johanneswilkens, 2023-03-02:
I also have the issue that they won't update, I press a button every 2 sec to wake them when pressing the update button, after a while they are indicating they are updating so I leave them to do their thing, but when I come back the next day they are still on 2.3.080 and didn't update. I haven't tried updating the ones which are currently running 2.3.014 because I have no issues with them.

mcmellow, 2023-03-11:
@cturra Thanks for your work around. This worked fine for me.

junosuarez, 2023-03-12:
> Unfortunatelly the zigbee2mqtt page for the device (https://www.zigbee2mqtt.io/devices/E1524_E1810.html) does not mention this approach for troubleshooting.

That documentation source code lives at https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/E1524_E1810.md - could someone from this thread consider contributing these instructions as a pull request over there?

brhahlen, 2023-04-12:
Comment to remove stale label

bjab, 2023-04-16:
> That documentation source code lives at https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/E1524_E1810.md - could someone from this thread consider contributing these instructions as [...]

Just updated docs. I've also disocvered, that inserting fresh battery helps with updated rejections :)
https://github.com/Koenkk/zigbee2mqtt.io/pull/2007

brhahlen, 2023-05-17:
Comment to remove stale label and because the issue is still present

closed 2023-06-24 by github-actions[bot]

GrumpyMeow, 2024-06-15:
Hi,
Leaving this for others as i finally was able to update the firmware of my Ikea remote. I swapped the battery with a fresh one. So apparently the remotes seem to check for a minimum battery-level. Mine reported a battery-level of "34". Of which i thought would be plenty to complete the update.