Koenkk/zigbee2mqtt #16244: SMSZB-120 missing sensors

federicosantolini, 2023-01-18:
### What happened?

Starting from the last two updates of Z2M, I am having problems with the SMSZB-120 siren. Before, I lost the ability to manage the siren independently from the smoke (which worked fine until December). Now it no longer passes me the value of the temperature sensor. Initially, I had to update because I could no longer stop the siren with the siren.turn_off service but since then everything has just gotten worse.

### What did you expect to happen?

See temperature value
[image]

### How to reproduce it (minimal and precise)

pair the SMSZB-120 siren to Z2M using conbee2 dongle

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

26720700

### Adapter

Conbee2

federicosantolini, 2023-02-22:
Still needing to be fixed.
[image]

closed 2023-04-02 by github-actions[bot]

garret, 2024-02-04:
I have the same issue after having just bought 4 of these smoke detectors (posted my comment here https://github.com/Koenkk/zigbee2mqtt/issues/17983#issuecomment-1925673850) Are you using a conbee usb stick? Maybe the temperature value can be solved by using another stick like the SONOFF one but it is strange that it was working before for you and now not anymore. Then it does not seem like a zigbee antenna issue. Have you tried to revert to a previous release of Z2M when everything was working?

Articole, 2024-03-30:
Hello,

I have the same problem with Conbee3 with last Firmware.
To getting Temperature and Battery work, i followed this in Z2M for the moment #14997