Koenkk/zigbee2mqtt #16245: [Feature request]: Set LED color on Bticino switches

federicosantolini, 2023-01-18:
### Is your feature request related to a problem? Please describe

Not related to an issue

### Describe the solution you'd like

I would like to have the ability to set the color of the LED on Bticino switches.

### Describe alternatives you've considered

no alternatives, just asking for a possibility.

### Additional context

On Bticino switches, it is already possible to control the LED separately (on and off). Since during the configuration the LED can be red, green, blue and purple, I assume it is a small RGB LED. It would be nice to be able to control it. It would have a use, for example, to indicate the state of an alarm (for example red if the alarm is armed).

Reinqusito, 2023-01-18:
+1 for this feature

giodaprile, 2023-01-18:
Me too for the feature. Thanks a lot

closed 2023-02-25 by github-actions[bot]

federicosantolini, 2023-04-12:
Please reopen it. I still like to have this feature