Koenkk/zigbee2mqtt #16246: Version not available in docker under Unraid->No update possible

canedje, 2023-01-18:
### What happened?

My Z2M docker under unraid shows version not available and because of that can not update.
Current version Z2M is 1.29.1
Version Unraid: 6.11.5
[image]

### What did you expect to happen?

Expected to see up-to-date or update-available

### How to reproduce it (minimal and precise)

It occurs without handling and is always there

### Zigbee2MQTT version

1.29.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonof zigbee 3.0 dongle plus-E

### Debug log

There is no debug log.
Docker is running oke
also updating by adding by hand to repository:1.29.2 doesn't update

lewistu, 2023-01-19:
Sounds similar to this issue: https://github.com/Koenkk/zigbee2mqtt/issues/16201

canedje, 2023-01-19:
your right. Looks like the same
I didn't found that issue

closed 2023-02-26 by github-actions[bot]