Koenkk/zigbee2mqtt #16250: [Feature request]: Support LED in Tube's Zigbee PoE (CC2652P2 variant)

longos, 2023-01-18:
### Is your feature request related to a problem? Please describe

So far, I have been using _Egony Stick V4_ with _EBYTE_ module. The LED status indication functionality was very helpful.
It was firmware CC1352P2_CC2652P_other_*.zip

Now I am using _Tube's Zigbee PoE_ with _RFSTAR RF-BM-2652P2_ module, (firmware CC1352P2_CC2652P_launchpad_*.zip) but there is no LED support in this firmware.

### Describe the solution you'd like

Is it possible to add LED support in this firmware ?

### Describe alternatives you've considered

For testing, I even soldered LEDs on pins DIO_6 and DIO_7 (This firmware supports LEDs for example in CircuitSetup.us module), but they do not work.

### Additional context

If this firmware supports LEDs in other solutions (on the same module), why don't LEDs work for Tube's Zigbee PoE ?

Koenkk, 2023-01-18:
No, this is a generic firmware with the goal to provide broad device support (with the sacrifice of not supporting all the coordinators features).

longos, 2023-01-19:
OK, that's clear. But from which it follows that the same firmware on some implementations (of the same module) supports LED, and on others - not.
All the time we are talking about the same chip on the same chip.

Koenkk, 2023-01-19:
@longos if the same pins are used as on the official TI launchpad (dev kit) it will work, if adapter makers deviate from that, it won't.

closed 2023-02-26 by github-actions[bot]