Koenkk/zigbee2mqtt #16253: Web-frontend will not start due no input validation on settings page

jjarven, 2023-01-18:
### What happened?

Zigbee2mqtt web frontend version 0.6.121 does not start.

### How to reproduce it (minimal and precise)

Open zigbee2mqtt web frontend. 
Select a zigbee device, go to settings, enter QOS value 3 and submit value.

When zigbee2mqtt is restarted, it will not start; socat.log has error "Refusing to start because configuration is not valid, found the following errors" and QOS value can only be 0, 1 or 2.

Luckily one can edit configuration.yaml manually and correct the issue.
--
Any incorrect values MUST NOT disable startup - it must use good known default.

### Zigbee2MQTT version

1.29.2 

### Adapter firmware version

20221102

### Adapter

CC2652

jjarven, 2023-02-20:
This is still an issue

jjarven, 2023-03-23:
.

closed 2023-05-01 by github-actions[bot]

TolianIPB, 2023-06-12:
The same problem.
Zigbee2MQTT Current version: 1.31.2-1
FW version : 0.6.10
Hardware : TTGO T-Internet-POE
ESP32 model : ESP32-D0WDQ5

`Refusing to start because configuration is not valid, found the following errors:
- QOS for 'XXXXXXX' not valid, should be 0, 1 or 2 got 5`

MKtuzi, 2024-05-12:
The same issue.