Koenkk/zigbee2mqtt #16258: [Feature request]: As part of #15699 please add support for Candle Mode to White Ambiance Hue bulbs.

lundyfpv, 2023-01-19:
### Is your feature request related to a problem? Please describe

As part of Issue #15699 effects (Candle, Fireplace & Color Loop) were added to Hue bulbs that support color ambiance and temperature. It would be nice if we could add candle mode to the Hue bulbs that are White Ambiance bulbs. The hue app seems to support candle mode on the model I have but it doesn't show up in Z2M under Exposes. 

https://www.zigbee2mqtt.io/devices/046677551780.html

This is the model that I am able to verify supports candle mode in the Hue app. 

### Describe the solution you'd like

Be able to call the candle effect in Z2M for White Ambiance Hue bulbs.

### Describe alternatives you've considered

None

### Additional context

None

Bastian007, 2023-01-19:
Same applies to this device: **Hue white and color ambiance LightStrip plus**
https://www.zigbee2mqtt.io/devices/8718699703424.html

Would be great if the new modes can be added there as well, Hue App supports it.

Koenkk, 2023-01-19:
For both it will be enabled in the next release on 1 February.

closed 2023-01-19 by Koenkk

Berlusconi13, 2023-01-26:
Candlelight is also available/working on
Zigbee Modell: LWO001 (Hue white Filament bulb G93 E27 bluetooth)

Koenkk, 2023-01-29:
@Berlusconi13 done.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Berlusconi13, 2023-01-31:
Tks👍

Berlusconi13, 2023-02-01:
@Koenkk 
Its works on another devices:
7602031P7 (Hue Go)
1742930P7 (Hue Impress)
1746430P7 (Hue Resonate)

Koenkk, 2023-02-01:
Enabled!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

lundyfpv, 2023-02-01:
Is there a better way for us to report which Hue Bulbs support these effects? Looks like my bulb was actually slightly different than the one I originally posted about. Both support it but my bulb is actually the following: https://www.zigbee2mqtt.io/devices/929002478401.html#philips-929002478401

Just seems pretty ineffecient to report every bulb that needs support for certain effects. If there is a better way I'm all ears!

Berlusconi13, 2023-02-02:
I would implement candle for all white ambiance and all effects for the color lights. That would be the best solution imo

olfon, 2023-05-04:
@Koenkk 
The effect is also supported by these models:

8718699673147 (LWA001)
9290023349 (LWA009)

Could you please add them? Thanks!

Koenkk, 2023-05-04:
@olfon it should already be enabled for these

olfon, 2023-05-04:
@Koenkk: Many thanks for your fast response!

I am running zigbee2mqtt v1.30.4 and for the mentioned models the candle effects are missing in the dropdown. However, it is enabled (and works) if I change these lines in these files

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/philips.js#L466
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/philips.js#L711

to

`        extend: philips.extend.light_onoff_brightness({disableHueEffects: false}),`

Am I missing something?

Koenkk, 2023-05-05:
@olfon ah it was disabled by default for brightness only, I've enabled it now.

konstafed, 2023-05-19:
929002241201 also supports candle and fire effects in hue app, however its not exposed in the effects selector. Tried to add ({disableHueEffects: false}) but no success with this particular model (zigbee2mqtt v1.30.4). Can you please add support for filament lamps as well?