Koenkk/zigbee2mqtt #16259: Tuya ZigBee Smart Knob Switch

cloudbr34k84, 2023-01-19:
### Link

https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html

### Database entry

Unable

### Comments

I bought this from the same links on the device page and have ended up with one that is not supported by Z2M. I have added it to my Tuya Zigbee Hub (not connected or anything to do with Z2M)

Here is the Device info from Tuya Developer if that helps

```
"result": {
    "active_time": 1674096928,
    "category": "wxkg",
    "category_name": "Wireless Switch",
    "create_time": 1674096928,
    "gateway_id": "bf61b5ef56f168251axufi",
    "icon": "smart/icon/ay157243613457092I4D/1637047911e1c9c436231.png",
    "id": "bf19e917757bc2be84vhdi",
    "ip": "",
    "lat": "-26.8170",
    "local_key": "91f6768807d47ecf",
    "lon": "153.0591",
    "model": "ZT-SY-RD",
    "name": "ZC-YED 旋钮调光场景开关",
    "node_id": "70ac08fffe264ddb",
    "online": true,
    "owner_id": "4096575",
    "product_id": "qja6nq5z",
    "product_name": "ZC-YED 旋钮调光场景开关",
    "sub": true,
    "time_zone": "+10:00",
    "update_time": 1674096928,
    "uuid": "70ac08fffe264ddb"
```

kalestarler, 2023-02-15:
Having the same issue - bought this knob because it was on the supported devices list. Model number is the same - ERS-10TZBVK-AA, however it shows up as "Unsupported" after pairing. 

Would the only way forward for now be to setup a new converter manually?

closed 2023-03-25 by github-actions[bot]

cloudbr34k84, 2023-03-25:
If it's says it's supported and yet it appears to not be then why is this closed. Either remove it from the support list or support it no??

junosuarez, 2023-03-28:
Could you share zigbee2mqtt debug logs that show it as unsupported when you try adding it to your zigbee2mqtt network? What I think is happening is multiple variants of this device being sold with the same model number, but appearing in zigbee with different identifiers. I have one of these physical devices and it works perfectly with my zigbee2mqtt network.