Koenkk/zigbee2mqtt #16261: 3A / Nue 2-gang light switch showing up as double GPO

fredskis, 2023-01-19:
### What happened?

Added new device (first of this type on my network) and it shows up as the wrong one. 

It identifies as `LXN-2S27LX1.0 `which in the vendor nue_3a.js definition is in fact listed as a double GPO, [NUE-AUWZO2](https://www.zigbee2mqtt.io/devices/NUE-AUWZO2.html): 
```
    {
        fingerprint: [
            {type: 'Router', manufacturerName: '3A Smart Home DE', modelID: 'LXN-2S27LX1.0', endpoints: [
                {ID: 1, profileID: 49246, deviceID: 0, inputClusters: [0, 4, 3, 6, 5, 4096, 8], outputClusters: [0]},
                {ID: 2, profileID: 49246, deviceID: 0, inputClusters: [0, 4, 3, 6, 5, 4096, 8], outputClusters: [0]},
            ]},
        ],
        model: 'NUE-AUWZO2',
        vendor: 'Nue / 3A',
        description: 'Smart Zigbee double power point',
        extend: extend.switch(),
        exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right')],
        meta: {multiEndpoint: true},
        endpoint: (device) => {
            return {left: 1, right: 2};
        },
    },
```

I believe the correct device should be [HGZB-042](https://www.zigbee2mqtt.io/devices/HGZB-042.html): 
```
    {
        zigbeeModel: ['FB56+ZSW1HKJ1.7', 'FB56+ZSW1HKJ2.5', 'FB56+ZSW1HKJ2.7'],
        model: 'HGZB-042',
        vendor: 'Nue / 3A',
        description: 'Smart light switch - 2 gang',
        extend: extend.switch(),
        exposes: [e.switch().withEndpoint('top'), e.switch().withEndpoint('bottom')],
        endpoint: (device) => {
            return {'top': 16, 'bottom': 17};
        },
        meta: {multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            await reporting.bind(device.getEndpoint(16), coordinatorEndpoint, ['genOnOff']);
            await reporting.bind(device.getEndpoint(17), coordinatorEndpoint, ['genOnOff']);
        },
    },
```

This seems to have happened in the past, somehow been resolved, and then happened again. 
#10497
#10934

I have attached a zip of my database.db: 
[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/10453285/database.zip)

### What did you expect to happen?

Device to be added as a double light switch, HGZB-042, not a double GPO, NUE-AUWZO2. 

### How to reproduce it (minimal and precise)

Add a new Nue / 3A double light switch to network. 

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

[log1.zip](https://github.com/Koenkk/zigbee2mqtt/files/10453281/log1.zip)

fredskis, 2023-02-01:
Hi @Koenkk sorry for directly mentioning you and dragging you here, looks like you may have fixed this in the past on #10497 so perhaps it's a simple fix. 
I attached my database.db to the original post above, if there's anything else I can provide, let me know please 😄

fredskis, 2023-02-01:
Ah that is a shame, seems this vendor does some strange things with their devices... 
I imagine if I plan on never using the 2 gang switches I should be able to locally override this with a custom external converter?

NZPy7h0n, 2023-02-08:
Is there anyway to manually overwrite this device by device?

closed 2023-03-19 by github-actions[bot]