Koenkk/zigbee2mqtt #16267: Livarno Home LED ceiling light: cannot change power_on_behavior

pohly, 2023-01-19:
### What happened?

All expected properties are found (state, brightness, color_temp, color_xy, effect, power_on_behavior, linkquality) and all of them can be set except for power_on_behavior.

### What did you expect to happen?

The default seems to be "on" (not visible in the "Exposes" tab, but that's how the light behaves). I'd like to change that to "off".

### How to reproduce it (minimal and precise)

Under "Exposes" in the zigbeemqtt, click on "power_on_behvior: off".

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZigBee 3.x.0 TI CP2102N

### Debug log

Jan 19 20:59:38 xbmc conmon[2056904]: Zigbee2MQTT:debug 2023-01-19 20:59:38: Received MQTT message on 'zigbee2mqtt/flur/licht/set' with data '{"power_on_behavior":"off"}'
Jan 19 20:59:38 xbmc conmon[2056904]: Zigbee2MQTT:debug 2023-01-19 20:59:38: Publishing 'set' 'power_on_behavior' to 'flur/licht'
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.169Z zigbee-herdsman:controller:endpoint Write 0x04cd15fffea8080c/1 genOnOff({"startUpOnOff":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.170Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x04cd15fffea8080c:14958/1 (0,0,1)
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.171Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":14958,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":12,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[16,13,2,3,64,48,0]}}
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.171Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,17,36,1,110,58,1,1,6,0,12,0,30,7,16,13,2,3,64,48,0,31]
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.183Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.183Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.183Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.183Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.184Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.187Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,12,202]
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.187Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,12,202]
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.187Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,12] - 202
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.187Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":12}
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.187Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.199Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,110,58,0,214]
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.199Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,110,58,0,214]
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.199Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [110,58,0] - 214
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.200Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":14958,"relaycount":0,"relaylist":[]}
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.200Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.247Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,26,68,129,0,0,6,0,110,58,1,1,0,39,0,243,227,181,0,0,6,24,13,4,134,3,64,110,58,29,148]
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.247Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,26,68,129,0,0,6,0,110,58,1,1,0,39,0,243,227,181,0,0,6,24,13,4,134,3,64,110,58,29,148]
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.247Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 26 - 2 - 4 - 129 - [0,0,6,0,110,58,1,1,0,39,0,243,227,181,0,0,6,24,13,4,134,3,64,110,58,29] - 148
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.247Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":14958,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":39,"securityuse":0,"timestamp":11920371,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[24,13,4,134,3,64]}}
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.249Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":13,"manufacturerCode":null,"commandIdentifier":4},"Payload":[{"status":134,"attrId":16387}],"Command":{"ID":4,"name":"writeRsp","parameters":[{"name":"status","type":32},{"name":"attrId","type":33,"conditions":[{"type":"statusNotEquals","value":0}]}]}},"address":14958,"endpoint":1,"linkquality":39,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.250Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Jan 19 20:59:38 xbmc conmon[2056904]: 2023-01-19T19:59:38.250Z zigbee-herdsman:controller:endpoint Write 0x04cd15fffea8080c/1 genOnOff({"startUpOnOff":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
Jan 19 20:59:38 xbmc conmon[2056904]: Zigbee2MQTT:error 2023-01-19 20:59:38: Publish 'set' 'power_on_behavior' to 'flur/licht' failed: 'Error: Write 0x04cd15fffea8080c/1 genOnOff({"startUpOnOff":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Jan 19 20:59:38 xbmc conmon[2056904]: Zigbee2MQTT:debug 2023-01-19 20:59:38: Error: Write 0x04cd15fffea8080c/1 genOnOff({"startUpOnOff":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
Jan 19 20:59:38 xbmc conmon[2056904]:     at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
Jan 19 20:59:38 xbmc conmon[2056904]:     at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22)
Jan 19 20:59:38 xbmc conmon[2056904]:     at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:105:13)
Jan 19 20:59:38 xbmc conmon[2056904]:     at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)
Jan 19 21:00:32 xbmc conmon[2056904]: 2023-01-19T20:00:32.844Z zigbee-herdsman:controller:endpoint Read 0x04cd15fffede3ab7/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
Jan 19 21:00:32 xbmc conmon[2056904]: 2023-01-19T20:00:32.844Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x04cd15fffede3ab7:52121/1 (0,0,1)
Jan 19 21:00:32 xbmc conmon[2056904]: 2023-01-19T20:00:32.845Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":52121,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":13,"options": [... cut ...] dcast":false,"destinationEndpoint":1}'
Jan 19 21:00:32 xbmc conmon[2056904]: Zigbee2MQTT:debug 2023-01-19 21:00:32: Received Zigbee message from 'dach/licht', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Jan 19 21:00:32 xbmc conmon[2056904]: 2023-01-19T20:00:32.951Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Jan 19 21:00:32 xbmc conmon[2056904]: Zigbee2MQTT:debug 2023-01-19 21:00:32: Succesfully pinged 'dach/licht' (attempt 1/2)

pohly, 2023-01-19:
This ceiling light also shows up fine in HomeAssistant when using this configuration.yaml:
```
homeassistant: true
advanced:
   homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false
```

However, in HomeAssistant, the devices UI only shows an on/off toggle and let's me configure the power on behavior. Changing that there fails the same way.

Should the other attributes also show up in HomeAssistant? I am on an older version of it (2022.9.1). I probably should update that before filing further issues, or is this a known problem?

Koenkk, 2023-01-21:
Lidl lights don't support power on behaviour, this feature will be disabled for all Lidl lights in the next z2m release.

closed 2023-01-21 by Koenkk