Koenkk/zigbee2mqtt #16268: [New device support]: HT-SLM-2

janogre, 2023-01-19:
### Link

https://heimgard.com/produkt/heimgard-smart-elektronisk-dorlas

### Database entry

{"id":6,"type":"EndDevice","ieeeAddr":"0x94deb8fffe0223a3","nwkAddr":37403,"manufId":4098,"manufName":"Heimgard Technologies","powerSource":"Unknown","modelId":"HT-SLM-2","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":10,"inClusterList":[0,1,3,4,5,32,257],"outClusterList":[3,25],"clusters":{"closuresDoorLock":{"attributes":{"soundVolume":3,"lockState":0}},"genBasic":{"attributes":{"modelId":"HT-SLM-2","manufacturerName":"Heimgard Technologies","appVersion":0}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":180}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe88282d","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":0,"zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1674158766072,"defaultSendRequestWhen":"active"}

### Comments

I tried following the page for unsupported devices, but got stuck after adding the external converter.
The external converter should be a copy of the excisting converter "home_control_as.js".
The difference is the model and the vendor, HT-SML-2 and Heimgard Technologies. (see external converter below)

Hopefully this was understandable.

Regards,
janOgre

### External converter

```shell
const exposes = require('../lib/exposes');
const fz = require('../converters/fromZigbee');
const tz = require('../converters/toZigbee');
const reporting = require('../lib/reporting');
const e = exposes.presets;
const ota = require('../lib/ota');

module.exports = [
    {
        zigbeeModel: ['HT-SLM-2'],
        model: 'HT-SLM-2',
        vendor: 'Heimgard Technologies',
        description: 'Heimgard door lock pro',
        fromZigbee: [fz.lock, fz.battery],
        toZigbee: [tz.lock, tz.lock_auto_relock_time, tz.lock_sound_volume],
        ota: ota.zigbeeOTA,
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
            await reporting.lockState(endpoint);
            await reporting.batteryPercentageRemaining(endpoint);
            await endpoint.read('closuresDoorLock', ['lockState', 'soundVolume']);
        },
        exposes: [
            e.lock(), e.battery(), e.auto_relock_time().withValueMin(0).withValueMax(3600), e.sound_volume()],
    },
];
```

closed 2023-02-27 by github-actions[bot]

bskjon, 2024-01-04:
Hi, were you able to obtain any infomration about the device from the manufacturer?
I'm kind of stuck trying to figure out how to get more or proper infomration from and about the device. For instance, the childlock tricks the converter into seeing it as the opposite state.
Getting a bit annoyed of getting the device in a lockout state and being unable to unlock the door with z2m..

MathiasReker, 2024-03-13:
@bskjon did you make it work with z2m?

bskjon, 2024-03-13:
> @bskjon did you make it work with z2m?

I've put it on ice for now. It seems like the data or commands are not of the standard or xommon types (with exception to the currently working ones) . I believe I would need to read the characteristics or similar in order to see what it actually supports or responds to.

MathiasReker, 2024-03-13:
I'm contemplating purchasing the HT-SLM-2 and I'm interested in contributing to the integration with z2m. Have you tested it on the latest firmware? Could you provide information on which features are currently functional and which ones still need to be implemented?

bskjon, 2024-03-13:
I have no diea which firmware its using, and i have yet to find a firmware on Heimgard's site. Seems like they have only released the wattle firmware. The only functionality that i can say works is unlock/lock, and thats it. Override lockout is not present, show pincode does nothing, and the lock/unlock state gets missinterpereted if you press the function key (childlock on/off).
z2m does not see the OTA or firmware version on mine at all.

bskjon, 2024-03-13:
> You can find the latest firmware here: https://heimgard.com/support/firmware-filer-til-vare-produkter

Yes, but the HT-SLM-2 is not listed there.

bskjon, 2024-03-21:
Inspecting the state of the device, it apperently the current version (not listed under OTA for some reason)
`    "update": {
        "installed_version": 10012,
        "latest_version": 10012,
        "state": "idle"
    },`

rophome, 2024-12-23:
Enyone got this device to return pin in the payload. i will really like to see in home assistant who have unlocked the door.

bskjon, 2024-12-23:
> Enyone got this device to return pin in the payload. i will really like to see in home assistant who have unlocked the door. 

No. I think we would have to create a custom converter to make it work properly. There are too many issues with the default implementation. I think we have to do something similar to the nimly pro/ultimate

bskjon, 2025-01-25:
For those of you who still have this device, the only way i can see we can get this to fully work, is if heimgard actually releases their non standard zigbee attributes for the 'closuresDoorLock cluster (i believe). 
I've now tried multiple ways to interact with the device, but its the same limitations everywhere. If they never release the docs, some of us have to aquire a heimgard hub, then proceed to sniff it 🙃 .
At the very least, im trying to implement the standard stuff that i managed to play with in deconz, also might be able to "prevent" it from saying the door is unlocked when you unlock the internal handle (child lock)

M4YOO, 2025-03-09:
I made it work. I first created my own external converter, to get all the needed attributes:
[https://pastebin.com/cSd2kHeR](url)

It will give you a sensor for action source sensor, and action user sensor. You can then create a helper to translate those numbers. 

I create this template sensor:
`{% if is_state('sensor.smartlock_action_source_sensor', '0') %}
  Keypad
{% elif is_state('sensor.smartlock_action_source_sensor', '1') %}
  Home Assistant
{% elif is_state('sensor.smartlock_action_source_sensor', '2') %}
  Button on the lock
{% elif is_state('sensor.smartlock_action_source_sensor', '3') %}
  RFID tag
{% elif is_state('sensor.smartlock_action_source_sensor', '4') %}
  Fingerprint
{% else %}
  Unknown entity
{% endif %}`

You can do the same with the action user. I.e.:
{% if is_state('sensor.smartlock_action_user_sensor', '0') %}
  Admin
{% elif is_state('sensor.smartlock_action_user_sensor', '1') %}
  My self
{% elif is_state('sensor.smartlock_action_user_sensor', '2') %}
  Wife
{% elif is_state('sensor.smartlock_action_user_sensor', '3') %}
  Neighbor
{% else %}
  Unknown person
{% endif %}