Koenkk/zigbee2mqtt #16270: CCTFR6100

lka1987, 2023-01-19:
### Link

https://www.se.com/ww/en/product/CCTFR6100Z3/radiator-thermostat-wiser-z3-white/

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0x04cd15fffe7cca59","nwkAddr":12919,"manufId":4190,"manufName":"Schneider Electric","powerSource":"Battery","modelId":"CCTFR6100"

### Comments

I could not find the info from the device 

### External converter

```shell
zigbeeModel: ['iTRV'],
        model: 'WV704R0A0902',
        vendor: 'Schneider Electric',
        description: 'Wiser radiator thermostat',
        fromZigbee: [fz.ignore_basic_report, fz.ignore_haDiagnostic, fz.ignore_genOta, fz.ignore_zclversion_read,
            fz.legacy.wiser_thermostat, fz.legacy.wiser_itrv_battery, fz.hvac_user_interface, fz.wiser_device_info],
        toZigbee: [tz.thermostat_occupied_heating_setpoint, tz.thermostat_keypad_lockout],
        exposes: [exposes.climate().withSetpoint('occupied_heating_setpoint', 7, 30, 1).withLocalTemperature(ea.STATE)
            .withRunningState(['idle', 'heat'], ea.STATE).withPiHeatingDemand()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            const binds = ['genBasic', 'genPowerCfg', 'hvacThermostat', 'haDiagnostic'];
            await reporting.bind(endpoint, coordinatorEndpoint, binds);
            await reporting.batteryVoltage(endpoint);
            await reporting.thermostatTemperature(endpoint);
            await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
            await reporting.thermostatPIHeatingDemand(endpoint);
            // bind of hvacUserInterfaceCfg fails with 'Table Full', does this have any effect?
            await endpoint.configureReporting('hvacUserInterfaceCfg', [{attribute: 'keypadLockout', reportableChange: 1,
                minimumReportInterval: constants.repInterval.MINUTE, maximumReportInterval: constants.repInterval.HOUR}]);
```

storacm, 2023-02-04:
Have the same device. Branded Schneider, but is a Drayton Wiser Radiator Thermostat which is marked a compatible with zigbee2mqtt on Zigbee DEvice Compability Repository. But the Device is not supported in zigbee2mqtt when branded Schneider.
Has anyone got it to work?

Farmanden, 2023-02-05:
Got it somewhat working, good enough for now, running_state and temperatureCalibration is not working (yet). 

[image]

`{
    "linkquality": 255,
    "battery": 75,
    "local_temperature_calibration": null,
    "pi_heating_demand": 73,
    "running_state": null,
    "voltage": 2.8,
    "elapsed": 36409,
    "local_temperature": 20.1,
    "system_mode": "heat",
    "occupied_heating_setpoint": 21
}`

Just installed zigbee2mqtt yesterday, so bare with me :)

    {
        zigbeeModel: ['CCTFR6100'],
        model: 'WV704R0A0902',
        vendor: 'Schneider Electric',
        description: 'Wiser radiator thermostat',
        fromZigbee: [fz.ignore_basic_report, fz.ignore_haDiagnostic, fz.ignore_genOta, fz.ignore_zclversion_read,fz.legacy.wiser_thermostat,
            fz.legacy.wiser_itrv_battery ,fz.hvac_user_interface, fz.wiser_device_info],
        toZigbee: [tz.wiser_sed_thermostat_local_temperature_calibration, tz.thermostat_local_temperature,tz.thermostat_occupied_heating_setpoint,.
        tz.thermostat_system_mode, tz.thermostat_keypad_lockout],
        exposes: [exposes.climate().withLocalTemperature().withSetpoint('occupied_heating_setpoint', 4, 30, 0.5).withSystemMode(['off', 'auto', 'heat'])
            .withPiHeatingDemand().withLocalTemperatureCalibration(-12.8, 12.7, 0.1, ea.STATE_SET).withRunningState(['idle', 'heat'], ea.STATE)],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            const binds = ['genBasic', 'genPowerCfg', 'hvacThermostat', 'haDiagnostic'];
            await reporting.bind(endpoint, coordinatorEndpoint, binds);
            await reporting.batteryVoltage(endpoint);
            await reporting.thermostatTemperature(endpoint);
            await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
            await reporting.thermostatPIHeatingDemand(endpoint);
            // bind of hvacUserInterfaceCfg fails with 'Table Full', does this have any effect?
            await endpoint.configureReporting('hvacUserInterfaceCfg', [{attribute: 'keypadLockout', reportableChange: 1,
                minimumReportInterval: constants.repInterval.MINUTE, maximumReportInterval: constants.repInterval.HOUR}]);
            //endpoint.saveClusterAttributeKeyValue('hvacThermostat', {schneiderWiserSpecific: 1, systemMode: 4, ctrlSeqeOfOper: 2});
        },
    },

storacm, 2023-02-17:
Is it getting implemented into zigbee2mqtt or do we need to add external converter?

closed 2023-03-28 by github-actions[bot]

etnoy, 2023-10-19:
For everyone else who lands here from Google regarding CCTFR6100 support, it was included in the 15.53.0(https://github.com/Koenkk/zigbee-herdsman-converters/compare/v15.52.0...v15.53.0) release of z2m: https://github.com/Koenkk/zigbee-herdsman-converters/pull/6047