Koenkk/zigbee2mqtt #16272: Salus Controls PS600 - random "model ID" (zigbeeModel)

marekbm, 2023-01-20:
### What happened?

PS600 generates a differen "model ID" (zigbeeModel) after each pairing. And also it changes it between it starts pairing and it is paired. The beginning is always the same but not the end.
Also if I add the whole new name into the "/zigbee-herdsman-converters/devices/salus_controls.js" then for example OTA gives me an error regarding non existing model ID.

### How to reproduce it (minimal and precise)

Is it possible to add something like "end of string" char or sequence? This is the first device that I have that do the mess like this.

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

20220219

### Adapter

LAUNCHXL-CC1352P-2

robertwigley, 2023-02-16:
I have the same problem. Random characters are generated after PS600 e.g. 'PS600   � ��  �   �u  �  �   �'.

I've had to create a custom external converter file to be able to add this, but it's very difficult to get the random characters into it correctly. I've only succeeded with one of the two sensors I have.

mikeyyuen, 2023-02-16:
Having the same problem. It looks like this line https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/index.js#L39 should us to put the `zigbeeModel` to just `PS600` and it should strip the null bytes from the devise as it registers. However I can't seem to get that to work not sure if anyone smarter might be able to spot why?

```js
    zigbeeModel = zigbeeModel ? zigbeeModel.replace(/\0.*$/g, '').trim() : null;
```

Incidentally i finally figured out how to get the right model id for a custom converter, I had to open database.db and copy/paste it out of there.

robertwigley, 2023-02-16:
> Incidentally i finally figured out how to get the right model id for a custom converter, I had to open database.db and copy/paste it out of there.

Nice one! At least that makes getting the correct ID possible on the first attempt. It took me many attempts copying from the log file and some were split across lines and simply didn't work, as I guess the characters were not being displayed properly.

On a separate subject, as I am a novice when it comes to these custom converters, is there a way to bring the battery information in? Apparently, there are some battery attributes including voltage available when it's added via ZHA.

robertwigley, 2023-02-17:
Thanks @mikeyyuen. I tried `exposes: [e.battery(), e.temperature()]`, but it doesn't seem to pull the battery data in. Never mind. The main thing is it's reporting temperature.

mikeyyuen, 2023-02-25:
@robertwigley the following external converter gave me battery support:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = {
    zigbeeModel: [
        'PS600<your model id here>',
    ],
    model: 'PS600',
    vendor: 'Salus Controls',
    description: 'Pipe temperature sensor',
    fromZigbee: [fz.temperature, fz.battery],
    toZigbee: [],
    meta: {battery: {voltageToPercentage: '3V_2500'}},
    exposes: [e.battery(), e.temperature()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(9);
        await reporting.bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'genPowerCfg']);
        await reporting.temperature(endpoint);
        await reporting.batteryVoltage(endpoint);
    },
    ota: ota.salus,
};
```

I had to change the reporting and binding a little, which i've not played with before so I'm not 100% ready to PR it!

If it doesn't cause me (or anyone else here) any issues, i'll PR the config change. I had to press the "reconfigure" button for the device in the about page to get it to show up immediately, I don't know if that's just me being impatient!

robertwigley, 2023-02-25:
Brilliant @mikeyyuen. I'm now seeing battery info. Thanks.

mikeyyuen, 2023-03-03:
I think this can now ne closed.

Koenkk, 2023-03-04:
Thanks @mikeyyuen !

closed 2023-03-04 by Koenkk