Koenkk/zigbee2mqtt #16273: Unable to connect to SLZB-06 via TCP socket

beol, 2023-01-20:
### What happened?

I have SLZB-06 flashed with [CC1352P2_CC2652P_other_coordinator_20220219](https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_other_coordinator_20220219.zip).

Socket connected however it always failed capabilities probing, timing out after 6s.
```
Zigbee2MQTT:info  2023-01-20 19:07:47: Starting zigbee-herdsman (0.14.83-hotfix.0)
Zigbee2MQTT:debug 2023-01-20 19:07:47: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"baudRate":115200,"path":"tcp://192.168.86.53:6638"}}'                                   
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[69,120,174,56,145,14,175,229,181,15,3,226,115,58,56,147],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"path":"tcp://192.168.86.53:6638"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening TCP socket with 192.168.86.53:6638 +0ms
  zigbee-herdsman:adapter:zStack:znp:log Socket connected +28ms
  zigbee-herdsman:adapter:zStack:znp:log Socket ready +2ms
  zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload +4ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:log Skip bootloader for CC2652/CC1352 +1s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
Zigbee2MQTT:error 2023-01-20 19:08:08: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-01-20 19:08:08: Failed to start zigbee
```

No problem whatsoever if I connect the adapter via USB to my pi and also docker on my Mac.

### What did you expect to happen?

Zigbee2MQTT connects to adapter.

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

20220219

### Adapter

SMLIGHT SLZB-06 Zigbee ethernet USB POE WiFi LAN adapter

### Debug log

```
➜  zigbee2mqtt git:(master) ✗ DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.29.2 start
> node index.js

Zigbee2MQTT:debug 2023-01-20 19:07:47: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
Zigbee2MQTT:info  2023-01-20 19:07:47: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2023-01-20.19-07-47' filename: log.txt
Zigbee2MQTT:debug 2023-01-20 19:07:47: Removing old log directory '/opt/zigbee2mqtt/data/log/2023-01-20.15-18-07'
Zigbee2MQTT:info  2023-01-20 19:07:47: Starting Zigbee2MQTT version 1.29.2 (commit #bb3e8f6)
Zigbee2MQTT:info  2023-01-20 19:07:47: Starting zigbee-herdsman (0.14.83-hotfix.0)
Zigbee2MQTT:debug 2023-01-20 19:07:47: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"baudRate":115200,"path":"tcp://192.168.86.53:6638"}}'                                   
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[69,120,174,56,145,14,175,229,181,15,3,226,115,58,56,147],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"path":"tcp://192.168.86.53:6638"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening TCP socket with 192.168.86.53:6638 +0ms
  zigbee-herdsman:adapter:zStack:znp:log Socket connected +28ms
  zigbee-herdsman:adapter:zStack:znp:log Socket ready +2ms
  zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload +4ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:log Skip bootloader for CC2652/CC1352 +1s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
Zigbee2MQTT:error 2023-01-20 19:08:08: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-01-20 19:08:08: Failed to start zigbee
Zigbee2MQTT:error 2023-01-20 19:08:08: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-01-20 19:08:08: Exiting...
Zigbee2MQTT:error 2023-01-20 19:08:09: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)       
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:58:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:101:27)
    at start (/opt/zigbee2mqtt/index.js:107:5)
```

phillipmj, 2023-01-24:
Hi Beol, 

I have the same adapter as you, running with no problems, did a esp firmware update on it (response is way faster)

beol, 2023-02-09:
> Hi Beol,
> 
> I have the same adapter as you, running with no problems, did a esp firmware update on it (response is way faster)

Thanks. Reached out the team who are working on the adapter and we’re sorting out the issue.

closed 2023-02-09 by beol

mfrimannm, 2024-06-14:
What was the solution?

beol, 2024-06-16:
I reached out to the manufacturer, and they sent me a replacement.

mfrimannm, 2024-06-19:
Okay, i found out it was because of this (Zigbee2MQTT: Error Network Commissioning Timeout (panID or extendedPanID already exists nearby)) or the guide here fixed it https://smlight.tech/manual/slzb-06/guide/faq-and-lifehacks/#zigbee2mqtt-error-network-commissioning-timeout-panid-or-extendedpanid-already-exists-nearby